esphome/feature-requests #12: Force update option for binary sensor

HugoVonHory, 2019-01-27:
**Is your feature request related to a problem/use-case? Please describe.**
Binary sensor value is logged to homeassistant DB only on value change event via MQTT. Displaying graphs from values doesn't show correct state, if last state change isn't in displayed timeframe (no data in result). Hartbeat filter option in config publishes repeatedly state to MQTT correctly, force_update option should enable loging to DB as well.

**Describe the solution you'd like:**
add configuration option "force_update"
https://www.home-assistant.io/components/binary_sensor.mqtt/#force_update

**Additional context:**

casperklein, 2019-01-27:
I fully understand your point of view and respect it. And as you said, for most use-cases there are alternative ways to get it done.

For me however it means, when migrating a device which previously used a component (MQTT) that supported force_update, to esphome native api, I have to adjust my automation (node-red) and it's getting more complex.

One last example, then I'll be quiet :-)

I get a notification, when my washing machine has finished. This is done by checking, if power usage is 3mins constant at a fixed low value (2 Watts).

(Power sensor updates every 60s.)

With force_update: Received 2W three times in a row? --> if yes --> notify

Without force_update: Received 2W? --> sleep 1min --> query HA if value is still 2W --> if yes --> sleep 1min --> query HA if value is still 2W --> if yes --> notify

HugoVonHory, 2019-01-29:
> There's a fill option just for that
It works only for future from last value in time frame (e.g. last value to now), but not for history to last value before time frame, or even if no value is there.
In general force update is optional, so it could be enabled only realy required and with appropriate update interval.

dainman, 2019-03-02:
I asked this same question in discord about a month ago and I have been playing with settings in Grafana, however my graphs are not as consistant as the were when using MQTT and the force_update.  Here are a few comments to your view points

1 - Fill previous only works if there is a measurement in the timeframe that you have selected.  Example I have alot of my graphs that are set to 6 hour resolution and in my home some time it can go 7-8 hours without a change so then the sensor disappears unless I widen the time frame out to say 12 hours

2 - as for unavailable that only shows in home assistant.  if you look in Grafana and again there is a measurement in the time frame it will fill previous as we told it to or if not it will show a blank

3 - I saw no difference in performance of my HA installation from using MQTT with force_update enabled to now using the API.  I may not have a huge install but I have about 6 ESP32s with 2 temps and a humidity sensor each

It would be nice to have a force_update with a variable interval, however parity with MQTT force_update would be acceptable as well

messier433, 2019-07-21:
Hi,

there is a simple way to override the force_update flag inside the python code:
e.g. For ESP home sensors, cou can add following lines to  /srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/esphome/sensor.py

```python
class EsphomeSensor(EsphomeEntity):
    ...
    """Force update fix."""
    @property
    def force_update(self) -> bool:
        """Return True if state updates should be forced.
        If True, a state change will be triggered anytime the state property is
        updated, not just when the value changes.
        """
        return True
```

This forces updates for all ESP home sensors. So its an ugly hack but it does the job for now...

@OttoWinter: Grafanas fill(previous) has issues when there is no previous data in the current time window. Also after grouping mean values, a 0 power value is not shown as zero anymore but as the average of 0 and the neighboaring samples. And in addition it would be nice to differentiate if a sensor is sending no data or the same data

corneyl, 2019-08-26:
Another reason for allowing the `force_update` option is if you want to count totals. I have e.g. a pulse sensor measuring the rotations of a disk in a power meter. If for two periods of time this is the same the state will not be updated and if you then sum the total of the pulses (using the statistics sensor in HA), the amount of pulses in the second window are not counted because there was no state update.

OttoWinter, 2019-08-27:
@corneyl So why not use the pulse_counter + integration sensor in esphome then?

corneyl, 2019-08-28:
That would make the problem smaller, but it's still possible that two 5-minute windows have the same amount of pulses. Making the integration window longer would solve this, but then you lose data when updating the configuration/firmware.

OttoWinter, 2019-08-29:
@corneyl That's if you do the integral on the HA side, but if you would use the integration sensor in esphome you would not have that issue.

messier433, 2019-08-30:
@OttoWinter: Just a side note: the integration sensor in esphome is still not integrating correctly: https://github.com/esphome/issues/issues/533 ;)

haufes, 2019-10-21:
@OttoWinter  I also would like to see force_update as an option for sensors. I am as well monitoring power usage (on my house) using a Pulse Counter and pushing it into influx. I've considered using fill(previous), but this issue with that, it isn't possible to distinguish between the metric not changing and no metric at all. 

EG. When the power is out the graph would display the last metric and would show more consumption than actually happened.

Other idea that maybe options that fill some of the need and wouldn't fill the db
- only force if the measured value is over a unit, in a range or not a unit.
- if the esp is determined to be offline emit a value for the sensor (But this wouldn't help if HA was also offline)

PS: Love ESPHome, great job!

closed 2019-10-21 by OttoWinter