esphome/feature-requests #4: Support for MAX31865 board

grantoboy, 2019-01-27:
What new integration would you wish to have?

MAX31865

If possible, provide a link to an existing library for the integration:

https://github.com/adafruit/Adafruit_MAX31865

Please describe your use case for this integration and alternatives you've tried:

Used to convert resistance to digital especially designed for RTD thermal resistance temperature detectors.

These resistance probes are used in places digital probes are not use (hot water systems and boilers)

This board allows the probes data to be converted to digital data.

Additional context:

devanl, 2019-06-26:
Put this together but I wanted to test before submitting.
https://github.com/devanl/esphome/commit/4d0030ecbb71f6e9f2a23fa67aa110950654ab85

Not sure when it will happen...

Misiu, 2019-11-20:
Any updates on this? I'm planning to add MAX31865 with PT100 so it would be really useful for me.

devanl, 2019-11-20:
Should be ok to close this?
https://github.com/esphome/esphome/pull/832

Misiu, 2019-11-20:
@devanl sorry, I didn't found that issue before. I think this issue can be closed then.

Was the support for MAX32865 already released in latest version? I can't find it listed on the devices page

OttoWinter, 2019-11-20:
@Misiu It will be in the next version

Closing as per @devanl

closed 2019-11-20 by OttoWinter

Misiu, 2019-11-22:
@OttoWinter and when can I expect a new release?
Sorry for the question, but I'm really new to ESPHome.
1.13.0 was released in May, 1.14.0 in November, but I hope we won't have to wait  6 months for 1.15.0 🙂

glmnet, 2019-11-22:
@Misiu 

You can use the **dev** version (even if you are on hass.io there is a specific add-on) and then you can use the bleeding edge ESPHome version (just rebuild the hass.io image), use it for the nodes you need the latest bits, you can have mixed nodes running different versions.