esphome/feature-requests #5: LIS3DH tripple-axis accelerometer

Mynasru, 2019-01-27:
**What new integration would you wish to have?**
Lis3DH tripple-axis accelerometer

**If possible, provide a link to an existing library for the integration:**
[Adafruit](https://github.com/adafruit/Adafruit_LIS3DH)
[Sparkfun](https://github.com/sparkfun/SparkFun_LIS3DH_Arduino_Library)

**Please describe your use case for this integration and alternatives you've tried:**
Amazingly cheap and versatile accelerometer + 3 10bit ADCs + knock detection + low power consumption + I2C + SPI + temperature change sensor. This sensor can for example wake up your esp when a double knock is detected.

**Additional context:**
-

lboue, 2024-01-03:
We also have the [esp-idf Driver for the LIS3DH 3-axes digital output accelerometer
](https://github.com/gschorcht/lis3dh-esp-idf).

adavidzh, 2024-01-06:
Not sure if relevant, but there's also a [Seeed Arduino driver](https://github.com/Seeed-Studio/Seeed_Arduino_LIS3DHTR).

lboue, 2024-03-10:
The is LIS3DH accelerometer for detecting taps/orientation on [Adafruit Matrix Portal S3](https://learn.adafruit.com/adafruit-matrixportal-s3).

scottbob09, 2025-03-25:
still nothing?