esphome/feature-requests #6: DSMR Meter

olivierfaber, 2019-01-27:
**What new integration would you wish to have?**
DSMR aka Slimme Meter 

**If possible, provide a link to an existing library for the integration:**
https://opencircuit.nl/Blog/Slimme-meter-uitlezer
https://github.com/mrWheel/DSMRlogger2HTTP
https://github.com/matthijskooijman/arduino-dsmr

**Please describe your use case for this integration and alternatives you've tried:**
I have used a USB to serial DSMR integration in Home Assistant before, but now my HASS.IO is virtualised and it's nowhere close to the meter which rules out a direct USB connection.

**Additional context:**
A DIY version to connect a ESP8266 directly to the meter (powered by the meter) is quite easy to do and there  is also a PCB kit:
https://opencircuit.nl/Product/13859/Slimme-meter-uitlezer-V3-bouwpakket-met-ESP-01

Pimmetje, 2019-04-12:
@Mynasru If u need help testing or something else please let me know.

Mynasru, 2019-04-18:
@Pimmetje I got the hardware working, but do not yet get reliable results from the software. I plan to port [this](https://github.com/matthijskooijman/arduino-dsmr) code to ESPHome (if allowed by the licence). I'll put the code on my GitHub and contact you for testing.

GraafG, 2019-05-18:
I like to add some existing libraries:
https://github.com/romix123/ESPEasyDSMR5/blob/master/_P110_P1WifiGateway.ino
https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini-and-P1-Energy-Meter

I'm using D1 Mini with espeasy firmware at this moment.
D1 and this module: http://www.esp8266thingies.nl/wp/ (I have some boards for d1 mini for sale by the way)

olivierfaber, 2019-06-26:
> 
> 
> @Pimmetje I got the hardware working, but do not yet get reliable results from the software. I plan to port [this](https://github.com/matthijskooijman/arduino-dsmr) code to ESPHome (if allowed by the licence). I'll put the code on my GitHub and contact you for testing.

Any news? Is there anything I can do?

Pimmetje, 2019-06-26:
Note that the code does not yet support DSMR 3 https://github.com/matthijskooijman/arduino-dsmr/issues/22 (that what my meter is using).

olivierfaber, 2019-08-01:
> 
> 
> I'm working on this, already made a working custom sensor. Next month I'll try to integrate it into Esphomelib.

Could you share your custom sensor config?

geelenbert, 2019-08-08:
I would like this feature too. 

can you share the custom sensor config? Maybee it is something we or someone else can work with.

Theoreet, 2019-08-28:
there's some tasmota coding as you can read : [http://www.esp8266thingies.nl/wp/software/](url)
+1 from me with this feature request

Theoreet, 2019-09-23:
> I'm working on this, already made a working custom sensor. Next month I'll try to integrate it into Esphomelib.

could you maybe tell us something about it, of give us an update ?

geelenbert, 2019-09-24:
> I'm working on this, already made a working custom sensor. Next month I'll try to integrate it into Esphomelib.

yes, i'm interested in this too. Anything we can do to help?

Taigar2015, 2019-10-28:
Also interested.
I have ESMR5.
Any news/updates?

nldroid, 2019-11-27:
I created a custom component but didn't test it (because i don't have a cable yet)

https://github.com/nldroid/DsmrP1CustomSensor

rspaargaren, 2019-12-15:
I am working on a custom library for the DSMR reader. Based on nldroid his input but not based on softwareserial. In the Yaml I am initiating the UART:

uart:
  id: uart_bus
  rx_pin: GPIO3
  baud_rate: 115200

And within the custom library I am reading the buffer line or line:

  void update() override {  
  ESP_LOGD("DmsrCustom","Updating..");  
  digitalWrite(D5,HIGH); //Data is only submitted if pin is HIGH
  if (available())
    {
      ESP_LOGD("DmsrCustom","Data ready..start if loop");       
      memset(telegram, 0, sizeof(telegram));
      while (available())
      {
        ESP.wdtDisable();
        int len = readBytesUntil('\n', telegram, P1_MAXLINELENGTH);
        ESP.wdtEnable(1);
        telegram[len] = '\n';
        telegram[len + 1] = 0;
        yield();
        ESP_LOGD("DmsrCustom","*%s",telegram);
        bool result = decode_telegram(len + 1);

etc... etc...

If I run the custom library without a polling interval I can read the full message but as soon as I activate pollinginterval I can only read the second half of the message. Based on my limited guess is that without the pollinginterval the routine is fast enough to emply the uart buffer line for line and even wait for the new line to arrive. Within the polling interval the UART buffer is not updated within the custom routine and only the last part can be read what was still in the buffer. (FIFO) so the first part is already out..?

rspaargaren, 2019-12-21:
A working custom sensor can be found at:
https://github.com/rspaargaren/DSMR_ESPHOME
This library is based on the dsmr.h -> https://github.com/matthijskooijman/arduino-dsmr/blob/master/src/dsmr.h. So please read this page for more info on how to connect. My code is a bit WIP.

mcflexus, 2019-12-22:
trying to compile it for esp32 but it won't because D5 is not defined.
D5 is called from setup and update functions in dsmr_p1_sensor.h
is this D5 pin the rts pin?

bramvdpol, 2019-12-30:
I ran rspaargaren's code on a wemos d1, however I was not getting any power readings. I suspect this may be due to way I wired the p1 port, with RTS permanently high. I commented the logging part back in, while the "READING READY" messages got through, no data was logged. Was running [this](https://github.com/fliphess/esp8266_p1meter) with few issue previously.

However I read back a bit and noticed @nldroid updated [his version](https://github.com/nldroid/DsmrP1CustomSensor) yesterday , which works fine after fixing the include name (dsmr_p1_uart.h -> dsmr_p1_sensor.h) in the .yaml. Thanks!

JoeyPoortman, 2020-02-22:
@bramvdpol How did you wire it up?? I used a 4 wire rj11 connector and put a 1kOhm resistor between rts and vin (wich gives ~5v). And the only thing i'm getting in espHome is : Reading Ready

bramvdpol, 2020-02-22:
You probably need to invert the serial signal coming from the meter, depending on your specific model. There are a few where this is not necesary but for most meters it is. Mine is a XS210 ESMR5 meter which needs inverting. I searched a bit but couldn't find the exact schematic I used, but it is something similar to http://compuron.nl/IoT/meter/index.html. I also needed to add a 1K pull up resistor between data and 5V but this may not be needed for your meter.

Good luck

neographikal, 2020-02-23:
Would like this as a standard component to discard my own custom solution :)

nldroid, 2020-02-23:
It's not that hard to use the custom component. See https://github.com/nldroid/CustomP1UartComponent and open read_p1.yaml

JoeyPoortman, 2020-02-23:
@bramvdpol Thanks for the quick reply!

I used this script: https://github.com/nldroid/CustomP1UartComponent
And used the hardware inverter you linked. Wired it up but no bueno. I must be doing something wrong with the hardware inverter or the wires from the cable i guess!

D5 is hooked up to pin 2 without a resistor.
D2 is hooked up to the collector pin on the bc547
Pin 5 is hooked up to the base on the bc547
Pin 3 is connected to ground aswell as the emitter of the bc547

This is exactly like the picture with the 74c04 but i'm not using that ofcourse so i guess that my issue is somewhere there right?

nldroid, 2020-02-23:
I had problems with transistors too (probably my fault) so i got the hardware inverter. Worked immediatly.

Here's more info about other types of inverters:

http://domoticx.com/p1-poort-slimme-meter-uitlezen-hardware/

JoeyPoortman, 2020-02-23:
@nldroid Probably going to order a module and pick this up when i get it.. Its such a waste of time. 3 different esphome setups i've found all use different pinouts or hardware and none have some sort of plan to show you what you should do. This one for example:
https://github.com/rspaargaren/DSMR_ESPHOME Utterly useless without any form of instructions :(

rspaargaren, 2020-02-23:
@JoeyPoortman A Readme was on the to do list.... but if you google a bit on the DSMR meters it is quite obvious on the pinout side. Most of the meters require an inverter for the communication. And the read out pin can either be contineously high or connected to and IO on the ESP.

Mynasru, 2020-02-24:
> I created a custom component but didn't test it (because i don't have a cable yet)
> 
> https://github.com/nldroid/DsmrP1CustomSensor

It works for me with a Wemos D1 mini and a simple transistor inverter. I was struggling to get it to work at first. However, I remembered that there was a message request signal, and I did not configure that pin (it is set as D5 in the .h file). So my mistake. I will let it run for a while and check how it is reporting.

JoeyPoortman, 2020-02-25:
@rspaargaren I would like to help make a readme, maybe with pictures?. You say its easy to google, but i found 3 different wirings so its not "that easy"!! 

So if i can make your coding work (wich looks really clean !) with a BC547 transistor i would be very happy!

nldroid, 2020-02-25:
The 'problem' is that there is more than one way to Rome. There's no 'best' way. You can use the one that suits you best, For me it was using a hardware inverter.

Ik you look on this page: 

http://domoticx.com/p1-poort-slimme-meter-uitlezen-hardware/

You'll find an example with BC547 inverter.

JoeyPoortman, 2020-02-25:
@nldroid Thanks you again! That one uses a BC547 indeed, but it also doesn't use a trigger like mentioned above. So i should in that diagram wire 5v from the arduino from D5 into the BC547 and that should be it right? (that didn't work earlier but i'll try on a breadboard again! )

JoeyPoortman, 2020-02-25:
Still doesn't work after spending another hour on it. I do have a Landis+Gyr E360 meter. Maybe its different from the widely used E350?

rspaargaren, 2020-02-26:
Still doesn't work after spending another hour on it. I do have a Landis+Gyr E360 meter. Maybe its different from the widely used E350?

If you look on the following page:
http://domoticx.com/p1-poort-slimme-meter-hardware/
You can see that the Gyr E360 has an EMSR5.0 and the Gyr E350 the DSMR4.0. That might make a difference.

rspaargaren, 2020-02-26:
> 
> 
> @rspaargaren I would like to help make a readme, maybe with pictures?. You say its easy to google, but i found 3 different wirings so its not "that easy"!!
> 
> So if i can make your coding work (wich looks really clean !) with a BC547 transistor i would be very happy!

Hi I worked on the code together with NLdroid. His code is quite similar to mine and has a better readme. 
I have used the following scheme to read the values:
[image]

JoeyPoortman, 2020-04-11:
I caved and bought a board from Esp8266thingies. Worked right out of the "box". Now only thing is to link it to influxdb to store the data!

I uploaded the same sketch to my old setup and it also worked right away so idk what i did wrong but something wasn't right software wise!

dynasticorpheus, 2020-04-14:
@JoeyPoortman Are you now using ESPHOME on the board purchased from Esp8266thingies? Would like to move as well hence wondering if you have a working setup or I should simply stick to the software it was supplied with.

JoeyPoortman, 2020-04-14:
@dynasticorpheus I am using the default software the board was delivered with. But i can't get it to work with HA, there is just nothing published. So i'm tempted to just flash esphome on it! The seller has a full mailbox so i can't ask for help anymore :(

GraafG, 2020-04-14:
FYI I have some esp8266thingies boards for sale on: https://tweakers.net/aanbod/2306796/p1-shield-voor-wemos.html

They are tested and working, and would be willing to sponser to devs working to get this working in ESPhome.

rspaargaren, 2020-04-14:
> 
> 
> FYI I have some esp8266thingies boards for sale on: https://tweakers.net/aanbod/2306796/p1-shield-voor-wemos.html
> 
> They are tested and working, and would be willing to sponser to devs working to get this working in ESPhome.

Nice piece of hardware! Would not be to hard to adapt the scheme for esphome. However our libraries are working on custom library basis

JoeyPoortman, 2020-04-14:
@GraafG Do you use the same software as esp8266thingies delivers them with??

I think if you just drop the code from above onto it, it works straight away. I don't have the original firmware that is on the board so if things go sideways i can't reflash it, that is my only problem atm.. 

[image]

It works like a charm!

dynasticorpheus, 2020-04-14:
Latest binary can be downloaded here http://www.esp8266thingies.nl/wp/wp-content/uploads/2019/05/ESPEasyDSMR5v4.bin_.zip

Source: https://github.com/romix123/ESPEasyDSMR5

dynasticorpheus, 2020-04-14:
@JoeyPoortman  I actually found a surprisingly stable solution to get the data fed into my own home assist setup however not necessarily very straightforward :)

enable remote logging --> syslog-ng outputting json formatted log file --> read by hass file sensor

JoeyPoortman, 2020-04-15:
@dynasticorpheus  Where do you see the option Enable remote logging? I can't find it anywhere. 

I'll try to load esphome on it today, now that i have the original files! thanks for that!

JoeyPoortman, 2020-04-15:
@dynasticorpheus I can't find the option "enable remote  logging". 

I did just install esphome on it, it works after changing the pin number for reading. It is really slow and goes offline for a second to update. But atleast i have something in HA right away!

Romix said in a email that the tutorial to hook the board up to HA is on his website, i can't find any details as to what you have to setup on the device itself! There are no mqtt messages from the device with any setting (it does however connect to the mqtt broker running in HA). 

I would like to have the speed of the espeasydsmr5v4 output directly into influxdb, then i can setup anything i want to display.

[image]
The small gaps are the time the device is offline.

dynasticorpheus, 2020-04-15:
@dynasticorpheus  You need to set below options which you can find via Tools - Advanced:

Syslog IP: 192.168.x.x  (= home assistant instance with syslog-ng running) 
Syslog Level: 2
UDP port: 514

JoeyPoortman, 2020-04-15:
@dynasticorpheus Thanks you for the example. I'm running right into the next wall, Syslog-ng isn't a package i can easily install. Is it best to install it using Portainer you think?

ESPmonster, 2021-01-28:
@JoeyPoortman  I also have/had a lot of trouble getting the ESP8266thingies board to work with HA. The method on Romix' site works (http://www.esp8266thingies.nl/wp/p1-wifi-gateway-parasitic/) --> i.e., setup (generic) HTTP as controller in ESPeasy and add DSMR, IP and port in configuration.yaml (and meter_readings as a group). (Note that I could only find the DSMR integration after looking for 'ignored integrations'!) However, in my case this method only works for ~20 min until the data flow stops and then I have to restart HA (without errors in DSMR debugging log on HA or ESPeasy)! 

On the same website you can find a Node-red script that could connect to the port and publish the data as separate MQTT topics but I fail to establish the TCP connection. 

For ESPhome, did you change the reading pin to RX or D6? 

@dynasticorpheus  I might also try your solution but I'm running hassio so Syslog-ng doesn't seem an easy thing to do.

JoeyPoortman, 2021-01-28:
I gave up on the software side from Esp8266thingies, its unstable and clunky (sorry). 

I went with the DSMR Reader solution, it has to run in its own docker container outside of HA. But its buttery smooth, its been running since March without any glitches! Pm me if you want to get some details in Discord :)

ESPmonster, 2021-01-28:
@JoeyPoortman Yes please! My Discord is ESPmonster#7102

dynasticorpheus, 2021-01-28:
@ESPmonster  I actually moved away from Esp8266thingies software in full and flashed below esphome based solution which is rock stable and creates required sensors in hass by its own without the need for anything else. 

https://github.com/tbrasser/CustomP1UartComponent

The only thing that I needed to change in read_p1.yaml next to the WIFI AP credentials was GPIO21 into GPIO3:

```
uart:
  - rx_pin: GPIO3
    baud_rate: 115200
    id: uart_bus
```

ESPmonster, 2021-01-28:
@dynasticorpheus Excellent! I'll try this firmware this week.

In order to make this findable for other I'll add some tags: ESP8266thingies Wemos D1 mini lite home assistant DSMR sensor P1 gateway reader MQTT kwh

ESPmonster, 2021-01-28:
@dynasticorpheus I installed the P1UARTcomponent firmware. Works OK. For the esp8266thingies board (on a Wemos D1 mini lite) I also had to change the UART pin to:
```
logger: 
  level: DEBUG
  hardware_uart: UART1
```
What are your settings in dsmr_p1_sensor.h/dsmr_p1_uart.h? 
I have very frequent wifi disconnections (~3x/5 min) even though signal strength is optimal and I never had this problem with this or any other ESP devices. My current settings are:
```
#define P1_MAXTELEGRAMLENGTH 1700
#define DELAY_MS 29000 // Delay in miliseconds before reading another telegram
#define WAIT_FOR_DATA_MS 20000
```

dynasticorpheus, 2021-01-30:
@ESPmonster Forget to mention I also changed platform and board settings, perhaps this is the culprit? I pushed all my changes to below fork:

https://github.com/dynasticorpheus/CustomP1UartComponent/commit/e790cf353cf6dc3ed76ddbb2875ce9923360ab45

rspaargaren, 2021-01-30:
Hi I am currently working on a dsmr component for esphome. So we do not have to use the custom code. I will release a beta version soon for testing

rspaargaren, 2021-01-30:
There are some delays in the custom component which are blocking the progress in the code. This is disturbing the progress in overall code and WiFi reception can drop. I have the same issue with the custom code.

dynasticorpheus, 2021-01-30:
@ESPmonster @rspaargaren  Perhaps because I reduced update frequency from 60 to 3 seconds this is not a problem on my end. Anyway good to know there will be official component soon, great news.

ESPmonster, 2021-02-01:
@dynasticorpheus Thanks for the info and forking! I also adjusted board settings and further played with the setting to no avail. In fact, my other ESPhome devices are suddenly also affected. Turning off the P1 wemos only helps a little so perhaps it has something to do with my router.
@rspaargaren looking forward to your beta, thanks

closed 2021-08-10 by OttoWinter