esphome/feature-requests #8: Add TX power option in BLE iBeacon

uryupinsk, 2019-01-27:
Hi, I'm currently experimenting with iBeacons and room detection in Owntracks.

I want to be able to do presence detection at room level, but as I live in a small flat, the BLE Tx power is too high and all beacons are detected at all time.

It would be nice if you could add a TX power option to your iBeacon sketch to be able to powering them down a little.

Best regards.

duiffie, 2020-04-18:
This would be a nice enhancement indeed! Any update on it?

sja, 2021-09-28:
I had a look in the code but not familar enough with C++ to implmement that.
As of IDF, you can use this, e.g. [after this line of esp32_ble_beacon](https://github.com/esphome/esphome/blob/dev/esphome/components/esp32_ble_beacon/esp32_ble_beacon.cpp#L77):

        /**
        * @brief  Set BLE TX power
        *         Connection Tx power should only be set after connection created.
        * @param  power_type : The type of which tx power, could set Advertising/Connection/Default and etc
        * @param  power_level: Power level(index) corresponding to absolute value(dbm)
        * @return              ESP_OK - success, other - failed
        */
        esp_err_t esp_ble_tx_power_set(esp_ble_power_type_t power_type, esp_power_level_t power_level)