esphome/feature-requests #14: max7219 dot matrix

sandro-defender, 2019-02-01:
**Is your feature request related to a problem/use-case? Please describe.**

add max7219 dot matrix support.

**Describe the solution you'd like:**

home assistan publishing "events,massages.. on max7219 dot matrix .

**Additional context:**

sending text to display on dot matrix.

ASMfreaK, 2019-05-10:
So, I've created a custom component for this, it wasn't a hard hack. I've created a [custom repo](https://github.com/ASMfreaK/esphome_max7219) for this.
@OttoWinter  Is there a generic class for inbuffer drawing?

OttoWinter, 2019-05-10:
@ASMfreaK Yes, see `DisplayBuffer`

Xeroxxx, 2019-05-19:
Is this going to be pulled to main repo?

sandro-defender, 2019-05-20:
Hope to see it soon in main repo...

cvanaalst, 2019-08-05:
Hope to see it soon in the main repo - is there a simple way to use the custom repo from ASMfreaK ?

ASMfreaK, 2019-08-13:
@everyone
Do anyone know a good (preferably open-source) pixel font to use with such display? I'll need it for testing and examples in docs.

thundergreen, 2019-08-13:
any news here mister dev? :)

T-REX-XP, 2019-08-14:
> to everyone
> Do anyone know a good (preferably open-source) pixel font to use with such display? I'll need it for testing and examples in docs.
> I'd also like if the font be fully unicode compatible.

here is https://www.wfonts.com/font/pixel-unicode

codegrau, 2019-09-16:
Looking forward to any news regarding MAX7219 Dot Matrix support <3

saikek, 2019-10-06:
Any news here ? What is missing to make a merge ?

minsuke, 2019-11-25:
[image]
Can I use this one? 

Thx for your custom codes

T-REX-XP, 2019-11-25:
> [image]
> Can I use this one?
> 
> Thx for your custom codes

Sure, but not with this firmware)) esp home still not support this type of display

minsuke, 2019-11-25:
> > [image]
> > Can I use this one?
> > Thx for your custom codes
> 
> Sure, but not with this firmware)) esp home still not support this type of display

thx for your quick reply : ) I think I wait for some time..

jisala, 2020-01-03:
Im really looks forward for ESPhome comparability

angeltonio, 2020-01-11:
Nothing about the support on ESPhome to this dot matrix??...

Dreamoffice, 2020-04-14:
any news about the MAX7219 support?

widapro, 2020-04-14:
still waiting for MAX7219 ESPHome support...  meanwhile waiting, I wrote this library for MAX7219 which works with MQTT protocol and can be integrated with smart home systems.
Maybe it can help anybody https://github.com/widapro/pixel_led_mqtt_panel

For example, [Home-assistant can publish events. message or sensors value to MAX7219 ](https://github.com/widapro/pixel_led_mqtt_panel#then-configure-automation-in-home-asisstant)

bazorante14, 2020-04-16:
very much looking forward to it.

ceaswaran, 2020-04-19:
Yes looking forward for this, any update on when this will be added?

rspaargaren, 2020-04-27:
I am working on the direct max7219 digit integration within ESPHOME. Any testers here. At this moment I can display a single digit on each section but with a bit of flickering. So I need a bit of fine tuning.

Beltshassar, 2020-04-27:
I have the hardware, but where should I drop the files for esphome to notice?

rspaargaren, 2020-04-27:
https://github.com/rspaargaren/esphome/tree/NextionUpdate/esphome/components/max7219digit these files need to be added to the esphome files. Basically complete directory.

Beltshassar, 2020-04-27:
@rspaargaren 
Great - I have cloned the branch and linked the max7219digit folder into the components folder of my installation.

There's a few warnings concerning max7219digit.cpp during compiling and all pixels in the display is on after flashing.

I use a Wemos D1 mini (probably a clone)

Virtualenv:
  Python 3.7.7
  ESPHome 1.14.3

```
INFO Reading configuration /var/db/esphome/demo1.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /var/db/esphome/disco
Processing disco (board: d1_mini; framework: arduino; platform: espressif8266@2.2.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.6
|   |-- <ESPAsyncTCP-esphome> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
|-- <SPI> 1.0
Compiling .pioenvs/disco/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/disco/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/disco/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/disco/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/disco/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/disco/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/disco/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/disco/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/disco/src/esphome/components/api/util.cpp.o
Compiling .pioenvs/disco/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling .pioenvs/disco/src/esphome/components/display/display_buffer.cpp.o
Compiling .pioenvs/disco/src/esphome/components/gpio/switch/gpio_switch.cpp.o
Compiling .pioenvs/disco/src/esphome/components/logger/logger.cpp.o
Compiling .pioenvs/disco/src/esphome/components/max7219digit/max7219digit.cpp.o
Compiling .pioenvs/disco/src/esphome/components/ota/ota_component.cpp.o
Compiling .pioenvs/disco/src/esphome/components/spi/spi.cpp.o
Compiling .pioenvs/disco/src/esphome/components/switch/automation.cpp.o
src/esphome/components/max7219digit/max7219digit.cpp: In member function 'uint8_t esphome::max7219digit::MAX7219Component::print(uint8_t, const char*)':
src/esphome/components/max7219digit/max7219digit.cpp:119:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }  // end of sendString
 ^
Compiling .pioenvs/disco/src/esphome/components/switch/switch.cpp.o
Compiling .pioenvs/disco/src/esphome/components/web_server_base/web_server_base.cpp.o
Compiling .pioenvs/disco/src/esphome/components/wifi/wifi_component.cpp.o
Compiling .pioenvs/disco/src/esphome/components/wifi/wifi_component_esp32.cpp.o
Compiling .pioenvs/disco/src/esphome/components/wifi/wifi_component_esp8266.cpp.o
Compiling .pioenvs/disco/src/esphome/core/application.cpp.o
Compiling .pioenvs/disco/src/esphome/core/component.cpp.o
Compiling .pioenvs/disco/src/esphome/core/controller.cpp.o
Compiling .pioenvs/disco/src/esphome/core/esphal.cpp.o
Compiling .pioenvs/disco/src/esphome/core/helpers.cpp.o
Compiling .pioenvs/disco/src/esphome/core/log.cpp.o
Compiling .pioenvs/disco/src/esphome/core/preferences.cpp.o
Compiling .pioenvs/disco/src/esphome/core/scheduler.cpp.o
Compiling .pioenvs/disco/src/esphome/core/util.cpp.o
Compiling .pioenvs/disco/src/main.cpp.o
Generating LD script .pioenvs/disco/ld/local.eagle.app.v6.common.ld
[... 111 lines cut ...]
Compiling .pioenvs/disco/FrameworkArduino/sqrt32.cpp.o
Compiling .pioenvs/disco/FrameworkArduino/time.cpp.o
Compiling .pioenvs/disco/FrameworkArduino/uart.cpp.o
Compiling .pioenvs/disco/FrameworkArduino/umm_malloc/umm_malloc.cpp.o
Archiving .pioenvs/disco/libFrameworkArduino.a
Indexing .pioenvs/disco/libFrameworkArduino.a
Linking .pioenvs/disco/firmware.elf
Retrieving maximum program size .pioenvs/disco/firmware.elf
Checking size .pioenvs/disco/firmware.elf
Building .pioenvs/disco/firmware.bin
DATA:    [====      ]  44.4% (used 36388 bytes from 81920 bytes)
PROGRAM: [====      ]  39.0% (used 407240 bytes from 1044464 bytes)
Creating BIN file ".pioenvs/disco/firmware.bin" using ".pioenvs/disco/firmware.elf"
========================= [SUCCESS] Took 33.70 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of disco.local
INFO  -> 192.168.1.149
INFO Uploading /var/db/esphome/disco/.pioenvs/disco/firmware.bin (411392 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from disco.local using esphome API
```

rspaargaren, 2020-04-27:
Yes the "warning" is known and I will clean that up. First time all my pixels where green as well but after complete reboot. (disconnect from power) it worked. Although blinking but I am working on that. I use a wemos d1 mini clone as well.

rspaargaren, 2020-04-27:
I did a bit of furhter update on the code and proof of concept I can at least show the numbers. Now need to have a view on the update routine.
[image]

Beltshassar, 2020-04-28:
Ah well a reboot makes all the difference 
[image]

ceaswaran, 2020-04-28:
> @Beltshassar https://github.com/rspaargaren/esphome/tree/NextionUpdate/esphome/components/max7219digit these files need to be added to the esphome files. Basically complete directory. and to the configurationfile you can add: # Example configuration entry
> 
> spi:
> clk_pin: D0
> mosi_pin: D1
> 
> display:
> 
> * platform: max7219digit
>   cs_pin: D2
>   num_chips: 4
>   lambda: |-
>   it.print("0123");

**Wr exactly these files go? I mean which folder? I don't see any components folder mentioned above either under config folder or under esphome folder. Please help**

Beltshassar, 2020-04-28:
It probably depends on your OS and how your installation was made.
I'm running espHome as a Freenas-plugin, which in reality is a installation inside a FreeBSD jail.
My components path was here:
/usr/local/srv/esphome/lib64/python3.7/site-packages/esphome/components/

cvanaalst, 2020-04-28:
I am running HASSIO with the ESPHOME-addon; I am not a docker 
specialist, but found that the /opt/esphome/esphome/components/ contains 
all the component-packages .. I have the Portainer addon and that's how 
I found it. But I have no idea how to add the max7219digit files to that 
docker image ...

rspaargaren, 2020-04-28:
@cvanaalst I do not have an updated docker image of my build and it is work in progress. The easiest way at the moment is to map a directory for the docker container.

rspaargaren, 2020-04-29:
Got it in scrolling mode now but cannot post a video on this channel...

Beltshassar, 2020-04-29:
Nice, do I need any additional configuration for scrolling?
I've pulled the latest changes and the text is nice, no more flickering. But theres no scrolling even though I've added extra text.

rspaargaren, 2020-04-29:
No I will create an instruction on that. It needs a bit of finetune to control the loop! I will leave a message when ready

RoadkillUK, 2020-04-30:
Hi all, I was just looking to see if it was possible to use these 8x8 matrices, then I come along this thread, cool.  Anyway, I'm on HASSIO here and I have your component working as far as yourself.

I'm just here to say that @cvanaalst, the simplest way to add components for testing is to create a ``` custom_components ``` folder in your ``` /config/esphome/ ``` folder

example:
``` config/esphome/MAX7219digit/ ```

> **Note
Additionally, ESPHome has a custom_components mechanism like Home Assistant does. So for testing you can also create a new custom_components folder inside of your ESPHome config folder and create new integrations in there.**

Hope that helps someone out.

cvanaalst, 2020-04-30:
@RoadkillUK thx I’ll give that a try !
Chris

Verstuurd vanaf mijn iPhone

ceaswaran, 2020-04-30:
> * max7219digit

i get the following error while compiling

```Failed config

display.max7219digit: [source /config/esphome/max7219digit.yaml:27]
  
  Platform not found: 'display.max7219digit'.
  platform: max7219digit
  cs_pin: D2
  num_chips: 4
  lambda: it.print("01234567");```

RoadkillUK, 2020-04-30:
Blx. @cvanaalst I got it wrong, just noticed !!  The description is correct, the example is wrong !!

``` config/esphome/custom_components/MAX7219digit/ ```

I've edited the original too.

ceaswaran, 2020-04-30:
> Blx. @cvanaalst I got it wrong, just noticed !! The description is correct, the example is wrong !!
> 
> `config/esphome/custom_components/MAX7219digit/`
> 
> I've edited the original too.

Thanks 

but i see more errors while compiling
please help
```INFO Reading configuration /config/esphome/max7219digit.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/max7219digit
Processing max7219digit (board: nodemcuv2; framework: arduino; platform: espressif8266@2.2.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.6
|   |-- <ESPAsyncTCP-esphome> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
|-- <SPI> 1.0
Compiling /data/max7219digit/.pioenvs/max7219digit/src/main.cpp.o
Compiling /data/max7219digit/.pioenvs/max7219digit/src/max7219digit.cpp.o
Generating LD script /data/max7219digit/.pioenvs/max7219digit/ld/local.eagle.app.v6.common.ld
Compiling /data/max7219digit/.pioenvs/max7219digit/lib4d9/ESP8266WiFi/BearSSLHelpers.cpp.o
Compiling /data/max7219digit/.pioenvs/max7219digit/lib4d9/ESP8266WiFi/CertStoreBearSSL.cpp.o
In file included from src/main.cpp:18:0:
src/max7219digit.h:19:7: error: redefinition of 'class esphome::max7219digit::MAX7219Component'
 class MAX7219Component : public PollingComponent,
       ^
In file included from src/esphome.h:17:0,
                 from src/main.cpp:3:
src/esphome/components/max7219digit/max7219digit.h:19:7: error: previous definition of 'class esphome::max7219digit::MAX7219Component'
 class MAX7219Component : public PollingComponent,
       ^
In file included from /root/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include/sys/ctype.h:7:0,
                 from /root/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include/ctype.h:5,
                 from /root/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/cctype:42,
                 from /root/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/localefwd.h:42,
                 from /root/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/string:43,
                 from src/esphome/core/component.h:3,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome.h:2,
                 from src/main.cpp:3:
src/max7219font.h:3:46: error: redefinition of 'const byte MAX7219_Dot_Matrix_font [256][8]'
 const byte MAX7219_Dot_Matrix_font [256] [8] PROGMEM = {
                                              ^
In file included from src/esphome.h:18:0,
                 from src/main.cpp:3:
src/esphome/components/max7219digit/max7219font.h:3:12: error: 'const byte MAX7219_Dot_Matrix_font [256][8]' previously defined here
 const byte MAX7219_Dot_Matrix_font [256] [8] PROGMEM = {
            ^
Compiling /data/max7219digit/.pioenvs/max7219digit/lib4d9/ESP8266WiFi/ESP8266WiFi.cpp.o
*** [/data/max7219digit/.pioenvs/max7219digit/src/main.cpp.o] Error 1
========================== [FAILED] Took 2.51 seconds ==========================```

RoadkillUK, 2020-04-30:
That my friend is WAY beyond me, I'm uploading mine to an ESP32, don't know if that makes a difference.

rspaargaren, 2020-04-30:
Hi all,

The location of the files should be in:
esphome/esphome/components/max7219digit/

Not in CUSTOM components!

RoadkillUK, 2020-04-30:
[image]

This works for Hassio

[image]

rspaargaren, 2020-04-30:
Ok did not know that, but you can install the files in the directory as indicated by me as well. Or at least that is where I am having them during the development phase.
So now I still have some features I am thinking of implementation:
- Scrolling (for larger text display) or cool effect from left to right or visa versa -> Almost done
- ICONS -> basically already possible as long as we add them to the font table.
- Blinking dots or double dots for example time display
- More custom pixel control
- Smaller font(s) of for example 3 dots height and variable width to show two lines

Maybe you have some more ideas....?

rspaargaren, 2020-04-30:
@ceaswaran  Are you having a configuration for the original MAX7219 in your config as well?

ceaswaran, 2020-04-30:
> @ceaswaran Are you having a configuration for the original MAX7219 in your config as well?

nope @rspaargaren below is my esphome .yaml

```
esphome:
  name: max7219digit
  platform: ESP8266
  board: nodemcuv2

  includes:
    - custom_components/max7219digit/max7219digit.cpp
    - custom_components/max7219digit/max7219digit.h
    - custom_components/max7219digit/max7219font.h 
spi:
  clk_pin: D0
  mosi_pin: D1

display:
  - platform: max7219digit
    cs_pin: D2
    num_chips: 4
    lambda: |-
      it.print("01234567");```

RoadkillUK, 2020-04-30:
@rspaargaren  You're doing a great job, if I think of anything I'll let you know.

RoadkillUK, 2020-04-30:
@ceaswaran 

This is mine, I've trimmed off the crap (I had other things on it too).

```
esphome:
  name: clock
  platform: ESP32
  board: nodemcu-32s

wifi:

  networks: 
    - ssid: !secret wifi
      password: !secret wifi_password
    - ssid: !secret wifi2
      password: !secret wifi_password2

  manual_ip:
    static_ip: 192.168.0.238
    gateway: 192.168.0.1
    subnet: 255.255.255.0
    dns1: 192.168.0.1
    dns2: 8.8.8.8

logger:

api:
  password: !secret api_password

ota:
  password: !secret ota_password

spi:
  clk_pin: GPIO16
  mosi_pin: GPIO17

display:
  - platform: max7219digit
    cs_pin: GPIO18
    id: my_matrix
    num_chips: 4
    lambda: |-
      it.print("MILK");

```

rspaargaren, 2020-04-30:
@ceaswaran 
indeed remove the includes part:
    - custom_components/max7219digit/max7219digit.cpp
    - custom_components/max7219digit/max7219digit.h
    - custom_components/max7219digit/max7219font.h
esphome is locating these file during compiling by default. either in component (or in custom-component ;-) directory)

ceaswaran, 2020-04-30:
I see the following error after removing includes & compiling

```Archiving /data/max7219digit/.pioenvs/max7219digit/libFrameworkArduino.a
Linking /data/max7219digit/.pioenvs/max7219digit/firmware.elf
/data/max7219digit/.pioenvs/max7219digit/src/max7219digit.cpp.o: In function `esphome::max7219digit::MAX7219Component::get_setup_priority() const':
max7219digit.cpp:(.text._ZNK7esphome12max7219digit16MAX7219Component18get_setup_priorityEv+0x4): multiple definition of `esphome::max7219digit::MAX7219Component::get_setup_priority() const'
/data/max7219digit/.pioenvs/max7219digit/src/esphome/components/max7219digit/max7219digit.cpp.o:max7219digit.cpp:(.text._ZNK7esphome12max7219digit16MAX7219Component18get_setup_priorityEv+0x4): first defined here
/data/max7219digit/.pioenvs/max7219digit/src/max7219digit.cpp.o: In function `esphome::max7219digit::MAX7219Component::dump_config()':
max7219digit.cpp:(.text._ZN7esphome12max7219digit16MAX7219Component11dump_configEv+0x64): multiple definition of `esphome::max7219digit::MAX7219Component::dump_config()'
/data/max7219digit/.pioenvs/max7219digit/src/esphome/components/max7219digit/max7219digit.cpp.o:max7219digit.cpp:(.text._ZN7esphome12max7219digit16MAX7219Component11dump_configEv+0x64): first defined here
/data/max7219digit/.pioenvs/max7219digit/src/max7219digit.cpp.o: In function `esphome::max7219digit::MAX7219Component::update()':
max7219digit.cpp:(.text._ZN7esphome12max7219digit16MAX7219Component6updateEv+0x10): multiple definition of `esphome::max7219digit::MAX7219Component::update()'
/data/max7219digit/.pioenvs/max7219digit/src/esphome/components/max7219digit/max7219digit.cpp.o:max7219digit.cpp:(.text._ZN7esphome12max7219digit16MAX7219Component6updateEv+0x10): first defined here
/data/max7219digit/.pioenvs/max7219digit/src/max7219digit.cpp.o: In function `esphome::max7219digit::MAX7219Component::set_intensity(unsigned char)':
max7219digit.cpp:(.text._ZN7esphome12max7219digit16MAX7219Component13set_intensityEh+0x0): multiple definition of `esphome::max7219digit::MAX7219Component::set_intensity(unsigned char)'
/data/max7219digit/.pioenvs/max7219digit/src/esphome/components/max7219digit/max7219digit.cpp.o:max7219digit.cpp:(.text._ZN7esphome12max7219digit16MAX7219Component13set_intensityEh+0x0): first defined here
/data/max7219digit/.pioenvs/max7219digit/src/max7219digit.cpp.o: In function `esphome::max7219digit::MAX7219Component::set_num_chips(unsigned char)':
max7219digit.cpp:(.text._ZN7esphome12max7219digit16MAX7219Component13set_num_chipsEh+0x0): multiple definition of `esphome::max7219digit::MAX7219Component::set_num_chips(unsigned char)'
/data/max7219digit/.pioenvs/max7219digit/src/esphome/components/max7219digit/max7219digit.cpp.o:max7219digit.cpp:(.text._ZN7esphome12max7219digit16MAX7219Component13set_num_chipsEh+0x0): first defined here
/data/max7219digit/.pioenvs/max7219digit/src/max7219digit.cpp.o: In function `esphome::max7219digit::MAX7219Component::send_byte_(unsigned char, unsigned char)':
max7219digit.cpp:(.text._ZN7esphome12max7219digit16MAX7219Component10send_byte_Ehh+0x8): multiple definition of `esphome::max7219digit::MAX7219Component::send_byte_(unsigned char, unsigned char)'
/data/max7219digit/.pioenvs/max7219digit/src/esphome/components/max7219digit/max7219digit.cpp.o:max7219digit.cpp:(.text._ZN7esphome12max7219digit16MAX7219Component10send_byte_Ehh+0x8): first defined here
/data/max7219digit/.pioenvs/max7219digit/src/max7219digit.cpp.o: In function `esphome::max7219digit::MAX7219Component::display()':
max7219digit.cpp:(.text._ZN7esphome12max7219digit16MAX7219Component7displayEv+0x18): multiple definition of `esphome::max7219digit::MAX7219Component::display()'
/data/max7219digit/.pioenvs/max7219digit/src/esphome/components/max7219digit/max7219digit.cpp.o:max7219digit.cpp:(.text._ZN7esphome12max7219digit16MAX7219Component7displayEv+0x18): first defined here
/data/max7219digit/.pioenvs/max7219digit/src/max7219digit.cpp.o: In function `esphome::max7219digit::MAX7219Component::send_to_all_(unsigned char, unsigned char)':
max7219digit.cpp:(.text._ZN7esphome12max7219digit16MAX7219Component12send_to_all_Ehh+0xc): multiple definition of `esphome::max7219digit::MAX7219Component::send_to_all_(unsigned char, unsigned char)'
/data/max7219digit/.pioenvs/max7219digit/src/esphome/components/max7219digit/max7219digit.cpp.o:max7219digit.cpp:(.text._ZN7esphome12max7219digit16MAX7219Component12send_to_all_Ehh+0xc): first defined here
/data/max7219digit/.pioenvs/max7219digit/src/max7219digit.cpp.o: In function `esphome::max7219digit::MAX7219Component::setup()':
max7219digit.cpp:(.text._ZN7esphome12max7219digit16MAX7219Component5setupEv+0x30): multiple definition of `esphome::max7219digit::MAX7219Component::setup()'
/data/max7219digit/.pioenvs/max7219digit/src/esphome/components/max7219digit/max7219digit.cpp.o:max7219digit.cpp:(.text._ZN7esphome12max7219digit16MAX7219Component5setupEv+0x30): first defined here
/data/max7219digit/.pioenvs/max7219digit/src/max7219digit.cpp.o: In function `esphome::max7219digit::MAX7219Component::send64pixels(unsigned char, unsigned char const*)':
max7219digit.cpp:(.text._ZN7esphome12max7219digit16MAX7219Component12send64pixelsEhPKh+0x14): multiple definition of `esphome::max7219digit::MAX7219Component::send64pixels(unsigned char, unsigned char const*)'
[... cut ...]
max7219digit.cpp:(.text._ZN7esphome12max7219digit16MAX7219Component10set_writerEOSt8functionIFvRS1_EE+0x10): multiple definition of `esphome::max7219digit::MAX7219Component::set_writer(std::function<void (esphome::max7219digit::MAX7219Component&)>&&)'
/data/max7219digit/.pioenvs/max7219digit/src/esphome/components/max7219digit/max7219digit.cpp.o:max7219digit.cpp:(.text._ZN7esphome12max7219digit16MAX7219Component10set_writerEOSt8functionIFvRS1_EE+0x10): first defined here
collect2: error: ld returned 1 exit status
*** [/data/max7219digit/.pioenvs/max7219digit/firmware.elf] Error 1
========================= [FAILED] Took 14.28 seconds =========================```

RoadkillUK, 2020-04-30:
Well, @ceaswaran.  I've just set up a ESP8266, uploaded your code and it's absolutely fine, sorry I can't help any further.

[image]

rspaargaren, 2020-04-30:
@ceaswaran I think I know why compiling is not working and it requires some change in code from my side. The solution as far as I know now is to clean up your compiling directory.

Beltshassar, 2020-04-30:
> Maybe you have some more ideas....?

Are you using any libraries? I came across these in another esp/max project.

https://github.com/MajicDesigns/MD_Parola
https://github.com/MajicDesigns/MD_MAX72XX

Maybe you can use some of it, at least there's some inspiration 😉

cvanaalst, 2020-04-30:
these all work:

         lambda: |-
           it.print("HELLO");

         lambda: |-
           it.strftime("%H:%M", id(hatime).now());

         lambda: |-
           it.printf("%.0f", id(wifi_sig).state);

but font is just to big to use it as a clock eg '19:23'; last char '3' 
is not displayed ...

other option I tried is the pages, but that does not seem to work - 
would be nice to alternate messages; text -> time -> temp -> ....

my apologies for spamming you with all my ideas, but I really like this 
component and I use it a lot (but have to code it in arduino ide)

cvanaalst

ceaswaran, 2020-04-30:
> Well, @ceaswaran. I've just set up a ESP8266, uploaded your code and it's absolutely fine, sorry I can't help any further.
> 
> [image]+
i deletd all & re created it works thanks

RoadkillUK, 2020-04-30:
> > Maybe you have some more ideas....?
> 
> Are you using any libraries? I came across these in another esp/max project.
> 
> https://github.com/MajicDesigns/MD_Parola
> https://github.com/MajicDesigns/MD_MAX72XX
> 
> Maybe you can use some of it, at least there's some inspiration 😉

@rspaargaren 

I agree, I think you should use a more native library.  Let me know if you need any testing done.

ceaswaran, 2020-04-30:
> -- Chris van Aalst cva@skynet.be

**Yes For time font size is too big to accommodate we need to have small font**

rspaargaren, 2020-04-30:
@RoadkillUK @ceaswaran i have used other libraries as inspiration. I think I will “finalize” this library first. By enabling scrolling and a (blinking) dot or double dot. That will not be to hard to do. Then I will push this to the source and work on the extended version. That will require a different approach and indeed I have already reviewed the library as mentioned and I will not reinvent the wheel... just adopt it within this platform ;-) and tweek a bit

Beltshassar, 2020-04-30:
Sounds like a solid plan

cvanaalst, 2020-04-30:
For info this is what I tested; simple alternating Time & Wifi signal ...

We need a font that is 6wide (5 dots for char & 1 dot blank), this is 
what is mostly used, where did you get/created the max7219font.h file ? 
Can I help in creating a 6dot wide version ?

esphome:
   name: max7219
   platform: ESP8266
   board: nodemcuv2

wifi:
   ssid: "xxxxxx"
   password: "xxxxxxxxxx"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

time:
   - platform: homeassistant
     id: hatime

spi:
   clk_pin: D5
   mosi_pin: D7

globals:
   - id: my_global_bool
     type: bool
     restore_value: no
     initial_value: 'true'

sensor:
   - platform: wifi_signal
     name: "WiFi Signal Sensor"
     id: wifi_sig

#this works
#      it.print("Hello");
#      it.printf("%.0f", id(wifi_sig).state);
#      it.strftime("%H:%M", id(hatime).now());
#this does not work
#      it.printf(50, 60, "%.0f", id(wifi_sig).state);

display:
   - platform: max7219digit
     cs_pin: D8
     intensity: 1  #has no effect
     update_interval: 2s
     id: my_matrix
     num_chips: 4
     lambda: |-
       if (id(my_global_bool)) {
         it.strftime("%H:%M", id(hatime).now());
       }
       else {
         it.printf("%.0f", id(wifi_sig).state);
       }

interval:
   - interval: 10s
     then:
       - lambda: |-
           if (id(my_global_bool)) {
             id(my_global_bool) = false;
           }
           else {
             id(my_global_bool) = true;
           }

RoadkillUK, 2020-05-01:
@rspaargaren, anything to test? ;)

rspaargaren, 2020-05-01:
@RoadkillUK hope to show you tomorrow something now I have to solve a crashed pi first

[188 more replies]

closed 2020-06-29 by glmnet