esphome/feature-requests #16: iAQ core sensor support ?

vincegre, 2019-02-01:
What new integration would you wish to have?
Atmospheric volatil component sensors: iAQ Core and family (I2C based)

If possible, provide a link to an existing library for the integration:
https://github.com/hafu/IAQCore

Please describe your use case for this integration and alternatives you've tried:
Very small, pretty cheap and lot more reliable than competing products, very low consumption (perfect for battery/solar project), easy wiring (I2C).

[iAQ-core_DS000334_1-00.pdf](https://github.com/esphome/ESPHome-Core/files/2807097/iAQ-core_DS000334_1-00.pdf)
Datasheet of one of the sensor ;)

Thanks

voicevon, 2019-03-06:
There is another sensor called SGP30 from Adafruit. The features is quite similar to iAQ Core. 

[Datasheet ](https://www.mouser.com/ds/2/682/Sensirion_Gas_Sensors_SGP30_Datasheet_EN-1148053.pdf)

[Arduino Library ](https://github.com/adafruit/Adafruit_SGP30)

vincegre, 2019-03-24:
@OttoWinter any update on integration of these sensors ? thanks

ariedeleen, 2019-06-16:
Bosch Sensortec Environmental Cluster (BSEC) Software is updated to v1.4.7.3 | April 10th, 2019
[](https://www.bosch-sensortec.com/bst/products/all_products/bsec)

In the download zip there is the BSEC (IAQ) library for ESP8266 and ESP32 and other 32, 16 and 8 bit MCU platforms.

Is  it possible to use this as a reference to get the BSEC_OUTPUT_IAQ value 0-500 as mentioned in the BST-BME680-BSEC_1.4.7.3_Generic_Release/integration_guide/Integration-Guide-AN008-47.pdf page 8.
 to implement iAQ?

It would be great to have a card in lovelace that shows the in/outdoor air quality as a colorbar from green to red. Or set an alarm trigger for example.

OttoWinter, 2019-06-17:
@ariedeleen From all I can tell their BSEC library is still closed source, so it cannot be added to esphome because of GPLv3 license.

ruimarinho, 2020-08-02:
Me and @mcspr have been working together during the last month to [add support for BME680 using BSEC](https://github.com/xoseperez/espurna/pull/2295) on [espurna](https://github.com/xoseperez/espurna). The experience is completely seamless, requiring zero modifications on the user side - it just works.

Personally, I am a big fan of _esphome_ as well and use it extensively on my ESP32 projects, so in the future I'd like to bring the same level of support here, time permitting.

I hope it serves as inspiration in case others want to jumpstart that work. Our approach to licensing was:

* To include Bosch's own Arduino library as part of a user-downloaded dependency (the package being BSD-3-Clause licensed), with all copyright notices carried over.
* To require an opt-in flag to enable support for the sensor, mentioning on every document that enabling the flag is confirmation of agreement of the Bosch's license.
* To not distribute a pre-compiled binary with support for this sensor, instead requiring users to proactively seek to enable the correct flag.

This is truly a fantastic sensor which deserved a special attention to detail.

trvrnrth, 2020-09-01:
@ruimarinho If you're interested I have a working ESPHome custom component that I've had running without issue on a couple of Wemos D1 minis for a while now that might save a bit of time getting things up and running. Now that the BSEC Arduino library has a Platform IO library manifest with the necessary incantations to get the library linked in there's no need for manual download of the library or anything (see https://github.com/BoschSensortec/BSEC-Arduino-library/pull/50). The only remaining compilation gotcha now is that I have to maintain a patched arduino framework in order to get the library placed somewhere it will fit per https://github.com/BoschSensortec/BSEC-Arduino-library#4-additional-core-specific-modifications. I imagine on an ESP32 it would all "just work".

mcspr, 2020-09-15:
@trvrnrth I believe you misunderstood the changes we made, see:
https://community.bosch-sensortec.com/t5/MEMS-sensors-forum/BSEC-library-installation-instructions-for-ESP8266/td-p/14862
<sub>(*lone-tumbleweed.gif*)</sub>

We simply repackage the provided library .a file, after doing `mv *.o *.c.o` on the contents, no need to patch the framework files when they already do the stuff we want.

trvrnrth, 2020-09-18:
@mcspr Neat! That makes things much simpler! I was indeed completely unaware of those changes.

I've opened a PR with a re-packaged library upstream at https://github.com/BoschSensortec/BSEC-Arduino-library/pull/51 as I'd made the changes in my fork anyway in order to support the [component I've put together](https://github.com/trvrnrth/esphome-bsec-bme680). Hopefully it will get picked up and some action taken with it.

For reference here as well the re-packaging command I used is as follows:
```
ar t libalgobsec.a | awk '{
  old=$1;
  new=$1;
  sub(/\.o/, ".c.o", new);
  system("ar x libalgobsec.a "old);
  system("ar dS libalgobsec.a "old);
  system("mv "old" "new);
  system("ar q libalgobsec.a "new);
  system("rm "new)
} END {
  system("ar s libalgobsec.a")
}'
```

trvrnrth, 2020-11-22:
For anyone following along here the [latest release of the BSEC Software Library (v1.6.1480)](https://github.com/BoschSensortec/BSEC-Arduino-library/releases/tag/v1.6.1480) now includes a version of the library packaged as discussed above for esp8266.

Now that little hurdle is out of the way I have marked https://github.com/esphome/esphome/pull/1313 which adds a component to support the BME680 sensor via BSEC as ready for review.

closed 2022-07-05 by nagyrobi