esphome/feature-requests #18: ESP32 Touch Raw Values

arnie580, 2019-02-01:
**Is your feature request related to a problem/use-case? Please describe.**
I have a capacitive rain sensor that can be read by the touch interface on ESP32.

**Describe the solution you'd like:**
A new sensor component to output the raw readings of the touch sensor. When in test mode usable values are output but there is no easy way to read into home assistant or process the values.

**Additional context:**
Only intermittent values would be needed, therefore regular readings would not need to be taken in the same way as the binary sensor. For example for my project I would be looking to read the value at most once every 30 seconds.

larsenglund, 2020-05-26:
Did anyone start working on this?
I would like to use this for a soil moisture sensor to know when to water my plants.

Swamp-Ig, 2022-02-08:
I think I can see how to code this up. I wonder if it would be accepted?

nagyrobi, 2022-07-06:
> I think I can see how to code this up. I wonder if it would be accepted?

Of course!

kbx81, 2023-07-18:
I can probably take a stab at this

kbx81, 2023-07-23:
Added a method in https://github.com/esphome/esphome/pull/5116 to enable getting the raw values which could be used as the lambda for a [template sensor](https://esphome.io/components/sensor/template.html).

suside, 2024-04-01:
I think this can be closed, example usage `Chip is ESP32-D0WDQ6 (revision v1.0)` with esphome `Version: 2024.3.1`:

```yaml
esp32_touch:
  id: esp32_touch_1

binary_sensor:
  - platform: esp32_touch
    id: esp32_touch_pad
    pin: GPIO4
    threshold: 0

sensor:
  - platform: template
    name: "Raw touch value"
    lambda: |-
        return id(esp32_touch_pad).get_value();
    update_interval: 3s
```

- untouched `'Raw touch value': Sending state 1149.00000  with 1 decimals of accuracy`
- touched `'Raw touch value': Sending state 225.00000  with 1 decimals of accuracy`
- touched over insulator `'Raw touch value': Sending state 884.00000  with 1 decimals of accuracy`

closed 2024-04-02 by kbx81

Swamp-Ig, 2024-04-02:
It would be good if this could be documented. I will look at creating a PR to update the documentation. This issue is closed, so have created another issue.