esphome/feature-requests #19: [Feature Request] Please add CO2 sensor CCS811

pluskal, 2019-02-01:
**What new integration would you wish to have?**

CO2 sensor CCS811

* CO2 sensor with temperature and humidity sensor.
* I2C communication.

Integration can extend already implemented support for MH-Z19 #94 

**If possible, provide a link to an existing library for the integration:**
https://github.com/sparkfun/SparkFun_CCS811_Arduino_Library
https://github.com/maarten-pennings/CCS811

**Please describe your use case for this integration and alternatives you've tried:**

CO2 measuring in a room.

**Additional context:**
A lot of useful information about sensor [CJMCU-811](https://revspace.nl/CJMCU-811)
Multiple boards with this sensor:
* [SparkFun](https://www.sparkfun.com/products/14193)
* [Adafruit](https://www.adafruit.com/product/3566)
* [CJMCU](https://www.banggood.com/CJMCU-811-CCS811-Carbon-Monoxide-CO-VOCs-Air-Quality-Digital-Gas-Sensor-p-1157216.html?cur_warehouse=CN)

Edit: SPI->I2C

pashar1, 2019-02-23:
looking for support of the sensor as well

casnacaj, 2019-03-21:
I got the CCS811 working using the custom sensor. See https://github.com/casnacaj/esphome_lkpr/blob/master/lkpr_bedroom.yaml

pashar1, 2019-03-21:
i see that you have bmp280 as well in that setup.. are you feeding temp and humidity to CCS811?

casnacaj, 2019-03-22:
No, I calibrate the CCS811 to 25°C. I have PCB module with CCS811+SI7021+BMP280 chip. I don't know how to read another esphome sensor value from my custom sensor. But good idea - I may calibrate CCS811 to temperature read from SI7021. Here is the C++ code behind: https://github.com/casnacaj/esphome_lkpr/blob/master/ccs811_si7021.h

closed 2019-05-13 by OttoWinter