esphome/feature-requests #22: [Question] Is it possible to Transmit 433Mhz code recieved from MQTT command ?

animavitis, 2019-02-02:
Hi

Is it possible to send MQTT command to esphomeyaml device with code that should be then send via Remote Transmitter Component?? ie.:

From my Home assistant i want to send something like that:
```
        topic: "esphomeyaml/rftransmit/command"
        payload: "{"rc_switch_raw":{"code":"001010011001111101011011","protocol":"1"}}"
```

then i want esphomeyaml to send automatically this as RF code  (without define this code as Switch in eshomeyaml config file).

Any idea how to do this??

dlashua, 2019-03-06:
even a sensor.remote_receiver component that just sent the raw data would be useful. Getting something like:

````
[4088, -1542, 1019, -510, 513, -1019, 510, -509, 511, -510, 1020,
          -1020, 1022, -1019, 510, -509, 511, -510, 511, -509, 511, -510,
          1020, -1019, 510, -511, 1020, -510, 512, -508, 510, -1020, 1022,
          -1021, 1019, -1019, 511, -510, 510, -510, 1022, -1020, 1019,
          -1020, 511, -511, 1018, -1022, 1020, -1019, 1021, -1019, 1020,
          -511, 510, -1019, 1023, -1019, 1019, -510, 512, -508, 510, -511,
          512, -1019, 510, -509]
````

in HomeAssistant seems pretty nasty, but something could convert it on the HA end to a more manageable string to perform automations with. 

The alternative, at the moment, is to flash a "dumping" firmware to the ESP device. Trigger the sending device. Get the dump code. Add that code to the ESPHome YAML. Then reflash the ESP device. And while this process certainly provides the cleanest end result (a entity in HASS for every code handled in exactly the way that code should be handled) it's quite tedious for tinkerers.

OttoWinter, 2019-03-06:
ESPHome can dump these codes too you know - see remote_receiber

dlashua, 2019-03-15:
Yes, but ESPHome dumps them to the log. (per https://esphome.io/components/switch/remote_transmitter.html#finding-remote-codes). This means, that to add a new 433Mhz code, I have to view the log for the ESPHome device, trigger the 433Mhz, look for the code, add the code to the YAML for the ESPHome device, recompile the code, and upload again.

Having this data visible in a text_sensor would mean I could act on it in Home Assistant or via MQTT without having to recompile for every new 433Mhz device I add.

OttoWinter, 2019-03-15:
@dlashua All of those actions are not a ton of work, especially when you batch the operation. So for example set up esphome to dump everything, then press all buttons and record the data, while adding it to the YAML. I've done it before and it's not much work. Besides, I doubt it would be any quicker through HA because you would still need to copy-paste it.

And then there's the fact that recompiling is not that slow when you don't add/remove a component.

I can tell you that due to technical reasons what you're asking is impossible. Text Sensor strings have a limited length because of how the TCP stack on these ESPs works - that's also why the raw messages are split into multiple lines in the logs.

dlashua, 2019-03-15:
I totally get that the current platform limits what it is capable of.

I had debating changing the custom code I run on Arduino to provide exactly
this functionality over to ESPHome, hence the reason I jumped on in support
of this "Issue". But, it's also not a problem to just keep running my
existing code as it works just fine.

OttoWinter, 2019-05-27:
Ok, with 1.13 this will be possible - all remote transmits are now "actions", with templatable values.

closed 2019-05-27 by OttoWinter