esphome/feature-requests #23: [Feature Request] MQTT-microphone (Hermes protocol)

vevsvevs, 2019-02-02:
**What new integration would you wish to have?**
MQTT-microphone for ESP32.

**If possible, provide a link to an existing library for the integration:**

**Please describe your use case for this integration and alternatives you've tried:**
In my opinion, HassOS is the best easy-deployment solution for HA-beginners and for home automation at all! But it also has many limitations related to used platform, especially in case of OVA-based images (which i personally use). The most disappointing thing is a lack of support of Bluetooth (device tracking) and audio (voice control) USB-dongles. BT issue was fully-solved with your amazing ESPHome component, but the audio problem still remains. There is a very promising project [Rhasspy](https://github.com/synesthesiam/rhasspy-hassio-addon), which can take audio input data from MQTT payload ([Hermes protocol](https://docs.snips.ai/ressources/hermes-protocol)) for later processing. IMO it would be amazing to get such integration in ESPHome!

**Additional context:**

sehraf, 2020-10-16:
I second this, especially with the esp32 audio kit.

mrhatman, 2022-12-15:
So I am really interested in doing the development work for this over my Christmas break off  work. There is a great example [here](https://github.com/Romkabouter/ESP32-Rhasspy-Satellite). 

I am thinking we allow for as much customization as we can for these devices. Base features:
* Configurable SiteID
* Trigger for Hermes connection
* Trigger for wake word detected/Listen detected
* Trigger for Listen ended
* Trigger for no matches detected 
* Trigger for Audio playback started
* Trigger for Audio playback ended

ser, 2023-08-22:
will it be ready for christmas '23? <3