esphome/feature-requests #25: MQTT convention: Homie

davidgraeff, 2019-02-02:
I know this is a HA targeted project, but it could be used with auto discovery very easily in openHAB as well if a second mqtt convention could be supported:

homieiot.github.io

WDYT? :)

knopserl, 2019-02-10:
I would also like to use this great solution in conjunction with OpenHAB!
I like the ESP32 HW and I need a flexible gateway/controller with support also BLE to integrate the new Xiaomi Mijia Temperature/Humidity LCD Sensor

That would be super cool!

https://de.aliexpress.com/item/Original-Xiao-mi-mi-jia-Bluetooth-Temperatur-Hu-mi-dity-Sensor-Lcd-bildschirm-Digitale-Thermometer-Feuchtigkeit/32843440689.html

peterhoeg, 2019-06-10:
1.11.0 has long come and gone - is there any plans to implement this?

davidgraeff, 2019-06-10:
Not by the ESPHome developers themselves, I guess

peterhoeg, 2019-06-10:
Since @OttoWinter tagged this issue to a milestone, I was kind of hoping... ;-)

peterhoeg, 2019-06-12:
@OttoWinter is it safe to assume there are no immediate plans to support the homie spec? Would you accept a PR that adds support?

OttoWinter, 2019-06-12:
@peterhoeg No not immediate plans. Also I think there will be other problems with homie convention, as ESPHome is more set up for single-topic publishing, but I'd take a PR that implements homie and doesn't cause issues for the normal MQTT protocol