esphome/feature-requests #26: Support for DFPlayer Mini SKU:DFR0299

silfa718, 2019-02-02:
**What new integration would you wish to have?**

DFPlayer Mini SKU:DFR0299

https://www.dfrobot.com/wiki/index.php/DFPlayer_Mini_SKU:DFR0299

**If possible, provide a link to an existing library for the integration:**

**Please describe your use case for this integration and alternatives you've tried:**

I would like to get my nodemcu play audio files as triggered sounds effects.

**Additional context:**

I couldn't find to many other modules/sensors that would work. It would be great to find one that had an integrated speaker in it. The alternative I was looking at was to find something with an audio output jack to plug in some old computer speakers.

glmnet, 2019-06-19:
Gonna start with this, I'm building the yaml syntax to get some feedback on design
An integration with [media on home assistant](https://www.home-assistant.io/components/media_player/) seems overkill 

Will work over UART bus
````yaml
dfplayer_mini:
  id: # optional to use lambdas
  uart_id: # optional, if you have many uarts
  volume: 15 # 0..30 initial volume, default to 15
````

Actions: (will add actions for the following lambdas)
* `id(dfplayer_mini).set_volume(int)` #0..30
* `id(dfplayer_mini).play_track(int)`  # df player plays files given an id
* `id(dfplayer_mini).pause()`
* `id(dfplayer_mini).start()` #resume from pause

Here is some more info: https://community.home-assistant.io/t/dfplayer-mini-mp3-player-with-esphome/122454

ronschaeffer, 2019-06-22:
Great that's you're starting with this. Could also be useful to have actions to query the current status (for use in automations) and to reset the DFPlayer.

closed 2019-10-19 by OttoWinter