esphome/feature-requests #27: other mqtt channel

voicevon, 2019-02-02:
**Is your feature request related to a problem? Please describe.**
I have an sensor far away from my house(let's say 1km distance).There is no wifi, even no AC  power.
**Describe the solution you'd like**
Will esphomelib support LuRaWan , or GSM  connection to server?

**Additional context**
Right now, esphomelib has organized all data to an object `mqtt_message`(include topic ,payload, qos, etc.), Then it called publish() method in `mqtt_client_`  which is an AsyncMqttClient  without source code, but inside depended library. Obviously the `mqtt_client` binds AsyncTcp+Wifi. I would like to name it "Channel".

GSM module connect  to esp32 through UART (another channel),  All communication between the esp32 and GSM module  is AT-commands. 

Imaging Esphomelib `mqtt component ` has a member called `channel`. Just modify the value of `channel`  form `wifi` to `gsm`.  The esp32 will turn to  the new channel.  Because I have tested the node under `channel_wifi`, so the node should be no issue on other channel.   Isn't it awesome?

thompsa, 2019-03-17:
The ESP32 has an experimental ppp client which would solve the issue of serial not having open/close schematics. https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/kconfig.html#config-ppp-support

dj-fiorex, 2023-01-18:
2 years have passed, almost 3, is there any news? i would like to implement a LoRa comunication channel!

nagyrobi, 2025-02-07:
Making progress in https://github.com/esphome/esphome/pull/8187, and first implementation is for UART: https://github.com/esphome/esphome/pull/8214. This is a generalised approach, which will make possible to develop simpler components for exchanging data directly between nodes by any means, like TCP, ESP-Now, LoRa, etc.