esphome/feature-requests #28: MCP39F501 energy sensor (Shelly 2)

oscar-b, 2019-02-02:
**What new integration would you wish to have?**
MCP39F501, the energy monitoring chip present on the Shelly 2 (https://shelly.cloud/shelly2/)
Datasheet: https://www.microchip.com/wwwproducts/en/MCP39F501

**If possible, provide a link to an existing library for the integration:**
https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xnrg_04_mcp39f501.ino

subzero79, 2019-02-08:
This would be nice to have. There is also new Shellie’s coming with power meters, no idea if they will come with the same chip.

blademckain, 2019-03-11:
yeah very nice features

tubalainen, 2019-05-21:
Also looking very much forward to this! <3

Poeschl, 2019-08-26:
Any progress on this?

max246, 2023-05-15:
I would say to close ti since :
Shelly 2 is discontinued and replaced by Shelly 2.5]

@jesserockz