esphome/feature-requests #29: MAX44009 / GY-49 ambient light sensor support ?

vincegre, 2019-02-02:
Is it planned to get support or is there a workaround already for MAX44009 / GY-49 ambient light sensor ? It's an I2C module pretty efficient for light sensing.
Basic description of it here: https://www.aliexpress.com/snapshot/0.html?spm=a2g0s.9042647.6.2.2bfd4c4ddKFYKF&orderId=507448985225914&productId=32804395737

Thanks

Vincèn

vincegre, 2019-03-24:
@OttoWinter any chance to get that sensor implemented anytime soon ? thanks ;)

Tim-Dixon, 2019-07-01:
In case anyone stumbles across this thread, which seems to be the only one regarding this sensor, the above code provided me with some errors after implementation due to changes in the esphome library. I by no means can write in C++, however by attemting to read the error codes, I was able to sucessfully compile the program with the below code. Pay attention to the I2C pins defined in both the C++ code, and make sure you define them in your YAML config. If there are any errors or something I missed I'm open to anyone's help. Like I said, this is merely a few changes after reading error codes from compiling. Hope this helps! :)

```
#include "esphome.h"
#include "Max44009.h"

using namespace esphome;

class MAX44009Sensor : public sensor::PollingSensorComponent {
 public:
    Max44009 tsl = Max44009(0x4A, 21, 22); //these are the i2c pins being used

  MAX44009Sensor(const std::string &name, uint32_t update_interval) : sensor::PollingSensorComponent(
    name, update_interval) {}
  
  void update() override {
    float lux = tsl.getLux(); // in lux
    int err = tsl.getError();
    if(err) 
      ESP_LOGI("luxmeter", "Got error code: %1i", err);  
    else   
      //ESP_LOGD(TAG, "Got lux=%.1flux", lux);   
      publish_state(lux); 
  }

   std::string icon() override {
    return "mdi:theme-light-dark";
  }

    std::string unit_of_measurement() override { return "Lux"; }
  int8_t accuracy_decimals() override { return 2; } // 2 decimal places of accuracy.
};
'
```

jamien82, 2019-07-13:
Hi,
Im having trouble with this verifying.
Any ideas

```
esphome:
  name: weather
  platform: ESP8266
  board: d1_mini_pro
  includes:
  - Max44009jamie.h

wifi:
  ssid: "embedded"
  password: "esp8266esp8266"

# Enable logging
logger:

i2c:
  sda: 4
  scl: 5
  scan: True

ads1115:
  - address: 0x48
  
dallas:
  - pin: GPIO2
  
sensor:
  - platform: ads1115
    multiplexer: 'A0_GND'
    gain: 6.144
    name: "ADS1115 Channel A0-GND"
  - platform: bme280
    temperature:
      name: "BME280 Temperature"
      oversampling: 16x
    pressure:
      name: "BME280 Pressure"
    humidity:
      name: "BME280 Humidity"
    address: 0x76
    update_interval: 60s
  - platform: dallas
    address: 0x40031724D569FF28
    name: "Temperature #1"
  - platform: custom
    auto *luxmeter = App.register_component( new MAX44009Sensor("Light Level", 30 * 1000)); 
    App.register_sensor(luxmeter);
    sensors:
    - name: "My Custom lux Sensor"
      unit_of_measurement: lux
      accuracy_decimals: 1
    
  
   
 
    
    
# Enable Home Assistant API
api:

ota:

```
Validate window
```
INFO Reading configuration...
ERROR Error while reading config: Invalid YAML syntax. Please see YAML syntax reference or use an online YAML syntax validator:

while scanning a simple key
  in "/config/esphome/weather.yaml", line 45, column 5:
        auto *luxmeter = App.register_co ... 
        ^
could not find expected ':'
  in "/config/esphome/weather.yaml", line 46, column 5:
        App.register_sensor(luxmeter);
        ^
```

randybb, 2019-10-18:
Any news about official integration?

LiuVik, 2019-10-20:
> В случае, если кто-то наткнется на этот поток, который, кажется, является единственным относительно этого датчика, приведенный выше код предоставил мне некоторые ошибки после реализации из-за изменений в библиотеке esphome. Я никоим образом не могу писать на C++, однако, пытаясь прочитать коды ошибок, я смог успешно скомпилировать программу с приведенным ниже кодом. Обратите внимание на контакты I2C, определенные в обоих кодах C++, и убедитесь, что вы определяете их в своей конфигурации YAML. Если есть какие-то ошибки или что-то, что я пропустил, я открыт для любой помощи. Как я уже сказал, это всего лишь несколько изменений после считывания кодов ошибок из компиляции. Надеюсь, это поможет! :)
> 
> ```
> #include "esphome.h"
> #include "Max44009.h"
> 
> using namespace esphome;
> 
> class MAX44009Sensor : public sensor::PollingSensorComponent {
>  public:
>     Max44009 tsl = Max44009(0x4A, 21, 22); //these are the i2c pins being used
> 
>   MAX44009Sensor(const std::string &name, uint32_t update_interval) : sensor::PollingSensorComponent(
>     name, update_interval) {}
>   
>   void update() override {
>     float lux = tsl.getLux(); // in lux
>     int err = tsl.getError();
>     if(err) 
>       ESP_LOGI("luxmeter", "Got error code: %1i", err);  
>     else   
>       //ESP_LOGD(TAG, "Got lux=%.1flux", lux);   
>       publish_state(lux); 
>   }
> 
>    std::string icon() override {
>     return "mdi:theme-light-dark";
>   }
> 
>     std::string unit_of_measurement() override { return "Lux"; }
>   int8_t accuracy_decimals() override { return 2; } // 2 decimal places of accuracy.
> };
> '
> ```

Hi Tim.
help me pls - MAX44009 do not working ,i create MAX4409.h with yours code, put's him into hassio and in config i showed next:
i2c:
  sda: 4 
  scl: 5 
  scan:  True
  id:  bus_a

sensor:
  platform: |-
    custom auto *luxmeter = App.register_component( new MAX44009Sensor("Light Level", 30 * 1000)); App.register_sensor(luxmeter);
  sensors: 
    - name: My Custom lux Sensor
      unit_of_measurement: lux
      accuracy_decimals: 1

whats wrong ?

Tim-Dixon, 2019-10-31:
@LiuVik can you give us a little more information on your setup and where you placed files, code, etc? Your text was a little confusing.

egranto, 2020-03-23:
Create a file on  _\\homeassistant\config\esphome_ i named mine **cjmcu.h**
add 
`#include "esphome.h"
#include <Max44009.h>
#include <Wire.h>

class max44009 : public PollingComponent, public Sensor {
Max44009 light = Max44009(0x4A, 4, 5); //these are the i2c pins being used
 public:

   // constructor
  max44009() : PollingComponent(60000) {}
 
  void setup() override {
	  Wire.begin();
  }
  void update() override {
    float lux = light.getLux(); // in lux
      publish_state(lux); 
  }
  std::string icon() override {
    return "mdi:theme-light-dark";
  }

    std::string unit_of_measurement() override { return "Lux"; }
    int8_t accuracy_decimals() override { return 2; } // 2 decimal places of accuracy.
};`

on your YAML just need 
`
  includes:
    - cjmcu.h

`
and then 
`  - platform: custom
    lambda: |-
      auto my_sensor = new max44009();
      App.register_component(my_sensor);
      return {my_sensor}; 

    sensors:
      name: "Light Sensor:"`

grimmsq, 2020-05-14:
Hello, I've been struggling with these for some time now, maybe you can help me

**I've created a file called lux.h that I placed in \config\esphome with the following content**

"#include "esphome.h"
#include "Max44009.h"
class max44009 : public PollingComponent, public Sensor {
 public:
  Max44009 light = Max44009(0x4A, 4, 5); //these are the i2c pins being used
// constructor
  max44009() : PollingComponent(60000) {}

void update() override {
  float lux = light.getLux(); // in lux
  int err = light.getError();
  if(err) 
      ESP_LOGI("luxmeter", "Got error code: %1i", err);  
    else   
      //ESP_LOGD(TAG, "Got lux=%.1flux", lux);   
      publish_state(lux); 
}
std::string icon() override {
return "mdi:theme-light-dark";
}

std::string unit_of_measurement() override { return "Lux"; }
int8_t accuracy_decimals() override { return 2; } // 2 decimal places of accuracy.
};"

**After that I added this in the .yaml file**

esphome:
  name: seraf
  platform: ESP8266
  board: d1_mini
  includes:
    - lux.h
  libraries:
    - "MAX44009 library"
  
wifi:
  ssid: "XXXXX"
  password: "XXXXX"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Seraf Fallback Hotspot"
    password: "XXXXX"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
i2c:
 sda: 4
 scl: 5
 scan: True
 id: bus_a
ota:
sensor:
 - platform: custom
    lambda: |-
      auto my_sensor = new max44009();
      App.register_component(my_sensor);
      return {my_sensor};
    sensors:
      name: "Light Sensor:"

**When I compile it i get the following error:**

In file included from src/main.cpp:19:0:
src/lux.h:8:3: error: 'Max44009' does not name a type
   Max44009 light = Max44009(0x4A, 4, 5); //these are the i2c pins being used
   ^
src/lux.h: In member function 'virtual void max44009::update()':
src/lux.h:14:15: error: 'light' was not declared in this scope
   float lux = light.getLux(); // in lux

Can you please tell me what am I doing wrong?
Thanks in advance!

nvvasilev, 2020-05-17:
Hello, @grimmsq 
I managed to make the sensor work with the following configuration:

In the ESPHome config folder I created the file `ESPHomeMax44009Sensor.h`. 
Here is the path to the file: `homeassistant/config/esphome/CustomLibs/ESPHomeMax44009Sensor.h`
My .yaml file is one directory up in: `homeassistant/config/esphome/weatherstation.yaml`.

Here is the contents of the **ESPHomeMax44009Sensor.h**:

```
#include "esphome.h"
#include "Max44009.h"

class MAX44009Sensor : public PollingComponent, public Sensor  {
 public:
  Max44009 myLux = Max44009(0x4A, 4, 5); //these are the i2c pins being used

  MAX44009Sensor() : PollingComponent(15000) {} // 15 seconds update interval
  
  void setup() override {}
  
  void update() override 
  {
    float lux = myLux.getLux(); // in lux
    
    publish_state(lux); 
  }
};
```

And this is how I use it in **weatherstation.yaml**:

```
substitutions:
  display_name: outsideWeather

esphome:
  name: weather_station
  platform: ESP8266
#  board: d1_mini_pro
  board: d1_mini
  includes:
    - CustomLibs/ESPHomeMax44009Sensor.h
  libraries:
    - "Max44009"

wifi:
  ssid: !secret iot_wifi_ssid
  password: !secret iot_wifi_pass
  fast_connect: true
  manual_ip:
    static_ip: !secret ip_weatherstation
    gateway: !secret ip_gateway
    subnet: !secret ip_subnet
    dns1: !secret ip_dns1

#api:

# Enable logging
logger:

mqtt:
  broker: !secret mqtt_broker
  username: !secret mqtt_username
  password: !secret mqtt_password
  birth_message:
    topic: solarsensor/birthdisable
    payload: disable
  will_message:
    topic: solarsensor/willdisable
    payload: disable

ota:

i2c:
 - id: ic_1
   sda: 4
   scl: 5
   scan: false

sensor:
  - platform: hdc1080
    temperature:
      name: ${display_name} HDC1080 Temp
    humidity:
      name: ${display_name} HDC1080 Humi
    address: 0x40
    update_interval: 30s

  - platform: dht
    temperature:
      name: ${display_name} DHT Temperature
    humidity:
      name: ${display_name} DHT Humidity
    pin: D4
    update_interval: 30s

  - platform: custom
    lambda: |-
      auto lux_sensor = new MAX44009Sensor();
      App.register_component(lux_sensor);
      return {lux_sensor};
  
    sensors:
      name: ${display_name} Lux
      unit_of_measurement: lx

  - platform: adc
    pin: A0
    name: ${display_name} Battery
    update_interval: 15s
    filters:
      - multiply: 5.5238

deep_sleep:
  run_duration: 40s
  sleep_duration: 5min
```
I followed the official ESPHome Custom Sensor Component guide [here](https://esphome.io/components/sensor/custom.html) and the digiblurDIY's video about Solar Powered Weather Station [here](https://www.youtube.com/watch?v=VUqOIPVbeF0).

This works for me and I hope it works for you as well.

mar00, 2020-07-04:
mam problem z codem 
program przy weryfikacji zatrzymuje sie na linii .
brak jest dokladnego wyjasnienia przyczyny error. 
SSDAX44009I2C:13:18: error: no matching function for call to 'MAX44009::MAX44009(int)'

 MAX44009 Lux(0x4A);

drtaul, 2020-07-12:
Tried the different approaches provided in this thread with no success getting the MAX44009 to operate with basic ambient light readings. Finally took the code from [Dan Tudose](https://github.com/dantudose/MAX44009/tree/master/src) and [implemented a sensor component using native ESPHome APIs](https://github.com/drtaul/hatweaks/tree/master/esphome/components/max4409) (removed the arduino references making it dead simple).

This is all towards completing a DIY weather station that integrates with Home Assistant and weewx weather station data server. I have starting capturing what I did including this sensor and others in a [github repository](https://github.com/drtaul/hatweaks).

Hope it is helpful. 

I will note that as part of this effort I discovered how to add ESPHome components locally via the 'custom_components' subdirectory. I just happened upon a discussion of this facility that made my life so much easier.

Coolie1101, 2020-10-13:
@drtaul  I just tried implementing your custom component for the [Max44009 Sensor](https://github.com/drtaul/hatweaks/tree/master/esphome/components/max4409), validation passed, but Esphome is throwing the following errors when trying to upload, any I idea what the issue may be?

"In file included from src\main.cpp:83:0:
src\max44009.h:10:6: error: multiple definition of 'enum esphome::max44009::MAX44009Constants'
 enum MAX44009Constants {
      ^
In file included from src\esphome.h:24:0,
                 from src\main.cpp:3:
src\esphome/components/max44009/max44009.h:10:6: error: previous definition here
 enum MAX44009Constants {
      ^
In file included from src\main.cpp:83:0:
src\max44009.h:20:6: error: multiple definition of 'enum esphome::max44009::MAX44009Registers'
 enum MAX44009Registers {
      ^
In file included from src\esphome.h:24:0,
                 from src\main.cpp:3:
src\esphome/components/max44009/max44009.h:20:6: error: previous definition here
 enum MAX44009Registers {
      ^
In file included from src\main.cpp:83:0:
src\max44009.h:31:6: error: multiple definition of 'enum esphome::max44009::MAX44009CfgMasks'
 enum MAX44009CfgMasks {
      ^
In file included from src\esphome.h:24:0,
                 from src\main.cpp:3:
src\esphome/components/max44009/max44009.h:31:6: error: previous definition here
 enum MAX44009CfgMasks {
      ^
In file included from src\main.cpp:83:0:
src\max44009.h:45:7: error: redefinition of 'class esphome::max44009::MAX44009Sensor'
 class MAX44009Sensor : public sensor::Sensor, public PollingComponent, public i2c::I2CDevice {
       ^
In file included from src\esphome.h:24:0,
                 from src\main.cpp:3:
src\esphome/components/max44009/max44009.h:45:7: error: previous definition of 'class esphome::max44009::MAX44009Sensor'
 class MAX44009Sensor : public sensor::Sensor, public PollingComponent, public i2c::I2CDevice {
       ^
Compiling .pioenvs\multi_sense_test\lib2d9\ESP8266WiFi\ESP8266WiFiGratuitous.cpp.o
Compiling .pioenvs\multi_sense_test\lib2d9\ESP8266WiFi\ESP8266WiFiMulti.cpp.o
Compiling .pioenvs\multi_sense_test\lib2d9\ESP8266WiFi\ESP8266WiFiSTA-WPS.cpp.o
Compiling .pioenvs\multi_sense_test\lib2d9\ESP8266WiFi\ESP8266WiFiSTA.cpp.o
Compiling .pioenvs\multi_sense_test\lib2d9\ESP8266WiFi\ESP8266WiFiScan.cpp.o
*** [.pioenvs\multi_sense_test\src\main.cpp.o] Error 1
===================================================================== [FAILED] Took 41.14 seconds ====================================================================="

CMDR-Sloma, 2020-10-25:
> Tried the different approaches provided in this thread with no success getting the MAX44009 to operate with basic ambient light readings. Finally took the code from [Dan [...]

I have tried it and as well other methods but none seems to work.

CMDR-Sloma, 2020-11-02:
> Hello, @grimmsq
> I managed to make the sensor work with the following configuration:
>
> In the ESPHome config folder I created the file `ESPHomeMax44009Sensor.h`.
> Here is the path to the file: [...]

I had to edit this in order to get it to work, the code for the sensor was inconsistent and ESPHome was returning errors during compiling. Once I replaced every instance of **MAX44009Sensor** with **MAX44009** in **ESPHomeMax44009Sensor.h** it compiled fine. The code below should work as long the pins for the sensor are the same:
```
#include "esphome.h"
#include "Max44009.h"

class MAX44009 : public PollingComponent, public Sensor {
 public:
  Max44009 solarLux = Max44009(0x4A, 4, 5); // these are the i2c pins being used, make sure to replace 4, 5 if you are using different ones
  // Constructor
  MAX44009() : PollingComponent(60000) {} // 60 seconds update interval

  void setup() override {}

  void update() override {
    float lux = solarLux.getLux(); // in lux

    publish_state(lux);
  }
};
```
Add to sensor in your ESPHome yaml:
```
  - platform: custom
    lambda: |-
      auto lux_sensor = new MAX44009();
      App.register_component(lux_sensor);
      return {lux_sensor};

    sensors:
      name: ${display_name} Lux
      unit_of_measurement: lx
```

Make sure your .yaml has the right number of spaces and check if the pins are the same as in **ESPHomeMax44009Sensor.h**.

@mar00 Sprawdź kod powyżej, powinno działać, o ile używasz tych samych pinów do komunikacji z sensorem (4,5). Na moim Wemosie piny 4,5 są oznaczone jako D1, D2.

kisaev83, 2021-10-09:
> Несколько недель назад я опубликовал внешний компонент для поддержки датчика MAX44009. Вот пример: https://github.com/berfenger/esphome_components/blob/main/example_max44009.yaml
> 
> Дайте мне знать, работает ли это для вас, прежде чем отправлять мерж-реквест в основную линию esphome.

Yes, it works for me. Thanks for the componen
[image]
t!

jerryaycock, 2021-12-03:
Anything new? I am trying to add weather station. All was well. Had bad sensor and stopped. New sensor came in and now can't compile. Could it be updated ESPHome ? I used file from above nvvasilev .            only test file with include and I get the following error ///// In file included from /data/test/.piolibdeps/test/Max44009/Max44009.cpp:40:0:
/data/test/.piolibdeps/test/Max44009/Max44009.h:31:18: fatal error: Wire.h: No such file or directory

jerryaycock, 2021-12-07:
I have tried for weeks. Getting nowhere. I thought I had a defective sensor, Ordered new one and still nothing. I have tried  includes:
  - CustomLibs/ESPHomeMax44009Sensor.h
  libraries:
    - "Max44009"  
it fails to load- I found  external_components:
  - source: github://berfenger/esphome_components
    components: [ max44009 ]
it loads but I get no readings. I am using Home Assistant.
this is my file.
external_components:
  - source: github://berfenger/esphome_components
    components: [ max44009 ]
esphome:
  name: beachweather
  platform: ESP32
  board: esp32doit-devkit-v1
wifi:
  ssid: xxxxx
  password: !secret wifi_pwd
  manual_ip:
    static_ip: 
    dns1: 8.8.4.4
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "xxxx
    password: "xxxx
captive_portal:
# Enable logging
logger:
# Enable Home Assistant API
api:
ota: 
  password: "xxxxxx"
uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600
i2c:
  sda: 21
  scl: 22
  scan: True
  id: bus_a
sensor:
  - platform: pulse_counter
    pin:
      # pin D6
      number: GPIO32
      mode: INPUT_PULLUP
    unit_of_measurement: 'in' ##change to mm if metric
    name: 'Rain gauge'
    icon: 'mdi:weather-rainy'
    id: my_rain
    internal: true
    count_mode:
      rising_edge: DISABLE
      falling_edge: INCREMENT
    internal_filter: 13us
    update_interval: 60s
    #buckets = pulses
    #mm_per_bucket=0.2794
    #rain = buckets*0.2794
    filters:
  #    - multiply: 0.2794 #for mm
      - multiply: 0.011 # to inches
    accuracy_decimals: 3
  - platform: integration
    name: "Rainfall per min"
    id: rain_per_min
    time_unit: min
    unit_of_measurement: 'in'
[... 123 lines cut ...]
[11:00:05][C][pmsx003:227]:     State Class: 'measurement'
[11:00:06][C][pmsx003:227]:     Unit of Measurement: 'µg/m³'
[11:00:06][C][pmsx003:227]:     Accuracy Decimals: 0
[11:00:06][C][pmsx003:227]:     Icon: 'mdi:chemical-weapon'
[11:00:06][C][pmsx003:228]:   PM10.0 'Particulate Matter <10.0µm Concentration'
[11:00:06][C][pmsx003:228]:     State Class: 'measurement'
[11:00:06][C][pmsx003:228]:     Unit of Measurement: 'µg/m³'
[11:00:06][C][pmsx003:228]:     Accuracy Decimals: 0
[11:00:06][C][pmsx003:228]:     Icon: 'mdi:chemical-weapon'
[11:00:06][C][captive_portal:151]: Captive Portal:
[11:00:06][C][ota:082]: Over-The-Air Updates:
[11:00:06][C][ota:083]:   Address: 
[11:00:06][C][ota:086]:   Using Password.
[11:00:06][C][api:134]: API Server:
[11:00:06][C][api:135]:   Address: 192.168.31.112:6053
[11:00:06][C][api:139]:   Using noise encryption: NO
[11:00:06][C][sntp:050]: SNTP Time:
[11:00:06][C][sntp:051]:   Server 1: '0.pool.ntp.org'
[11:00:07][C][sntp:052]:   Server 2: '1.pool.ntp.org'
[11:00:07][C][sntp:053]:   Server 3: '2.pool.ntp.org'
[11:00:07][C][sntp:054]:   Timezone: 'EST5EDT,M3.2.0,M11.1.0'
[11:00:07][C][mdns:084]: mDNS:
[11:00:07][C][mdns:085]:   Hostname: beachweather

Any Pointers appreciated.

i7i5, 2021-12-30:
Thanks a lot for your solution,
I've used this one a long time:
```
  - platform: custom
    lambda: |-
      auto lux_sensor = new MAX44009();
      App.register_component(lux_sensor);
      return {lux_sensor};

    sensors:
      name: ${display_name} Lux
      unit_of_measurement: lx
```

Is there any Pr already?

BR
i7i5

closed 2022-02-19 by OttoWinter