esphome/feature-requests #31: New integration request: matrix keypad

MatthewFlamm, 2019-02-02:
I'd like to have a keypad available, ideally with some method of aggregating multiple keys before sending to HA.  Perhaps having a 'send' key, and/or a fixed/specifiable message length.

There is an existing library here: https://github.com/Nullkraft/Keypad

My use case for now is to arm/disarm an alarm system in HA.

Syralist, 2019-08-05:
I just rigged together a custom sensor for a keypad. 
You can find it here: https://github.com/Syralist/esphomekeypad

MatthewFlamm, 2019-08-05:
Hot stuff @Syralist .  Selfishly, can you have add an explanation of how to use in the README?  I'm not knowledgable about custom components, or how to parse what functionality this has.  I would love to test it out.  Looks like I need to get a matrix keypad now.

MatthewFlamm, 2019-08-05:
I should have just opened the files, pretty easy to parse the functionality, even if I don't understand the c++ well enough.

Syralist, 2019-08-08:
@MatthewFlamm glad you were able to figure out my code ;) I just added another variant that gathers the keypresses into a string rather than publishing each key on its own.

pplucky, 2019-08-15:
> @MatthewFlamm glad you were able to figure out my code ;) I just added another variant that gathers the keypresses into a string rather than publishing each key on its own.

Your custom sensors look great, but due to my noobness, I'm still trying to figure out how to wire the keypad directly to the NodeMCU ESP8266.

Would you be able to provide schematics on physical wire connections, to make it easier for noobs like me?

Thanks in advance.

Syralist, 2019-08-20:
> ...
> 
> Would you be able to provide schematics on physical wire connections, to make it easier for noobs like me?
> 
> Thanks in advance.

I used an old keypad with the connectors directly at the back. Modern ones are even easier. On http://www.circuitbasics.com/how-to-set-up-a-keypad-on-an-arduino/ there is a good tutorial with pictures and schematics.
The pins for the columns and the rows are defined in the "keypad_sensor.h":
```
    byte colPins[n_cols] = {D3, D2, D1};
    byte rowPins[n_rows] = {D7, D6, D5, D4};
```
 I hope this helps.

rradar, 2021-01-25:
https://github.com/ssieb/custom_components/tree/master/keypad

closed 2022-07-01 by nagyrobi