esphome/feature-requests #35: There is no posibility to connect to wifi with WIFI 802.1x EAP (WPA2-Enterprise)

komw, 2019-02-02:
I would be great, if esphomelib devices can use a WPA2-Enterprise Wifi passwords. Currently only simple WPA2 Personal is supported.
I notived that there is a support in arduino for wpa2-enterprise for ESP8266 (https://github.com/esp8266/Arduino/issues/1032), so it shouldn't be hard to add it.

YSmetana, 2019-06-06:
Really need WPA2-Enterprise ability to work in office environment.

randybb, 2019-06-07:
I am going to use IoT nodes with ESPhome in 802.1x network, so WPA2-Enterprise mode is the must for me :( Some nodes will use wired connection, so maybe would be good have it for Ethernet as well.

Preisschild, 2019-09-18:
Just tested WPA2-PEAP on my ESP01 running arduino, so it should be possible

lolorc, 2019-11-28:
wpa eap works with the arduino ide example, I also been using wpa eap with https://github.com/shmuelzon/esp32-ble2mqtt
I guess it shouldn't be to difficult to add support for it.
I use wpa eap at work and at home :)

OttoWinter, 2019-11-29:
> I guess it shouldn't be to difficult to add support for it.

That's the problem it probably *is*. Dealing with WiFi on ESP32s (especially ESP8266s) is a very painful task, as so many things in the WiFi core of these MCUs is not working well (crashing in weird ways etc). The added cost of maintaining a WPA enterprise auth (which is probably even less stable than regular WPA2 on ESPs, as it has fewer users) is huge.

lolorc, 2019-12-01:
Ok, fair enough, thanks. Based on your answer I guess I'll choose stability :-)

wwebers, 2020-05-02:
Any progress in this? According to Espressif at least EAP-TLS should work, which is awesome to get a secured home automation environment. Is anyone working on this already? Even though Otto's argument of the added cost is valid, the added security shouldn't be ignored IMHO.

closed 2020-07-10 by glmnet