esphome/feature-requests #36: [Feature Request] Support MCP23017 port expander

definitio, 2019-02-02:
Hello,
It would be nice to have support for that expander.

papperone, 2019-03-07:
any news on this PR? I would relly love to have MCP23017 (and 23008 as same chip with just 8 GPIO) supported as I will be able to move so many of my project to ESPHome while today I'm stuck with other firmware or even custom written ones... :(

OttoWinter, 2019-03-08:
This has already been implemented here https://github.com/esphome/esphome/pull/466

closed 2019-03-08 by OttoWinter

Ausonius, 2019-03-27:
Hi I am trying to control through a wemos d1 esphome mcp23017 trying to enable 8 8-button relays by following the code I can only make 8 buttons work but unfortunately the relays or LEDs do not light up when the code is in place I wanted to know where I am wrong the problem arises on the OUTPUT do not enable LED or relay you can see where it is wrong

esphome:
  name: esp_realy_8ch
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "my SSID"
  password: "my password"
  manual_ip:
    static_ip: my ip
    subnet: 255.255.255.0
    gateway: my gateway

# Enable logging
logger:
  
# Enable Home Assistant API
api:
  password: 'start0esp'

ota:
  password: 'start0esp'

i2c:
  sda: 4
  scl: 5
  scan: True

# Example configuration entry
mcp23017:
  - id: 'mcp23017'
    address: 0x20

binary_sensor:
  - platform: gpio
    name: "Pulsante 1"
    pin:
      mcp23017: mcp23017
      number: 0
      mode: INPUT_PULLUP
      inverted: True
    on_press:
    - switch.toggle: relay_1

switch:
  - platform: gpio
    name: "Relay 1"
    id: relay_1
    pin:
      mcp23017: mcp23017
      number: 0
      mode: OUTPUT
      inverted: True

Ausonius, 2019-03-28:
on the MCP when using the OUTPUT mode it does not work nn there is no output from the voltage pins I suppose there is an error in the code made thousands of tests putting an LED on one of the pins in output but nn does not want to turn on in any way