esphome/feature-requests #37: Support RC522

thoscut, 2019-02-02:
It's a great 1.8.0 release and the support for NFC readers is a great addition.
I wondered if you plan to support the RC522 as well?

apeeters, 2019-03-01:
Does this library help? It seems to be more up-to-date: https://github.com/miguelbalboa/rfid

schmurtzm, 2019-03-08:
Yes this library seems nice, it is used in the great project [esp-rfid](https://github.com/esprfid/esp-rfid).
Hope this will help to implement it in esphome :)

nickrout, 2019-03-25:
I was fishing through my electronics drawer and have one of these, so look forward to support.

HausnerR, 2019-04-13:
Hey,
I created simple integration through ESPHome custom component API.
Have a look here: https://github.com/HausnerR/esphome-door-lock

It's really simple and exposes various data to HA.
Can be inspiration for your needs.

SteveEdson, 2020-09-16:
Can this be merged as an official integration? I have an M5Stack Atom with a RFID module and would love to be able to use it officially.

nicogramm, 2020-09-18:
I'd also love to see this officially supported.

glmnet, 2020-09-21:
I got some dirty code running already. Is there a need to support other thing than reading the uid only?

SteveEdson, 2020-09-21:
Not for me personally no. Maybe for others

sgvj, 2020-09-22:
Thanks all for your work so far. Looking forward to this.

glmnet, 2020-09-24:
So, who's gonna help testing it?

sgvj, 2020-09-24:
@glmnet can do!

SteveEdson, 2020-09-24:
Happy to help test too

Gio-dot, 2020-09-24:
I don't want to belittle the work of anyone, but there are much better NFC readers on the market (even if it seems that not lot of people know them): they are Wiegand readers; there are many forms of them, all ip 65-68 and they use a specific protocol standardized for access control purpose. There is already a feature request for it, but it seems that no one can help to create a custom component or an integration.
Take a look here:
https://www.aliexpress.com/af/wiegand-reader.html?d=y&origin=n&SearchText=wiegand+reader&catId=0&initiative_id=SB_20200923231554

glmnet, 2020-09-24:
Ok, we can use this thread to report any issues / success stories.

In case anybody needs some help setting up the installation of the component, the required yaml is that of pn532, grab files from [here](https://github.com/esphome/esphome/tree/add-rc522/esphome/components/rc522) and [here is an explanation](https://www.reddit.com/r/homeassistant/comments/da48im/how_could_in_use_an_esphome_component_thats_been/) I just googled of how to copy files into your installation.

blademckain, 2020-09-25:
i would like to try using my RC522 with esphome.
how should i connect the wires?
is it possible to connect via uart?

blademckain, 2020-09-25:
ok i saw that the library only supports spi
[here is the same library but more updated with uart support too](https://www.arduinolibraries.info/libraries/mfrc522-spi-i2c-uart-async)

sgvj, 2020-09-25:
This worked perfectly for me. Scanned tags show up in homeassistant instantly.
THANK YOU SO MUCH!

Being a bit green I had to figure out some things please correct me if I'm wrong with these assumptions
Q: What do I connect SDA(on the rc522) to on my dev-board?
A: I used the pin next to miso mosi and clk which has RX in it's definition
Q: What is the CS PIN mentioned in the configuration
A: CS connectes to SDA
Q: Which pins can I use for 'reset' in the configuration
A: Any pin capable of sending (TX or regular)

For others testing
Create 'custom_components' folder next to yaml and then 'rc522' folder and place the [files](https://github.com/esphome/esphome/tree/add-rc522/esphome/components/rc522) into that.
Then custom_components isn't read by standard release versions of esphome, use [these instructions](https://esphome.io/guides/faq.html#how-do-i-use-the-latest-bleeding-edge-version) to install dev.

glmnet, 2020-09-25:
`custom_components` mechanism should work in v1.15 which was released a few weeks ago.

@blademckain  you really mean uart? or i2c? 

Thankgs @sgvj 

SDA and SCK are the labels for I2C but only SPI is implemented now. However, SPI needs: CS (or sometimes called SS), CLK, MISO, MOSI at least... in the RC522 board when SPI is used, SDA is CS and CLK is SCK

As for the pins to use on the ESP8266 / ESP32, ESPHome supports almost any pin for SPI, it's wise to always try to use the pins given in the examples

blademckain, 2020-09-28:
@glmnet yeah i mean really UART
i2c and spi need a lot of wire! 

my esp32 near the door is now full and I don't have so many free pins!!

blademckain, 2020-09-28:
@glmnet 
I was wrong!!!
i2c only needs 2 wires too so i2c might be fine for my needs too!
sorry

schmurtzm, 2020-09-28:
Hi, I use [the code from HausnerR](https://github.com/HausnerR/esphome-door-lock) since a little more than one year and I'm happy with it. Thanks @HausnerR  by the way ;)

I've made [this jukebox audio launcher for my childs with it](https://youtu.be/pSX4PrZCyGU?t=394).
Happy to see that it is going to be natively implemented in ESPhome  :smiley:

bernardovalente, 2020-09-30:
I'm having this errors when compiling using the custom component:

`INFO Reading configuration /config/esphome/rfid_entrada.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/rfid_entrada
Processing rfid_entrada (board: nodemcuv2; framework: arduino; platform: espressif8266@2.6.2)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
|-- <SPI> 1.0
Compiling /data/rfid_entrada/.pioenvs/rfid_entrada/src/main.cpp.o
Compiling /data/rfid_entrada/.pioenvs/rfid_entrada/lib4d9/ESP8266WiFi/WiFiServer.cpp.o
Compiling /data/rfid_entrada/.pioenvs/rfid_entrada/lib4d9/ESP8266WiFi/WiFiServerSecureAxTLS.cpp.o
Compiling /data/rfid_entrada/.pioenvs/rfid_entrada/lib4d9/ESP8266WiFi/WiFiServerSecureBearSSL.cpp.o
Compiling /data/rfid_entrada/.pioenvs/rfid_entrada/lib4d9/ESP8266WiFi/WiFiUdp.cpp.o
Compiling /data/rfid_entrada/.pioenvs/rfid_entrada/liba41/ESPAsyncTCP-esphome/AsyncPrinter.cpp.o
Compiling /data/rfid_entrada/.pioenvs/rfid_entrada/liba41/ESPAsyncTCP-esphome/ESPAsyncTCP.cpp.o
src/main.cpp: In lambda function:
src/main.cpp:270:28: error: expected type-specifier before 'RFIDRC522Sensor'
       auto my_sensor = new RFIDRC522Sensor();
                            ^
src/main.cpp:272:51: error: could not convert '{<expression error>, <expression error>}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::text_sensor::TextSensor*>'
       return {my_sensor->rfid, my_sensor->rfidlast};
                                                   ^
src/main.cpp:273:3: warning: control reaches end of non-void function [-Wreturn-type]
   });
   ^
*** [/data/rfid_entrada/.pioenvs/rfid_entrada/src/main.cpp.o] Error 1
========================== [FAILED] Took 2.12 seconds ==========================`

any thoughts?

glmnet, 2020-09-30:
can you post full yaml?

bernardovalente, 2020-10-01:
Thanks for replying. the device yaml right?

```
esphome:
  name: rfid_entrada
  platform: ESP8266
  board: nodemcuv2

# If buzzer is enabled, notify on api connection success
  on_boot:
    priority: -10
    then:
    - wait_until:
        api.connected:
    - logger.log: API is connected!
    - rtttl.play: "success:d=24,o=5,b=100:c,g,b"

# Define switches to control LED and buzzer from HA
switch:
- platform: template
  name: "${friendly_name} Buzzer Enabled"
  id: buzzer_enabled
  icon: mdi:volume-high
  optimistic: true

# Enable SPI interface
spi:
  clk_pin: D0
  miso_pin: D1
  mosi_pin: D2
  
rc522:
  cs_pin: D8
  reset_pin: D3

text_sensor:
  - platform: custom
    lambda: |-
      auto my_sensor = new RFIDRC522Sensor();
      App.register_component(my_sensor);
      return {my_sensor->rfid, my_sensor->rfidlast};
    
    text_sensors:
     - name: "RFID ID (tempo real)"
     - name: "RFID (última leitura)"

# Define the buzzer output
output:
- platform: esp8266_pwm
  pin: D7
  id: buzzer

# Define buzzer as output for RTTTL
rtttl:
  output: buzzer

wifi:
  ssid: "Vodafone-F6DC5E"
  password: "JxCTWFKdh4"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Rfid Entrada Fallback Hotspot"
    password: "WxUazZyzot9b"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "kVyNQEqakoaBh3"
  services:
  - service: rfidreader_tag_ok
    then:
    - rtttl.play: "beep:d=16,o=5,b=100:b"

  - service: rfidreader_tag_ko
    then:
    - rtttl.play: "beep:d=8,o=5,b=100:b"

  - service: play_rtttl
    variables:
      song_str: string
    then:
    - rtttl.play: !lambda 'return song_str;'

ota:
  password: "kVyNQEqakoaBh3"
```

bernardovalente, 2020-10-01:
sorry had to delete the previous response.

you mean the nodemcu yaml?

```
esphome:
  name: rfid_entrada
  platform: ESP8266
  board: nodemcuv2

# If buzzer is enabled, notify on api connection success
  on_boot:
    priority: -10
    then:
    - wait_until:
        api.connected:
    - logger.log: API is connected!
    - rtttl.play: "success:d=24,o=5,b=100:c,g,b"

# Define switches to control LED and buzzer from HA
switch:
- platform: template
  name: "${friendly_name} Buzzer Enabled"
  id: buzzer_enabled
  icon: mdi:volume-high
  optimistic: true

# Enable SPI interface
spi:
  clk_pin: D0
  miso_pin: D1
  mosi_pin: D2
  
rc522:
  cs_pin: D8
  reset_pin: D3

text_sensor:
  - platform: custom
    lambda: |-
      auto my_sensor = new RFIDRC522Sensor();
      App.register_component(my_sensor);
      return {my_sensor->rfid, my_sensor->rfidlast};
    
    text_sensors:
     - name: "RFID ID (tempo real)"
     - name: "RFID (última leitura)"

# Define the buzzer output
output:
- platform: esp8266_pwm
  pin: D7
  id: buzzer

# Define buzzer as output for RTTTL
rtttl:
  output: buzzer

wifi:
  ssid: "xxxxxxxx"
  password: "xxxxxxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Rfid Entrada Fallback Hotspot"
    password: "xxxxxxxxxxx"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "xxxxxxx"
  services:
  - service: rfidreader_tag_ok
    then:
    - rtttl.play: "beep:d=16,o=5,b=100:b"

  - service: rfidreader_tag_ko
    then:
    - rtttl.play: "beep:d=8,o=5,b=100:b"

  - service: play_rtttl
    variables:
      song_str: string
    then:
    - rtttl.play: !lambda 'return song_str;'

ota:
  password: "xxxxxxx"
```

Thuurke, 2020-10-02:
@sgvj can you post you're yaml file? Also a bit green here ;)

> For others testing
> Create 'custom_components' folder next to yaml and then 'rc522' folder and place the [files](https://github.com/esphome/esphome/tree/add-rc522/esphome/components/rc522) into that. [...]

sgvj, 2020-10-02:
@Thuurke. Short and sweet.
```
esphome:
  name: devicename
  platform: ESP32
  board: esp-wrover-kit

wifi:

spi:
  clk_pin: GPIO14
  miso_pin: GPIO12
  mosi_pin: GPIO13

rc522:
  cs_pin: GPIO27
  reset_pin: GPIO4
  update_interval: 1s
  on_tag:
    then:
      - homeassistant.tag_scanned: !lambda 'return x;'

captive_portal:
logger:
api:
ota:
```

Thuurke, 2020-10-02:
Oke nice. The logs show:

[D][rc522:157]: Found new tag 'xx-xx-xx-xx'

But I still don't get how this. Which device_id shoudl I use and where do I find it?

  trigger:
    - platform: event
      event_type: tag_scanned
      event_data:
        device_id: 178D8E87-2945-43C6-9507-xxxxxxxxxxx
        tag_id: 4e32a633-241d-4191-8325-0e197a9bb05b

> @Thuurke. Short and sweet.
>

sgvj, 2020-10-02:
What are you trying to do? Once Homeassistant got the tags on Configuration>Tags page I just create an automation by clicking the robot. My setup dosen't require knowing the specific reading device so that is blank.
The automation looks like:

```
- id: '1601067933665'
  alias: Tag Skytrain card is scanned
  description: ''
  trigger:
  - platform: tag
    tag_id: 04-CD-37-A2-9F-4F-80
  condition: []
  action:
  - service: media_extractor.play_media
    data:
      media_content_id: https://www.youtube.com/watch?v=69RdQFDuYPI&list=OLAK5uy_kbwoNt5t9bhtUTrWKIw5J5tqePslHIbmE
      media_content_type: audio/youtube
    entity_id: media_player.kitchen_speaker
  mode: single
```

glmnet, 2020-10-02:
@bernardovalente your yaml is wrong, for using this component you don't require any custom stuff, check out the sample provided a few post above: https://github.com/esphome/feature-requests/issues/37#issuecomment-702802360

glmnet, 2020-10-02:
> [D][rc522:157]: Found new tag 'xx-xx-xx-xx'

This just tells you that a tag with that ID was read, and you can use that id if you want ESPHome to react to that tag in a binary_sensor, a local automation. Please refer to the [pn532 docs](https://esphome.io/components/binary_sensor/pn532.html) for a detailed explanation, how to use with local automations or how to use with Home Assistant. The same applies for this component.

Thuurke, 2020-10-02:
> This just tells you that a tag with that ID was read, and you can use that id if you want ESPHome to react to that tag in a binary_sensor, a local automation. Please refer to the [pn532 [...]

Oke I get it, but I would prefer that the esphome device sends every tag to homeassistant and that the tag handling is done by home assistant. Editing homeassistant yamls is easier than "hardcode"/update the esphome device. But I will experiment with it.

Thuurke, 2020-10-03:
Oke I figured it out.
You can do both ways.

In the esphome device with binary sensors
binary_sensor:
  - platform: rc522
    name: tagname
    uid: xx-xx-xx-xx
or using the new tag system
  on_tag:
    then:
      - homeassistant.tag_scanned: !lambda 'return x;'

Nice!!

> Oke I get it, but I would prefer that the esphome device sends every tag to homeassistant and that the tag handling is done by home assistant. Editing homeassistant yamls is easier than [...]

bernardovalente, 2020-10-03:
@glmnet I managed to compile and read tags after some changes on the yaml and changed esphome to dev, thanks!

@Thuurke i can see the tags on the log, registered a new tag on HA but it never gets scanned by HA, do yours get updated?
```
esphome:
  name: rfid_entrada
  platform: ESP8266
  board: nodemcuv2

# If buzzer is enabled, notify on api connection success
  on_boot:
    priority: -10
    then:
    - wait_until:
        api.connected:
    - logger.log: API is connected!
    - rtttl.play: "success:d=24,o=5,b=100:c,g,b"

# Define switches to control LED and buzzer from HA
switch:
- platform: template
  name: "${friendly_name} Buzzer Enabled"
  id: buzzer_enabled
  icon: mdi:volume-high
  optimistic: true

wifi:
  ssid: "xxxxxx"
  password: "xxxxxx"
  
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Rfid Entrada Fallback Hotspot"
    password: "xxxxxxxxxx"
    
spi:
  clk_pin: GPIO14
  miso_pin: GPIO12
  mosi_pin: GPIO13

rc522:
  cs_pin: GPIO3
  reset_pin: GPIO0
  update_interval: 1s
  on_tag:
    then:
    - homeassistant.tag_scanned: !lambda 'return x;'
    - if:
        condition:
          switch.is_on: buzzer_enabled
        then:
        - rtttl.play: "success:d=24,o=5,b=100:c,g,b"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "xxxxxx"
  services:
  - service: rfidreader_tag_ok
    then:
    - rtttl.play: "beep:d=16,o=5,b=100:b"
  
  - service: rfidreader_tag_ko
    then:
    - rtttl.play: "beep:d=8,o=5,b=100:b"
  
  - service: play_rtttl
    variables:
      song_str: string
    then:
    - rtttl.play: !lambda 'return song_str;'

ota:
  password: "xxxx"

# Define the buzzer output
output:
- platform: esp8266_pwm
  pin: GPIO15
  id: buzzer

# Define buzzer as output for RTTTL
rtttl:
  output: buzzer
```

Thuurke, 2020-10-03:
Did you register the esphome device as an home assistant integration? configuration->integrations->+-sign->esphome. After this it should show the scanned tags at confugration->tags

bernardovalente, 2020-10-03:
It's now updating tags, my house door lock tag (nfc?) is not compatible with this reader though and is not registering.

thanks for the help @glmnet @Thuurke

closed 2020-10-26 by jesserockz