esphome/feature-requests #38: Add more IR remote codes

escoand, 2019-02-02:
I was looking for a way to send and receive IR codes to and from my Samsung tv. Currently esphomelib is not able to handle them but just the raw telegram.

But I found a description of the codes for  Samsung and even more manufacturer.
I can confirm the Samsung codes are as described for my concrete device.
http://www.techdesign.be/projects/011/011_waves.htm

Could they get implemented and maybe in a more simple way than the current one-manufacture-one-class way. Maybe as a simple config file.

josh929800, 2019-02-16:
I am using the IRMQTTServer from the same repository. The thing I like about this server is whatever you feed into it. The code will decode it and you can then re-transmit that same code. E.G. I could feed it a remote code from remote central in pronto format. It would send the code out and receive it and tell me what the IR code is.  The catch is you have to prefix your code with a number that represents the type of code. https://github.com/markszabo/IRremoteESP8266/tree/master/examples/IRMQTTServer

freddan-teamleader, 2019-03-25:
I'd like some implementation like https://github.com/ToniA/arduino-heatpumpir

jay-wilkinson, 2019-05-26:
@josh929800 Thanks for that link. It appears to be a short term solution to a problem I have.
@OttoWinter I'd like to suggest and vote for the Pronto protocol.
Technical info: http://www.remotecentral.com/features/irdisp2.htm
Codes Database: http://files.remotecentral.com/library/3-1/index.html
The reason I need this is I'm trying to control an RCA 5.1 Receiver. RCA doesn't use the same modulation frequency as almost every other device on the planet. They used a carrier frequency of 56kHz. And of course any IR LED will work fine but the IR receiver has to be made for 56kHz. Without the correct receiver, I can't capture the codes in RAW format. The codes exist in Pronto and I've played with them using IRremoteESP8266 and they do work. Implementing this protocol would create a bridge to a lot of very old (10 - 20+ years) devices.
RCA IR info: https://www.sbprojects.net/knowledge/ir/rca.php

aenertia, 2019-06-29:
Upstream IRemoteESP8266 now has full support for ESP32 - including things like Daikin etc from there should now be relatively trivial.

OttoWinter, 2019-06-29:
@aenertia Yes, but that has never been the problem - ESPHome has support for ESP8266/ESP32 already and a pretty good API for sending/receiving codes. The individual protocols just need to be implemented.

Ariskayler, 2019-06-29:
Yeah. That would be so nice!

JoshuaSG, 2019-07-12:
Would it be difficult to implement the GlobalCache protocol? It's a nice universal code format that is easily searchable. It is implemented for the ESP8266 here on IRremoteESP8266 https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_GlobalCache.cpp

subzero79, 2019-08-01:
Is it possible to add LG2 to the available protocols. It was already in IRremoteESP8266 as you can see

https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_LG.cpp

OttoWinter, 2019-10-13:
I don't it will ever be possible to make something generic enough to encapsulate all protocols - so I will close this issue. All feature requests for specific protocols should go into separate feature requests per protocol so that contributors can coordinate efforts better.

closed 2019-10-13 by OttoWinter