esphome/feature-requests #41: MCP 23008

jaimehall63, 2019-02-02:
I see esphomelib already supports the pcf8574 I/O expander. I am working on a project that uses an MCP23008 I/O expander. Do you know if they are similar enough to just modify the existing code of the pcf8574 to use with the MCP23008? I am woking with an NCD.io 2 relay board that uses an ESP8266 communication module to talk to the MCP23008 on the board over I2C.

papperone, 2019-03-07:
MCP23008/23017 are easy to handle and I think they shoudl be part of core expanders of ESPHome, they are quite diffused and utilized in many applications, same protocol and PCF8574 (I2C) but up to 16 GPIOs
👍

OttoWinter, 2019-03-08:
This has already been implemented here https://github.com/esphome/esphome/pull/466

closed 2019-03-08 by OttoWinter

Tuckie, 2019-03-21:
@OttoWinter with that implementation work with the MCP23008 as well (if I don't exceed the pin count of the MCP23008)?

TomFahey, 2019-06-07:
No, the MCP23008 doesn't work with the current implementation, as if you check the spec sheets for both chips ([MCP23017](http://ww1.microchip.com/downloads/en/DeviceDoc/20001952C.pdf), [MCP23008](http://ww1.microchip.com/downloads/en/DeviceDoc/MCP23008-MCP23S08-Data-Sheet-20001919F.pdf) - pages 12, 4 respectively), you can see that the address registers are laid out differently due to there being twice the number of ports on the 23017 e.g 01h is IPOL for the 23008, but IODIRB for the 23017.

I've taken a shot myself at creating component files for the 23008 by tweaking those of the 23017, however I'm not really sure how to test it - can you do that as a custom component, as the docs only mention creating custom components using header files?

TomFahey, 2019-06-08:
@OttoWinter Ok, so I tested my tweaked version as a custom component and it seems to be working fine. I've tested all 8 pins in INPUT_PULLDOWN mode and the even pins in OUTPUT mode (don't have that many LEDs lying around...), didn't have any problems, should I do a PR?

OttoWinter, 2019-06-09:
@TomFahey Sure, that would be nice.