esphome/feature-requests #42: Template cover SET_COVER_POSITION support

brandond, 2019-02-02:
Do you have any plans to support [setting cover position](https://www.home-assistant.io/components/cover/#service-coverset_cover_position) for template covers? Additionally, is there a good way to tie a cover position sensor to the position of a slider in the Home Assistant UI? It seems like existing options require the sensor/slider to be pretty much separate from the cover and its open/closed state and open/stop/close buttons.

nevalain, 2019-02-06:
I would also like to use the set_cover_position in template cover.

I am using generic MG 946R servos with blinds. I assume that I need simple set_cover_position and I could calculate the PWM (with lamda)?

CTLS, 2019-03-12:
Would love to be able to use the tilt function with esphome.

PierreScerri, 2019-03-14:
I also have 11 roller shutters in the house and it would be good to have some form of position/set_positon functionality.

Thank you.

norges, 2019-04-10:
I try to use the cover component with my roller shutters at the moment and would love the possibility to have a position. Best case for me would be in percent wit a max/min value.

blademckain, 2019-04-15:
also for me it would be the maximum to be able to indicate the percentage of opening of the cover
++

OttoWinter, 2019-04-15:
Closed in https://github.com/esphome/esphome/pull/496

closed 2019-04-15 by OttoWinter