esphome/feature-requests #45: [Feature Request] nRF24L01 support

brandond, 2019-02-02:
I have a use case for some small battery-powered distributed sensors. Ideally this would be a good use for BLE or [Z-Wave](http://www.z-wave.com/shop-z-wave-smart-home-products/category/smart-sensors), but BLE support has a ton of overhead and Z-Wave is [proprietary and expensive](https://www.amazon.com/dp/B01IRBDBZY).

What I'm probably going to end up doing is setting up a few devices with STM32s that will spend most of their time in suspend, occasionally waking to make a sensor reading and send it off via a [nRF24L01](https://platformio.org/lib/show/433/RF24) radio if conditions are met. On the ESP side, it would just be a matter of wiring up the RF24 SPI/Enable/IRQ pins to GPIOs. 

 The chip is limited in hardware to 32 byte message payloads. Perhaps something like this:
```c
struct payload {
  unit8_t type;
  char unit[3];
  char value[14];
  char name[14];
};
```

Ideally, esphomelib would not need to define a list of sensors ahead of time, but simply reformat and pass the messages through to MQTT.

nekromant, 2020-05-31:
Hm, this would be useful. I happen to have my own pure-C optimized library for nrf24l01 (basically @ManiacBug 's arduino ported to pure C. I'll see if it compiles for esp8266 (it should). But it's pretty lowlevel.

diogosalazar, 2020-06-09:
The mysensors/MySensors project handles communication between nrF24l01 modules and seems to be able to register devices dynamically.
ESP8266 is supported as a gateway.

Would it make sense to integrate with that project instead of implementing from scratch?

nekromant, 2020-06-09:
As far as I got the nRF24L01 part, we have a huge bunch of hardware and stacks:
- Dumb sensors/boards using simple protocols. As simple as send/receive a predefined payload. (Should be easiest). 
- rf24mesh devices. rf24mesh doesn't specify any protocol, it's just read/write, no device method discovery/enumeration.
- mysensors that @diogosalazar mentioned. (That one is new to me, I'll have to check it out)
- My own libaura RPC over nRF24L01 ( https://gitlab.com/ncrmnt/aura-transport-rf24 ). 

Some nodes have encryption (e.g. car immobilizer tokens),  and that's a whole other layer of weirdness. I really want to help out with this task and have quite a lot of experience with nRF24L01 (non-arduino, pure C), but I don't know the inner workings of esphome good enough.

glmnet, 2020-06-11:
I discovered ESPHome because I hated MySensors, their code quality seems ok, but these things communicating never was reliable for me, probably I was doing something wrong, I don't know.

As for today I believe this FR is outdated, I don't see need for dynamic sensor creation, though it is possible, in fact it is done already, you can just pick the message analyze it and push to any template sensor / binary sensor / cover... whatever, if that is what you need.

@brandond  What ended happening here?

nekromant, 2020-06-17:
@glmnet I just learned about mysensors from this thread and rigged up a test mesh based on my older projects. (Yay, they're no longer sitting in the attic!). So far it looks like the best we have for nRF24L01. As for being outdated - I doubt that. nRF24L01 still provides better battery life for devices like leak sensors, compared to esp8266 and doesn't have a severely over-engineered std, like BLE. As for esphome, I think the only thing we can add - integrating a mysensors gateway in a few lines of yaml. For some cases this may be useful.

glmnet, 2020-06-17:
It’s a great platform. It did not work for me. 

I just don’t know why would you use it within an esphome controller node. They provide a esp8266 gateway. That is you can connect hass to it then to my sensors network. It’s fully supported and documented.

nekromant, 2020-06-20:
@glmnet  I guess you're right. The only thing I can think of - combining esphome node and mysensors gateway. As for non-mysensors protocols - perhaps nrf24l01 beacons and presense sensors come to my mind, but still with something like mysensors around, the whole idea of integrating nrf24l01 into esphome sounds like a waste of time (

Niorix, 2022-06-04:
Any news? I have an esphome-based LED dimmer and I would have controlled it with a touch R11 remote controler (https://aliexpress.ru/wholesale?catid=&Searchtext=R11%20Remote). It's apparently uses 2.4. NRF. Therefore, I cannot directly connect him with Esphome((

kluszczyn, 2022-08-01:
> The only thing I can think of - combining esphome node and mysensors gateway.

See: https://github.com/mannkind/ESPHomeMySensorsGatewayShim

nekromant, 2022-08-06:
> > The only thing I can think of - combining esphome node and mysensors gateway.
> 
> See: https://github.com/mannkind/ESPHomeMySensorsGatewayShim

Thanks a ton! That's precisely what I've been looking for!

otmezger, 2022-09-19:
I would love to be able to write yaml and leverage a nrf24l01 wireless network. let esphome do all the job ;-)

jamesmyatt, 2022-09-20:
I wonder if this could be modelled on [`bluetooth_proxy`](https://esphome.io/components/bluetooth_proxy.html) and [BTHome ](https://bthome.io/).

3esmit, 2022-11-09:
I need nRF24L01 support for integrating a device that uses a remote control on 2.4GHz. I have captured the signals and I can send it using HackRF, but for integrating with home assistant would be nice to have this support.

jhogendorn, 2022-11-16:
> I need nRF24L01 support for integrating a device that uses a remote control on 2.4GHz. I have captured the signals and I can send it using HackRF, but for integrating with home assistant would be nice to have this support.

I am also keen to get this use case working

zackfuchtel, 2023-08-03:
[x] AOL

sudipta26889, 2024-07-28:
ESPHome core doesn't have nRF24L01 support yet. right?

Do we have any external component?

derGreyson, 2024-09-22:
Is there any option to fix this manual?

AdamM68, 2025-01-08:
I'm still waiting for nRF24L01 support in esphome and home assistant.
Do Yoy know any working solution ?

Lukanex, 2025-03-20:
Hi, I would also like to use nRF24L01 or RMF73 with Home Assisntant (preferrably via ESPHome).
Any reccomendations, please?

I have looked at the [MySensors ](https://www.mysensors.org/) platform and it coveres some of my use-cases and has integration with HA. The need to manage another code base is one of the down-sides for me.

jamesmyatt, 2025-03-20:
What do you imagine nRF24L01 support in ESPHome to look like? nRF24L01 is just the radio, i.e. the physical layer.

Lukanex, 2025-03-20:
> What do you imagine nRF24L01 support in ESPHome to look like? nRF24L01 is just the radio, i.e. the physical layer.

That is a very good question.
For my first purpose: to be able to setup a custom network with one gateway and multiple sensors. Star, mashing not required.

But, now that I am thinking about it, it would be best to adhere to some standard protocol to really bring a value to the ESPHome users, right? I looked up that Enhanced ShockBurs (ESB) is common amongst hobbyists.

I am on the fence with MySensors. They support [ESP8266 and ESP32](https://github.com/mysensors/MySensors/tree/development/hal/architecture), none of the newer chips.
It would be super convenient to manage the MySensors nodes via the ESPHome interface :)

jhogendorn, 2025-03-20:
> What do you imagine nRF24L01 support in ESPHome to look like? nRF24L01 is just the radio, i.e. the physical layer.

My interest is in replacing RF based remotes, although this is more complex than just rf cloning.

nekromant, 2025-03-20:
Since github's pinging me about the issue, let me throw my 2 cents in: 
 
@jhogendorn These seldom use nrf24l01 - too expensive for that purpose, mostly some dumb 433 or similar transiever ICs. These are often used in keyboards/mouses - but getting those supported in esphome is (apart from being a fun pasttime) doesn't really have any use. 

In terms of mysensors - as @kluszczyn noticed, https://github.com/mannkind/ESPHomeMySensorsGatewayShim gets most of the job done. Perhaps some management via web ui might be useful, but that (IMO) would have to be done outside esphome. 

Another idea I had flying around - implement a esphome-like tool for building mysensors nodes(Man,. I wish I had the time!). But that's totally out of scope of this project.

jhogendorn, 2025-03-20:
> [@jhogendorn](https://github.com/jhogendorn) These seldom use nrf24l01 

Nah, i have a bunch of remotes that are utilising 2.4ghz chipsets, i've tried doing some sniffing on them with a hackrf but i'm somewhat out of my depth.

nekromant, 2025-03-20:
@jhogendorn For that case nrf24l01 component will make sense. 

Check out this hack https://github.com/Yveaux/NRF24_Sniffer This may be a good start for you

loucks1, 2026-04-25:
I have copied the nrf24/RF24 driver into an Esphome external component - I had to implement this to get my directolor solution to compile using esp-idf instead of the arduino I had been using.  I can read and send, so I think it is starting to get to a point where I can share it.  If someone has time to help me with it, I think we could whip it into shape pretty quickly.  I tried to keep it as close to nrf24/RF24 as possible.  The key point here is that I'm using the esphome spi and spi_device.  I would like to run a test with it use nrf24/RF24 to make sure that I can write and send to something using the official library.  If you want to help, reach out to me directly.  I think it is a long way from being ready to be included officially in esphome.