esphome/feature-requests #48: L298 Dual H-Bridge DC Motor Controller

Qu3uk, 2019-02-02:
**What new integration would you wish to have?**

L298 Dual H-Bridge DC Motor Controller
https://www.sparkfun.com/products/9670
**If possible, provide a link to an existing library for the integration:**
https://github.com/AndreaLombardo/L298N ?
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
This is what I have and rather not buy something else to convert an old project etc
**Additional context**

workingmanrob, 2019-04-11:
Hi guys,

I think I am also interested in this integration.  I have recently purchased a few nodemcu motor shield boards and I'm pretty sure it's the same chip used.  Oh I have 293 instead but I assume it's quite similar. 

https://hackaday.io/project/8856-incubator-controller/log/29291-node-mcu-motor-shield

The shield is based on the L293D quadruple high-current half-H drivers typically used to drive motors.

This is supposed to support either a single stepper motor or two dc motors.  I'm just trying to use a single dc motor.  Seems to be a popular kit for making a wifi remote controlled car.

https://www.instructables.com/id/Motorize-IoT-With-ESP8266/

Will try that snippet from Otto and see if things move.

Thanks again!

Oh found this - http://drzzs.com/diy-window-shades-motorized-automated/

I really just want motorized curtain control..

danpowpow, 2019-08-06:
Hi @OttoWinter ! 
I absolutely love esphome - well done on such a great piece of work :-)
has there been any development on this? i'm looking at using hbridges to give me dimming control of some polarity-reversing led string lights. I have code working on arduino, but am hoping to move to esphome for better hass integration.
the code i'm using uses three gpio from the arduino going to the L298N modulem, IN1 IN2 and PWM. i've repurposed code from here:
https://tronixlabs.com.au/news/tutorial-l298n-dual-motor-controller-module-2a-and-arduino/
thanks again for esphome!

RIKWEBB, 2019-08-23:
Any integration planned for the L293D?
This would help create an easy solution for Shades / Blinds with these 
ARCELI L293D Wifi Motor Drive Expansion Board Shield Module For NodeMCU ESP-12E ESP8266
by QC EU
Learn more: https://www.amazon.co.uk/dp/B078CPZDW1/ref=cm_sw_em_r_mt_dp_U_OycyDbXBX2GX4

OttoWinter, 2019-08-27:
@danpowpow See the response I posted above.

@RIKWEBB Looks like the L293D is essentially the same as the L298, so also look at said post.

danpowpow, 2019-08-28:
thanks for the response @OttoWinter :-)
i'm not sure i completely follow you though. I understand ledc for pwm on the esp32, and have succeeded where that is concerned. the main issue i'm facing however is that in order to get my led festoon to light all the bulbs i have to keep switching the polarity very fast to the in1 and in2 pins of the L298N. not polarity reversing results in only 1/2 of the string working, as each led is wired up in reverse to that last one. the arduino code is dead easy:
  digitalWrite(in1, LOW);
  digitalWrite(in2, HIGH);  
  digitalWrite(in1, HIGH);
  digitalWrite(in2, LOW); 
but i can't quite see how to run this bit (or equivalent) in esphome. i've tried using the loop() iteration but it isn't fast enough.
Is esphome the right tool for this job?
thanks again,
dp

cnctech, 2020-01-09:
I hope that I am asking this question in the right place. It's my first time using this.
I have HASSIO up and "somewhat" running, and have been looking into a good closed loop DC motor control. After experimenting with EspHome, I feel that this is one of the most advanced and flexible options to do much control. Has anyone tried using the L298 with a DC gear motor with encoder, and using the "Rotary Encoder" function in EspHome, along with "Cover" templates to set "Home" for position? I do not want to use stepper motors.

glmnet, 2020-01-10:
> I hope that I am asking this question in the right place. It's my first time using this.
> I have HASSIO up and "somewhat" running, and have been looking into a good closed loop DC motor control. [...]

Certainly you can use ESPHome here, I haven't tried using the rotary encoder to control the position but that sounds clever, you'll have to implement the closed loop yourself as ESPHome does not provide any PID algorithm yet.

shafr, 2020-03-27:
It took me 2 years to wrap my head around this thing (Mostly I was afraid to drill my window frame and was waiting for correct 3d model, PCB was ready and waiting).

```
# IN1 = 16
# IN2 = 12
# IN3 = 15
# IN4 = 01

stepper:
  - platform: uln2003
    id: my_stepper
    pin_a: 12 
    pin_b: 16
    pin_c: 01
    pin_d: 15
    max_speed: 250 steps/s
    sleep_when_done: true

    # Optional:
    acceleration: inf
    deceleration: inf

switch:
  - platform: gpio
    pin: 02
    id: red_pin
  
  - platform: template
    name: "Set position 250"
    turn_on_action:
      - stepper.set_target:
            id: my_stepper
            target: 250

  - platform: template
    name: "Set position -250"
    turn_on_action:
      - stepper.set_target:
            id: my_stepper
            target: -250     
```

1) First of all - pin order here is **in1,in3, in2,in4**. 
2) `sleep_when_done` is crucial! Or your motor will overheat.
3) Consumption - idle 0.09 A, with motor spinning - 0.3 A.

shafr, 2020-04-12:
Well it worked with exactly one chip. 
I'm uploading the exact same code to another and motors have no torque at all. That is very very confusing.
I have no idea what's the problem.

glmnet, 2020-04-15:
@shafr this FR is about the L298 Dual H Bridge which is a driver not suitable for the 28BYJ-48 motor
The driver is meant for a **standard DC** motor which can be powered directly or reversed so the turn direction changes. The motor you are using in contrast is a **stepper** motor, hence you're using the stepper integration. I don't know what will happen if you try to use the L298 with the stepper, but it will not work well for sure.

wmnl25, 2020-12-20:
> Yes, that integration is planned - and it wouldn't even be too hard to integrate.
> The issue is that a whole "motor" abstraction layer would have to be created first and that's a lot of work since there are so many types of motors.
> For now, you can also control it using the PWM components. For example
> 
> ```yaml
> output:
> - platform: gpio
>   pin: D1
>   id: direction_pin
> - platform: esp8266_pwm  # or ledc if on ESP32
>   pin: D2
>   id: speed_pin
> 
> # In some call:
> # Set to forward dir
> - output.turn_on: direction_pin
> # Turn on motor
> - output.set_level:
>     id: speed_pin
>     level: 100%
> # Let motor drive for a big
> - delay: 1s
> # Turn off motor
> - output.turn_off: speed_pin
> # Repeat for Oóther direction, this time with turning off the direction pin
> ```

Can confirm that the L298 works fine this way. I use it to drive the lid of a (now smart) simplehuman bin. I adjusted the code a little bit so the lid can both be opened and closed. Shows up as cover in HA.

`output:
  - platform: esp8266_pwm
    pin: D5
    frequency: 1000 Hz
    id: sh_speed_pin
  - platform: gpio
    pin: D7
    id: sh_direction_pin1
    inverted: true
  - platform: gpio
    pin: D8
    id: sh_direction_pin2
    inverted: true

cover:
  - platform: template
    name: "Simplehuman bin"
    id: lid
    open_action:
      - light.turn_on: 
          id: ws2811led
          effect: strobe
      - output.turn_on: sh_direction_pin2
      - output.turn_on: sh_speed_pin
      - delay: 2s
      - output.turn_off: sh_direction_pin2
      - output.turn_off: sh_speed_pin
    close_action:
      - output.turn_on: sh_direction_pin1
      - output.turn_on: sh_speed_pin
      - delay: 200ms
      - output.turn_off: sh_direction_pin1
      - output.turn_off: sh_speed_pin
      - delay: 1s
      - light.turn_off: ws2811led
    stop_action:
      - output.turn_off: sh_direction_pin1
      - output.turn_off: sh_direction_pin2
      - output.turn_off: sh_speed_pin
    optimistic: true`

closed 2021-08-30 by jesserockz