esphome/feature-requests #49: New integration request: PZEM-004

rradar, 2019-02-02:
**What new integration would you wish to have?**
- Peacefair PZEM-004
- Peacefair PZEM-004T (same, same but without housing and display)

**If possible, provide a link to an existing library for the integration:**
- https://github.com/olehs/PZEM004T
- https://halfforked.wordpress.com/2018/05/18/low-cost-house-power-monitoring-with-pzem-004-for-home-assistant/
- http://pdacontrolen.com/electricity-consumption-meter-peacefair-pzem-004-esp8266-arduino-nano/
- https://www.instructables.com/id/Power-Peacefair-PZEM-004-ESP8266-Arduino-Nano/

**Product link:**
- https://www.aliexpress.com/item/ANENG-4in1-Digital-Ammeter-Voltmeter-Watt-Power-Energy-Meter-AC-80-260V-100A-PZEM-004/32847092493.html

**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I can't use esphomelib for my integrations :cry: 

**Additional context**
Want to measure the total energy consumption/load of a one-phase main connection in real time with esphomelib/yaml :+1:

MariusLaurinaitis, 2019-03-18:
Hello, any updates on this?
I've been testing the PZEM with ESPurna using hardware serial works just fine. 
xoseperez (Espurna) commented: Using PZEM004T with SoftwareSerial is not working. Actually, using an oscilloscope shows that the PZEM is not answering back when using software serial... not sure the reason, but it does when using hardware serial...

OttoWinter, 2019-03-19:
No update, I'm concentrating on the core of esphome now and I don't have this piece of hardware, so I'm not actively working on it.

The software serial is a different issue - what i can say is that esphome rolls its own software serial implementation that is somewhat more efficient and has better timing. To the pzem the hard/soft serials look the same, unless the soft serial has timing issues.

MariusLaurinaitis, 2019-03-19:
I can send you a device as a gift. I have a few of them! That would be my thanks :)

bpmittal, 2019-03-24:
i also need to use this energy meter with esphome , is that possible , any update possible ?

rradar, 2019-04-12:
If still needed I also could donate one!

SeByDocKy, 2019-05-18:
Same .... definitively need a ESPhome integration since Tasmota don't allow teleperiod <7s without freezing the serial port between.

rradar, 2019-05-31:
Sorry @SeByDocKy but what is teleperiod? Espurna also supports the pzemm-004 (can catch a binary here for the nodemcu/esp8266 https://github.com/mcspr/espurna-nightly-builder/releases/) but still relies on mqtt and can't make advantages of the great esphome api.

Anyway I'm with you that ESPhome definitely needs this integration :stuck_out_tongue:

SeByDocKy, 2019-05-31:
> Sorry @SeByDocKy but what is teleperiod? Espurna also supports the pzemm-004 (can catch a binary here for the nodemcu/esp8266 https://github.com/mcspr/espurna-nightly-builder/releases/) but still [...]

teleperiod is the time period between two telemetry packet in JSON format to be sent to the MQTT broker. With tasmota... if your tweek the original code to allow teleperiod to be under 10s, in pratice for below 7s, the serial link is freezing.... I am pretty sure it's not coming from the PZEM-004T side but more from tasmota.... I dream to have the ESPhome PZEM integration.... Well first I want to be sure than the serial communication with ESPhome allow such sampling rate

rradar, 2019-05-31:
> teleperiod is the time period between two telemetry packets in JSON format to be sent to the MQTT broker. With tasmota... if your tweak the original code to allow teleperiod to be under 10s, in pratice for below 7s, the serial link is freezing.... I am pretty sure it's not coming from the PZEM-004T side but more from tasmota.... I dream to have the ESPhome PZEM integration.... Well first I want to be sure than the serial communication with ESPhome allow such sampling rate

I guess the limit is only given by the pzem-004(t) then. ESPhome software serial is capable doing almost anything - for now just pzem004 support is missing :wink:

SeByDocKy, 2019-07-03:
> > teleperiod is the time period between two telemetry packets in JSON format to be sent to the MQTT broker. With tasmota... if your tweak the original code to allow teleperiod to be under 10s, in pratice for below 7s, the serial link is freezing.... I am pretty sure it's not coming from the PZEM-004T side but more from tasmota.... I dream to have the ESPhome PZEM integration.... Well first I want to be sure than the serial communication with ESPhome allow such sampling rate
> 
> I guess the limit is only given by the pzem-004(t) then. ESPhome software serial is capable doing almost anything - for now just pzem004 support is missing 😉
> 
> EDIT: @SeByDocKy I can tell that a interval of 1 second requesting data (software serial) from the pzem004t and sending it by mqtt (also a 1 second) interval works with espurna.

It's stable ? after few hours ? days ? coz for me it's freezing after 3-4h ....

tsunglung, 2019-08-13:
[pzemac.diff.zip](https://github.com/esphome/feature-requests/files/3495113/pzemac.diff.zip)

the attached file is for support PZEM-004T V3.0. 
Already running for several days at my side.
If someone can help test, it will be great.

yaml conf
`
uart:
    tx_pin: D2
    rx_pin: D1
    baud_rate: 9600

sensor:
  - platform: pzemac
    energy:
      name: PZEM004T positive 110V Energy
    frequency:
      name: PZEM004T positive 110V Frequency
    voltage:
      name: PZEM004T positive 110V Voltage
    current:
      name: PZEM004T positive 110V Current
    power:
      name: PZEM004T positive 110V Power
      id: PZEM004T_P110_POWER
    update_interval: 1s
`

xppx99, 2019-08-14:
I would like to test it, but I don’t know what to do with the attachment you provided. Is there any guide on YouTube or wiki?

Thanks

tsunglung, 2019-08-15:
[pzemac.zip](https://github.com/esphome/feature-requests/files/3503962/pzemac.zip)

unzip it and put it to site-packages folder, like [python install folder]/lib/python3.7/site-packages/esphome/components/

yaml config
`uart:
    tx_pin: D2
    rx_pin: D1
    baud_rate: 9600

sensor:
  - platform: pzemac
    energy:
      name: PZEM004T Energy
    frequency:
      name: PZEM004T Frequency
    voltage:
      name: PZEM004T Voltage
    current:
      name: PZEM004T Current
    power:
      name: PZEM004T Power
      id: PZEM004T_POWER
    update_interval: 1s
`

SeByDocKy, 2019-08-15:
Is it possible to monitor the power factor, active and reactive power?

tsunglung, 2019-08-15:
PZEM-004T can monitor Power Factor, and active power.

SeByDocKy, 2019-08-15:
> PZEM-004T can monitor Power Factor, and active power.

I mean to add these output in the current as phone package

SeByDocKy, 2019-08-17:
Other important. Feature would be to monitor up to 4 pzem with the same esp8266

tsunglung, 2019-08-17:
> current as phone package

I did not get your point. what phone package?
 

> Other important. Feature would be to monitor up to 4 pzem with the same esp8266

I connect to 2 pzem. NodeMCU works fine.
for example. 
if use NodeMCU, NodeMCU have 9 GPIOs  and 2 Hardware UART( one is only have TX), 
4 software serail and 1 hardware uart can be used to connect 5 pzem

SeByDocKy, 2019-08-17:
> I did not get your point. what phone package?
>
>
> I connect to 2 pzem. NodeMCU works fine.
> for example.
> if use NodeMCU, NodeMCU have 9 GPIOs and 2 Hardware UART( one is only have TX),
> 4 software [...]

I. To add power factor, reactive power as new outputs of your esphome integration.
II. Can you give a yaml example with 2 pzem

tsunglung, 2019-08-17:
`uart:
  - id: pzem004tp110
    tx_pin: D2
    rx_pin: D1
    baud_rate: 9600
  - id: pzem004tn110
    tx_pin: D6
    rx_pin: D5
    baud_rate: 9600

time:
  - platform: sntp
    id: my_time

switch:
  - platform: uart
    uart_id: pzem004tp110
    name: "PZEM004T positive 110V Reset Energy"
    data: [0x01, 0x42, 0x80, 0x11]
  - platform: uart
    uart_id: pzem004tn110
    name: "PZEM004T negative 110V Reset Energy"
    data: [0x01, 0x42, 0x80, 0x11]

sensor:
  - platform: pzemac
    energy:
      name: PZEM004T positive 110V Energy
    frequency:
      name: PZEM004T positive 110V Frequency
    voltage:
      name: PZEM004T positive 110V Voltage
    current:
      name: PZEM004T positive 110V Current
    power:
      name: PZEM004T positive 110V Power
      id: PZEM004T_P110_POWER
    update_interval: 1s
    uart_id: pzem004tp110
  - platform: pzemac
    energy:
      name: PZEM004T negative 110V Energy
    frequency:
      name: PZEM004T negative 110V Frequency
    voltage:
      name: PZEM004T negative 110V Voltage
    current:
      name: PZEM004T negative 110V Current
    power:
      name: PZEM004T negative 110V Power
      id: PZEM004T_N110_POWER
    update_interval: 1s
    uart_id: pzem004tn110
  - platform: total_daily_energy
    name: "Total Daily positive 110V Energy"
    power_id: PZEM004T_P110_POWER
  - platform: total_daily_energy
    name: "Total Daily negative 110V Energy"
    power_id: PZEM004T_N110_POWER
`

SeByDocKy, 2019-08-18:
Thanks a lot very usefull

tsunglung, 2019-08-22:
an example to have 5 software serial (UART)

```
uart:
  - id: pzem004t1
    tx_pin: D3
    rx_pin: D1
    baud_rate: 9600
  - id: pzem004t2
    tx_pin: D4
    rx_pin: D2
    baud_rate: 9600
  - id: pzem004t3
    tx_pin: D6
    rx_pin: D5
    baud_rate: 9600
  - id: pzem004t4
    tx_pin: D0
    rx_pin: D7
    baud_rate: 9600
  - id: pzem004t5
    tx_pin: GPIO1
    rx_pin: GPIO3
    baud_rate: 9600

switch:
  - platform: restart
    name: "Power Monitor Restart"
  - platform: uart
    uart_id: pzem004t1
    name: "PZEM004T Channel 1 Reset Energy"
    data: [0x01, 0x42, 0x80, 0x11]
  - platform: uart
    uart_id: pzem004t2
    name: "PZEM004T Channel 2 Reset Energy"
    data: [0x01, 0x42, 0x80, 0x11]
  - platform: uart
    uart_id: pzem004t3
    name: "PZEM004T Channel 3 Reset Energy"
    data: [0x01, 0x42, 0x80, 0x11]
  - platform: uart
    uart_id: pzem004t4
    name: "PZEM004T Channel 4 Reset Energy"
    data: [0x01, 0x42, 0x80, 0x11]
  - platform: uart
    uart_id: pzem004t5
    name: "PZEM004T Channel 5 Reset Energy"
    data: [0x01, 0x42, 0x80, 0x11]

sensor:
  - platform: pzemac
    energy:
      name: PZEM004T Channel 1 Energy
    frequency:
      name: PZEM004T Channel 1 Frequency
    voltage:
      name: PZEM004T Channel 1 Voltage
    current:
      name: PZEM004T Channel 1 Current
    power:
      name: PZEM004T Channel 1 Power
      id: PZEM004T_CHANNEL1_POWER
    update_interval: 1s
    uart_id: pzem004t1
  - platform: pzemac
    energy:
      name: PZEM004T Channel 2 Energy
    frequency:
      name: PZEM004T Channel 2 Frequency
    voltage:
      name: PZEM004T Channel 2 Voltage
    current:
      name: PZEM004T Channel 2 Current
    power:
      name: PZEM004T Channel 2 Power
      id: PZEM004T_CHANNEL2_POWER
    update_interval: 1s
    uart_id: pzem004t2
  - platform: pzemac
    energy:
      name: PZEM004T Channel 3 Energy
    frequency:
      name: PZEM004T Channel 3 Frequency
    voltage:
      name: PZEM004T Channel 3 Voltage
    current:
      name: PZEM004T Channel 3 Current
    power:
      name: PZEM004T Channel 3 Power
      id: PZEM004T_CHANNEL3_POWER
    update_interval: 1s
    uart_id: pzem004t3
  - platform: pzemac
    energy:
      name: PZEM004T Channel 4 Energy
    frequency:
      name: PZEM004T Channel 4 Frequency
    voltage:
      name: PZEM004T Channel 4 Voltage
    current:
      name: PZEM004T Channel 4 Current
    power:
      name: PZEM004T Channel 4 Power
      id: PZEM004T_CHANNEL4_POWER
    update_interval: 1s
    uart_id: pzem004t4
  - platform: pzemac
    energy:
      name: PZEM004T Channel 5 Energy
    frequency:
      name: PZEM004T Channel 5 Frequency
    voltage:
      name: PZEM004T Channel 5 Voltage
    current:
      name: PZEM004T Channel 5 Current
    power:
      name: PZEM004T Channel 5 Power
      id: PZEM004T_CHANNEL5_POWER
    update_interval: 1s
    uart_id: pzem004t5
```

casio1964, 2019-08-27:
> [pzemac.zip](https://github.com/esphome/feature-requests/files/3503962/pzemac.zip)
>
> unzip it and put it to site-packages folder, like [python install [...]

Wher should i put these files in hassio?

fake-name, 2019-09-01:
Here's a config + c++ files for 3 PZEM-004T V3.0 + 1 PZEM-017 on a open-collector serial bus. 

[ICScraper.zip](https://github.com/esphome/feature-requests/files/3563010/ICScraper.zip)

It's a fair bit cleaner then @tsunglung's implementation, mostly because it just uses [modbusmaster](https://github.com/4-20ma/ModbusMaster) to handle all the communications. 

Also my wifi password, I guess. Whoops?

timido, 2019-09-19:
Wher should i put these files in hassio?

fake-name, 2019-09-20:
It's unrelated to hassio (well, maybe you can do it? There's an extension or something, but I just used the command line because I don't want to use hassio in the first place). 

You need a build environment that can [compile ESPHome](https://esphome.io/), then you edit the yaml file for your sensor config, and just do `esphome <filename>.yaml run`.

timido, 2019-09-20:
Good night I can not at all by working the PZEM-004 in hash with esphome

hannesbe, 2019-09-21:
> Wher should i put these files in hassio?

I'm also looking for a way to test this using hassio. The only way I know to run something custom, is using the dev version of the hassio addon, by specifying a branch name in the addon's config like `"esphome_version": "pzem004t"`. Since it's already been asked for a few times, I'm starting to guess it's not easily possible to use a zip file with a patch in hassio, if it's not available as an esphome branch.

dmit2k, 2019-09-25:
> [ICScraper.zip](https://github.com/esphome/feature-requests/files/3563010/ICScraper.zip)
> 
> It's a fair bit cleaner then @tsunglung's implementation, mostly because it just uses [modbusmaster](https://github.com/4-20ma/ModbusMaster) to handle all the communications.

Great work and nice realisation! Especially using modbusmuster: 

I was looking for common Modbus / RS485 support for ESPHome and that could become just a great starting point!

> It'd definitely be nice to get this (or something similar) integrated into the core. The PZEM-xxx devices are super handy!

All modbus devices are super handy as there is a lot of them especially in professional area! I would say that having core platform support for Modbus / RS485 would be a killer feature for ESPHome (similar to I2C or SPI). 

Then specific modbus/rs485 sensor implementation can be built just on top of this. Modbus allows having multiple devices on the same two wires (similar to I2C), so it is extremely useful for automation and telemetry. That would open the whole world of professional automation to ESPHome )))

fake-name, 2019-09-26:
> All modbus devices are super handy as there is a lot of them especially in professional area! I would say that having core platform support for Modbus / RS485 would be a killer feature for ESPHome (similar to I2C or SPI).

Note that doing modbus *properly* would be a LOT more complicated. This interface is exploiting the fact that the PZEM modules are doing half-assed modbus over an RS-232 interface. To do it properly you need an extra pin, and the ability to drive a transmit/receive control pin on the RS485 transciever. 

That itself would mean the serial interface module would have to be expanded to support multiple modes (the actual USART can do it, but basically nothing other then the ESP-IDF exposes the facilities). 

It'd be super neat, though!

frankiha, 2019-10-03:
> [pzemac.zip](https://github.com/esphome/feature-requests/files/3503962/pzemac.zip)
>
> unzip it and put it to site-packages folder, like [python install [...]

Hi,
Can you help me?
I'm using pzem-004t v3.0
I already applied the diff.
values ​​for energy is  Wh=0.00000 , see log.

sorry my english is not good

[13:05:02][D][pzemac:089]: V=221.8 I=0.041 P=0.5 F=59.9 PF=0.050000 Wh=0.000000
[13:05:02][D][sensor:092]: 'Pzem voltage': Sending state 221.80000 V with 1 decimals of accuracy
[13:05:02][D][sensor:092]: 'Pzem current': Sending state 0.04100 A with 2 decimals of accuracy
[13:05:02][D][sensor:092]: 'Pzem power': Sending state 0.50000 W with 1 decimals of accuracy
[13:05:02][D][sensor:092]: 'Pzem Total Daily Energy': Sending state 6.68713 Wh with 3 decimals of accuracy
[13:05:02][D][sensor:092]: 'Pzem frequency': Sending state 59.90000  with 1 decimals of accuracy
[13:05:02][D][sensor:092]: 'Pzem energy': Sending state 0.00000 Wh with 0 decimals of accuracy

uart:
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 9600
  
time:
  - platform: sntp
    id: my_time  

sensor:
  - platform: pzemac
    voltage: 
      name: "Pzem voltage"
    current: 
      name: "Pzem current"
    power:
      name: "Pzem power"
      id: PZEM004T_POWER
    energy:
      name: "Pzem energy"
    frequency:
      name: "Pzem frequency"
    update_interval: 1s

  - platform: total_daily_energy
    name: "Pzem Total Daily Energy"
    power_id: PZEM004T_POWER 
    
  - platform: wifi_signal
    name: "Pzem WiFi Signal"
    update_interval: 60s       
    
switch:
  - platform: uart
    name: "Pzem Reset Energy"
    data: [0x01, 0x42, 0x80, 0x11] 

Frank
From: Brazil.

tsunglung, 2019-10-04:
@frankiha it works great. About power, you may need to wait a while.

frankiha, 2019-10-04:
> @frankiha it works great. About power, you may need to wait a while.

yes, Works great! low current in my test.

How to get the power factor?

platform: pzemac
voltage:
name: "Pzem voltage"
current:
name: "Pzem current"
power:
powerfactor:   does not work
pf:    does not work 
factor: does not work

thank's

tsunglung, 2019-10-04:
power factor, factor..etc, are mathematically formula , not implemented. Maybe implement it after merged.

fake-name, 2019-10-04:
The hardware should absolutely return power factor and integrated power. At least the later PZEM modules do return it.

tsunglung, 2019-10-05:
you are right. power factor was reported but apparent power and integrated power are not.

yaml configuration need be added power factor

I also update the source code to improve the stable on reading from PZEM-004T
[pzemac.zip](https://github.com/esphome/feature-requests/files/3693097/pzemac.zip)

```
   powerfactor:
      name: TainanHouse Channel1 190V Power factor
```

randybb, 2019-10-08:
You can use such module https://electronics.stackexchange.com/questions/244425/how-is-this-rs485-module-working that will handle it for you.

SeByDocKy, 2019-10-09:
> you are right. power factor was reported but apparent power and integrated power are not.
>
> yaml configuration need be added power factor
>
> I also update the source code to improve the stable on [...]

Are the active and reactive power computed in the original arduino lib ? i think yes

sermayoral, 2019-10-19:
Hi everyone!

Thanks for working on pzem-004t. It is ready to be included in ESPHome?

closed 2019-10-20 by OttoWinter

Fusseldieb, 2019-10-21:
> Hi everyone!
> 
> Thanks for working on pzem-004t. It is ready to be included in ESPHome?

I think since Otto closed this issue, it will be in the next release. (I hope at least :P)

randybb, 2019-10-21:
Based on the documentation it has been included https://next.esphome.io/components/sensor/pzem004t.html / https://next.esphome.io/components/sensor/pzemac.html / https://next.esphome.io/components/sensor/pzemdc.html 
Many thanks!

Fusseldieb, 2019-10-21:
> Based on the documentation it has been included https://next.esphome.io/components/sensor/pzem004t.html / https://next.esphome.io/components/sensor/pzemac.html / https://next.esphome.io/components/sensor/pzemdc.html
> Many thanks!

Wow! This is just beautiful. 

Thank you for your hard work @tsunglung and @OttoWinter!

Just installed the the dev package ("next") in the Hassio addons and I'll be unwrapping my first PZEM!

tsunglung, 2019-10-21:
@OttoWinter 
There is an issue while integrating HomeAssistant.
I found that timeout is too short (5 seconds) in send_message_await_response_complex() of aioesphomeapi.
This will happen while connecting to up to 3 pzem-004t. Please check it.

OttoWinter, 2019-10-21:
@tsunglung Is that with the code you posted or with the most recent beta? I saw that your code had a delay() jn the main loop which would cause this problem. I have devices with many more sensors than that and those work fine.

tsunglung, 2019-10-21:
My code. I tried to change the delay in the loop. But no help. Only help after longer the timeout.

 I will test the new beta these weekend.

SeByDocKy, 2019-10-21:
When it will be available for the official stable release ?

SeByDocKy, 2019-10-21:
I just installed the dev version

Got this error

`INFO Reading configuration /config/esphome/pzem.yaml...
INFO Detected timezone 'CET' with UTC offset 1 and daylight savings time from 29 March 02:00:00 to 25 October 03:00:00
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/pzem
Processing pzem (board: nodemcuv2; framework: arduino; platform: espressif8266@1.8.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
Compiling /data/pzem/.pioenvs/pzem/src/main.cpp.o
Linking /data/pzem/.pioenvs/pzem/firmware.elf
/data/pzem/.pioenvs/pzem/src/pzemac.cpp.o: In function `esphome::pzemac::PZEMAC::update()':
pzemac.cpp:(.text._ZN7esphome6pzemac6PZEMAC6updateEv+0x4): multiple definition of `esphome::pzemac::PZEMAC::update()'
/data/pzem/.pioenvs/pzem/src/esphome/components/pzemac/pzemac.cpp.o:pzemac.cpp:(.text._ZN7esphome6pzemac6PZEMAC6updateEv+0x4): first defined here
collect2: error: ld returned 1 exit status
*** [/data/pzem/.pioenvs/pzem/firmware.elf] Error 1
========================= [FAILED] Took 16.99 seconds =========================`

my yaml file is

`esphome:
  name: pzem
  platform: ESP8266
  board: nodemcuv2
  arduino_version: 2.4.2

wifi:
  ssid: !secret esphome_ssid
  password: !secret esphome_password

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

uart:
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 9600
  id: pzem004t1

time:
  platform: sntp
  id: my_time
  
sensor:
  - platform: pzemac
    current:
      name: "enedis_courant"
    voltage:
      name: "enedis_tension"
    power:
      name: "enedis_puissance"
    frequency:
      name: "enedis_frequence"
    power_factor:
      name: "enedis_facteur_puissance"
    update_interval: 5s

 
switch:
  - platform: restart
    name: "Power Monitor Restart"
  - platform: uart
    uart_id: pzem004t1
    name: "enedis_reset"
    data: [0x01, 0x42, 0x80, 0x11]  `

OttoWinter, 2019-10-21:
@SeByDocKy Please use the clean command to clean all temporary build files - might be that the previous code by @tsunglung is still in your installation somewhere. Please also look through your installation dir (or wherever you put the manual pzemac patch) and remove them.

SeByDocKy, 2019-10-21:
> @SeByDocKy Please use the clean command to clean all temporary build files - might be that the previous code by @tsunglung is still in your installation somewhere. Please also look through your installation dir (or wherever you put the manual pzemac patch) and remove them.

Yes it worked ... sorry ... I had some files from the other implementation "in the middle of the garden". Now compile fine .... sorry for my mistake...

SeByDocKy, 2019-10-22:
With the official release how to use several PZEM each of them on their own UART ? The uart_id property is no more here.

randybb, 2019-10-22:
```address:```
https://next.esphome.io/components/sensor/pzemac.html

SeByDocKy, 2019-10-22:
> `address:`
> https://next.esphome.io/components/sensor/pzemac.html

Thanks but how to setup for example two uart ? coz address in uart definition don't accept digit

OttoWinter, 2019-10-22:
1. With this version of the integration you can connect many modbus devices to the same bus.

2. Using multiple buses got a bit more complicated.

```yaml
uart:
  - id: ubus1
    # ...
  - id: ubus2
    # ...
modbus:
  - id: mbus1
    uart_id: ubus1
  # repeat
sensor:
  - platform: pzemac
    modbus_id: mbus1
  # repeat
```

SeByDocKy, 2019-10-22:
> 1. With this version of the integration you can connect many modbus devices to the same bus.
> 2. Using multiple buses got a bit more complicated.
>
> ```yaml
> uart:
>   - id: ubus1
>     # ...
>   - id: [...]

Thanks a lot , I will try asap :)

minsuke, 2019-10-22:
[W][modbus:074]: Modbus CRC Check failed! 3E8A!=8A8A

continuosly modbus crc check fail.... I don't know what happened ....

minsuke, 2019-10-23:
I always thanks to you otto and every developers. 

ESPHOME is amazing tool for all over the world. 

I want to use pzem004t V3 in ESPHOME. 

I succeeded in Tastmota but I can't use ESPHOME. 

My enviroment. 

1. docker esphome. esphome(beta), esphome(dev) 
 -. I tried every version but can't activate. 

2.  my yaml 
 uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600

  sensor:
  - platform: pzemac
    current:
      name: "PZEM-004T V3 Current"
    voltage:
      name: "PZEM-004T V3 Voltage"
    power:
      name: "PZEM-004T V3 Power"
    frequency:
      name: "PZEM-004T V3 Frequency"
    power_factor:
      name: "PZEM-004T V3 Power Factor"
    update_interval: 60s

3. my loggers 

INFO Reading configuration /config/esphome/pzem004t.yaml...
INFO Starting log output from 192.168.0.28 using esphome API
INFO Connecting to 192.168.0.28:6053 (192.168.0.28)
INFO Successfully connected to 192.168.0.28
[13:20:15][I][app:100]: esphome version 1.15.0-dev compiled on Oct 23 2019, 13:13:28
[13:20:15][C][wifi:409]: WiFi:
[13:20:15][C][wifi:277]:   SSID: [redacted]
[13:20:15][C][wifi:278]:   IP Address: 192.168.0.28
[13:20:15][C][wifi:280]:   BSSID: [redacted]
[13:20:15][C][wifi:281]:   Hostname: 'pzem004t'
[13:20:15][C][wifi:285]:   Signal strength: -50 dB ▂▄▆█
[13:20:15][C][wifi:289]:   Channel: 13
[13:20:15][C][wifi:290]:   Subnet: 255.255.255.0
[13:20:15][C][wifi:291]:   Gateway: 192.168.0.1
[13:20:15][C][wifi:292]:   DNS1: 192.168.0.7
[13:20:15][C][wifi:293]:   DNS2: (IP unset)
[13:20:15][C][uart:134]: UART Bus:
[13:20:15][C][uart:136]:   TX Pin: GPIO1
[13:20:15][C][uart:139]:   RX Pin: GPIO3
[13:20:15][C][uart:141]:   Baud Rate: 9600 baud
[13:20:15][C][uart:142]:   Stop bits: 1
[13:20:15][C][uart:144]:   Using hardware serial interface.
[13:20:15][C][modbus:096]: Modbus:
[13:20:15](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)

SeByDocKy, 2019-10-23:
I didn't test it yet but at least it compiles fine, maybe you can try something like this
`esphome:
  name: pzem
  platform: ESP8266
  board: nodemcuv2
  arduino_version: 2.4.2

wifi:
  ssid: !secret esphome_ssid
  password: !secret esphome_password

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

uart:
  - id: ubus1
    tx_pin: GPIO1
    rx_pin: GPIO3
    baud_rate: 9600
  
modbus:
  - id: mbus1
    uart_id: ubus1

time:
  platform: sntp
  id: my_time
  
  on_time:
      - seconds: 59
        minutes: 59
        hours: 23
        then:  
          - sensor.template.publish:
              id: template_enedis_veille
              state: !lambda return id(enedis_puissance_du_jour).state;
  
sensor:
  - platform: pzemac
    current:
      name: "enedis_courant"
    voltage:
      name: "enedis_tension"
    power:
      name: "enedis_puissance"
      id: enedis_puissance
    frequency:
      name: "enedis_frequence"
    power_factor:
      name: "enedis_facteur_puissance"
    update_interval: 5s
    modbus_id: mbus1
    
  - platform: dht
    pin: GPIO14
    temperature:
      name: "tableau_temperature"
    humidity:
      name: "tableau_humidite"
    update_interval: 5s
    
  - platform: total_daily_energy
    name: "enedis_du_jour"
    power_id: enedis_puissance 
    unit_of_measurement: "kWh"
    accuracy_decimals: 2
    icon: mdi:power
    id: enedis_puissance_du_jour
    
  - platform: template
    name: "enedis_veille"
    id: template_enedis_veille
    unit_of_measurement: "kwh"
    accuracy_decimals: 2
    icon: mdi:power
    update_interval: 10s  
 
switch:
  - platform: restart
    name: "tableau_restart"
    
  - platform: uart
    uart_id: ubus1
    name: "enedis_reset"
    data: [0x01, 0x42, 0x80, 0x11]  `

tsunglung, 2019-10-24:
@OttoWinter 
a question, why not reporting energy?

OttoWinter, 2019-10-24:
@tsunglung Because you'd need a way to reset the energy value, and esphome already has `total_daily_energy` which does the same thing but automatically. If there's enough need and someone is willing to implement it (by adding a new action, like `sensor.pzemac.reset_energy`), then it would be ok to add again.

tsunglung, 2019-10-24:
@OttoWinter 
To reset energy is easy.  Use UART switch.
```
switch:
  - platform: uart
    uart_id: pzem004t1
    name: "Ch1 110V Reset Energy"
    data: [0x01, 0x42, 0x80, 0x11]
```

total_daily_energy is only daily energy. (meaning today's energy, 00:00:00~23:59:59)
The energy from pzem is meaning total energy after reset. They are different. I suggest that it shall be added.
By the way. do you have any idea how to implement yesterday's energy, like an [implement](https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xdrv_03_energy.ino#L859) of tasmota firmware?

[23 more replies]