esphome/feature-requests #51: Support of displays with ILI9341

njbuch, 2019-02-05:
**What new integration would you wish to have?**

I would like to be able to use displays that are based on the ILI9341.

**If possible, provide a link to an existing library for the integration:**
https://github.com/MartyMacGyver/ESP32_Adafruit_ILI9341

**Please describe your use case for this integration and alternatives you've tried:**

I am using the current https://esphomelib.com/esphomeyaml/components/display/ssd1306_i2c.html but these displays are on the small side.

**Additional context:**

ahd71, 2019-02-28:
+1 for this request.

There are some cheap 2.8" Touch screens that would be excellent panels (haven't tried the touch though)

SEK 74,71  25%OFF | 240x320 2.8" SPI TFT LCD Touch Panel Serial Port Module with PCB ILI9341 5V/3.3V
https://s.click.aliexpress.com/e/QkximFR

lonebaggie, 2019-03-31:
+1 for this request
M5 Stack uses this display https://m5stack.com

kostavelikov, 2019-04-24:
+1
It's a good display for a wide variety of use cases.

mr-sneezy, 2019-05-03:
+1 also.
I have three of these displays in my parts box, and would like to make a ESPhome diplay panel with one or more. I agree with a comment above, these displays are a nice size for a HA sensor or status display, and are cheap for the size.

cslehel, 2019-05-06:
+1 also. 
I would like to use the m5 stack display too with esphome. 
Thank You considering our requests.

giuarb, 2019-05-20:
+1 TTGO T4 is a great example of an expressif ESP32 and a ILI9341 plus 3 buttons, lots of uses in Home Assistant!
Please add it!

workingmanrob, 2019-05-21:
Hi all,

There is a platformio library that supports these screens and several other models (I have a couple ili9486s).

https://platformio.org/lib/show/1559/TFT_eSPI

I have installed but I'm not sure how to use it in esphome yet.

platformio lib --global install "TFT_eSPI"

Using the example sketches from their github the screen works a treat with touch support and everything.

brandond, 2019-06-08:
I recently picked up an ArduiTouch kit that came with one of these displays, along with a touch controller and SD card reader in a nice enclosure:
https://www.hwhardsoft.de/english/projects/arduitouch-esp/

I'd love to be able to drive it from ESPHome. I realize I'll have to do a lot of the drawing manually since it isn't offloaded like the nextion, but that's something I can work through. These displays frequently come with a SD reader, so it would also be nice to have support for drawing images stored on the card.

mvturnho, 2019-07-07:
Hi is someone already working on this? Maybe I can help?

mr-sneezy, 2019-07-07:
Not me, the coding behind ESPhome is beyond my ken,  but I can test with my displays also if wanted, with ESP8266 and ESP32. 
Even if we only got colour text display working, I'd be able to use it right away to replace old 16x2 LCD's in some pre-ESPhome projects.
I've been looking at the Nextion displays, but the price for equivalent size displays is close to 10x more which is over my hobby budget.

mvturnho, 2019-07-10:
Well I added ILI9341 support for M5Stack devices. 
In this implementation I do not use the buffer. all pixels are direct draw. This makes it slow because for every pixel in an object (circle, text, etc) we first set the address, start a write send the color data and end the write. 
When using the buffer we write the whole buffer all te time. This is also verry slow and takes up a lot of ram:  for 16 colorbits per pixel,  320*240 * 2 = 153kb

@OttoWinter and others some ideas on how to proceed would be appreciated.

You can clone and test from my forked repo. Use branch ili9341.
from gitbash;

```
git clone https://github.com/mvturnho/esphome.git
cd esphome
git checkout ili9341
pip install -e .
```
Then use the folowing yaml content:

```
#development config for ili9341
esphome:
  name: ili9341
  platform: ESP32
  board: nodemcu-32s

logger:
  level: VERBOSE
  
wifi: !include includes/esp32_wifi.yaml
mqtt: !include includes/mqtt_iot.yaml
ota:  !include includes/ota.yaml
spi: 
  clk_pin: 18
  mosi_pin: 23
  miso_pin: 19

binary_sensor:
  - platform: gpio
    name: "BUTTON_A_PIN"
    pin:
      number: 39
      inverted: False
      mode: INPUT_PULLUP
  - platform: gpio
    name: "BUTTON_B_PIN"
    pin:
      number: 38
      inverted: False
      mode: INPUT_PULLUP
  - platform: gpio
    name: "BUTTON_C_PIN"
    pin:
      number: 37
      inverted: False
      mode: INPUT_PULLUP

font:
  - file: "fonts/OpenSans-Regular.ttf"
    id: my_font
    size: 20

display:
  - platform: ili9341
    model: M5Stack
    cs_pin: 14
    dc_pin: 27
    led_pin: 32
    reset_pin: 33
    rotation: 0
    lambda: |-
      // Draw a circle in the middle of the display
      it.fill(ili9341::BLACK);
      it.filled_circle(it.get_width() / 2, it.get_height() / 2, 30, ili9341::RED);
      it.rectangle(10,10,100,100,ili9341::BLUE);
      it.print(0, 0, id(my_font), ili9341::GREEN,"Hello World!");
```

mvturnho, 2019-07-10:
To see the display in action:
[esphome on esp32 with ili9341](https://youtu.be/YGhVYtNOjAA)

Used this yaml:

font:
```yaml
  - file: "fonts/OpenSans-Regular.ttf"
    id: my_font
    size: 20
  - file: "fonts/OpenSans-BoldItalic.ttf"
    id: bold_italic
    size: 30

display:
  - platform: ili9341
    model: TFT_2.4
    cs_pin: 14
    dc_pin: 27
    led_pin: 32
    reset_pin: 33
    rotation: 270
    lambda: |-
      //function
      if(id(my_global_int) < 1 ) {
        it.rectangle(0,0,318,238,ili9341::WHITE);
        it.rectangle(1,1,316,236,ili9341::WHITE);
        it.horizontal_line(2,50,316,ili9341::BLUE);
        it.horizontal_line(2,51,316,ili9341::BLUE);
        it.print(120, 10, id(bold_italic), ili9341::GREEN,"ili9341");
        it.printf(30, 100, id(my_font), ili9341::YELLOW, "Counter value:");
        it.printf(30, 170, id(my_font), ili9341::YELLOW, "Wifi connection status:");
      } else {
        it.filled_rectangle(185,100,40,30,ili9341::BLACK);
        it.printf(190, 100, id(my_font), ili9341::RED, "%1d", id(my_global_int));
      }
      if(wifi_wificomponent->is_connected()) {
        it.filled_circle(270, 185, 10, ili9341::GREEN);
      } else {
        it.filled_circle(270, 185, 10, ili9341::RED);
      }
      id(my_global_int) += 1;
```

OttoWinter, 2019-07-10:
Yes I've been working on this a bit - the ili9341 driver itself is not that complicated, but the display base changes are a bit more work:

 - Color support for all display calls
 - Color support for images
 - Touch support for display base

Also, there is the question if this will work with the available memory on ESP8266s. If not, I'm considering declaring this as an ESP32-only feature. Another thing I'm considering for getting this faster is not support touch yet, and delaying that until a further release.

brandond, 2019-07-10:
I have my ili9341 + esp32 all wired up so I'm glad to test. I was hoping to use it to replace my thermostat and control some light groups around the house.

mvturnho, 2019-07-10:
> Yes I've been working on this a bit - the ili9341 driver itself is not that complicated, but the display base changes are a bit more work:
>
> * Color support for all display calls
> * Color support for [...]

Maybe we could make the touch device a separate component where you can define virtual buttons and sliders.

I tested this ili9341 component on an esp8266 but it does not work. Ik think it's a memory problem  althoug I do not allocate memory for the display buffer.

OttoWinter, 2019-07-10:
> Maybe we could make the touch device a separate component where you can define virtual buttons and sliders.

Well yes, kinda. I have something planned for display that would include this (basically, apply automation syntax for displays - user sets up individual widgets like text, buttons etc; touch trigger would be inline with the widget definition). This is also why it's taking time. Another option would be to provide raw touch x/y data, but that's probably not helpful and I rather like a good but delayed system than a bad but fast system.

workingmanrob, 2019-07-11:
Darn my two screens are already wired up to esp8266 boards and in snazzy little cases too.. all my esp32 boards and running camera modules so probably using most of their memory taking pictures.

I stripped the yaml down to pretty much the bare minimum and still get 

[14:57:46][E][display:016]: Could not allocate buffer for display!

and it boot loops.

All the example sketch files I tested with Ardiuno IDE from https://github.com/Bodmer/TFT_eSPI worked on my esp8266s but I guess they aren't running wifi or any other stuff.

mvturnho, 2019-07-11:
> Darn my two screens are already wired up to esp8266 boards and in snazzy little cases too.. all my esp32 boards are running camera modules so probably using most of their memory taking pictures.
> 
> I stripped the yaml down to pretty much the bare minimum and still get
> 
> [14:57:46][E][display:016]: Could not allocate buffer for display!
> 
> and it boot loops.
> 
> All the example sketch files I tested with Ardiuno IDE from https://github.com/Bodmer/TFT_eSPI worked on my esp8266s but I guess they aren't running wifi or any other stuff.

Yes I know it only works with esp32 right now. I tried esp8266 myself and it boot loops.

mvturnho, 2019-07-14:
Oké @OttoWinter how shall we proceed with this?
I ordered the esp32 with 8mb psram. This device could support DMA with double frame-buffering.

tonyek, 2019-08-06:
I've been trying to get the above to work with the  ESP-WROVER-KIT V4.1 development board as it also uses the ili9341 without any joy. I've changed all the pins to suit but esphome only seems accepts two models M5stack and TFT 2.4. Would it be possible to add  ESP-WROVER-KIT V4.1 .
 It's all working fine with adruino sketches, but I was hoping to display some room temperatures.

mvturnho, 2019-08-07:
Could you let me know what is wrong? You get compile errors or device crashes?
Basicaly the display model only effects the initialisation code, I cant imagine that your LCD is so different that the initialisations fails. Please use `TFT_2.4`, thats the standard ili9341 init codes.
I checked the LCD connections for the wrover kit, maybe you can try this config. 

```yaml
#development config for ili9341
esphome:
  name: ili9341
  platform: ESP32
  board: esp-wrover-kit
  platformio_options:
    board_build.mcu: esp32
    board_build.f_cpu: 240000000L

logger:
  level: VERBOSE
  
wifi: !include includes/esp32_wifi.yaml
mqtt: !include includes/mqtt_iot.yaml
ota:  !include includes/ota.yaml
spi:
  clk_pin: 19
  mosi_pin: 23
  miso_pin: 25

binary_sensor:
#  !include includes/m5_buttons.yaml
  - platform: display
    name: "Button"
    top_left: 10,10
    size: 100,10

globals:
   - id: my_global_int
     type: int
     restore_value: no
     initial_value: '0'

font:
  - file: "fonts/OpenSans-Regular.ttf"
    id: my_font
    size: 20
  - file: "fonts/OpenSans-BoldItalic.ttf"
    id: bold_italic
    size: 30

display:
  - platform: ili9341
    model: TFT_2.4
    cs_pin: 22
    dc_pin: 21
    led_pin: 5
    reset_pin: 18
    rotation: 0
    lambda: |-
      //function
      if(id(my_global_int) < 1 ) {
        it.rectangle(0,0,318,238,ili9341::WHITE);
        it.rectangle(1,1,316,236,ili9341::WHITE);
        it.horizontal_line(2,50,316,ili9341::BLUE);
        it.horizontal_line(2,51,316,ili9341::BLUE);
        it.print(120, 10, id(bold_italic), ili9341::GREEN,"ili9341");
        it.printf(30, 100, id(my_font), ili9341::YELLOW, "Counter value:");
        it.printf(30, 170, id(my_font), ili9341::YELLOW, "Wifi connection status:");
      } else {
        it.filled_rectangle(185,100,40,30,ili9341::BLACK);
        it.printf(190, 100, id(my_font), ili9341::RED, "%1d", id(my_global_int));
      }
      if(wifi_wificomponent->is_connected()) {
        it.filled_circle(270, 185, 10, ili9341::GREEN);
      } else {
        it.filled_circle(270, 185, 10, ili9341::RED);
      }
      id(my_global_int) += 1;

```

tonyek, 2019-08-07:
Hi thanks for getting back.
On first try there were no errors the file validated fine then compiled with no issue . It then uploaded ok and booted but display was blank. No errors showing on log.

Tried the new file you posted after downloading the fonts ( I used another the first try). It would not validate due to BINARY_SENSOR PLATFORM : display - not recognized

I # this section out and everything went as before I.E. validated ok and compiled then uploaded, booted . Still no errors showing on log. Display still blank .

Only thing I noticed is that there is no reference to led_pin in log. 

INFO Reading configuration /config/esp_wrover_kit_v4_1.yaml...
INFO Starting log output from 192.168.1.138 using esphome API
INFO Connecting to 192.168.1.138:6053 (192.168.1.138)
INFO Successfully connected to 192.168.1.138
[20:05:29][I][app:101]: esphome version 1.14.0-dev compiled on Aug  7 2019, 19:18:54
[20:05:29][C][wifi:399]: WiFi:
[20:05:29][C][wifi:267]:   SSID: 

[redacted]
[20:05:29][C][wifi:268]:   IP Address: 192.168.1.138
[20:05:29][C][wifi:270]:   BSSID: 

[redacted]
[20:05:29][C][wifi:271]:   Hostname: 'esp_wrover_kit_v4_1'
[20:05:29][C][wifi:275]:   Signal strength: -44 dB ▂▄▆█
[20:05:29][V][wifi:277]:   Priority: 0.0
[20:05:29][C][wifi:279]:   Channel: 2
[20:05:29][C][wifi:280]:   Subnet: 255.255.255.0
[20:05:29][C][wifi:281]:   Gateway: 192.168.1.1
[20:05:29][C][wifi:282]:   DNS1: 0.0.0.0
[20:05:29][C][wifi:283]:   DNS2: 0.0.0.0
[20:05:29][C][spi:079]: SPI bus:
[20:05:29][C][spi:080]:   CLK Pin: GPIO19 (Mode: OUTPUT)
[20:05:29][C][spi:081]:   MISO Pin: GPIO25 (Mode: INPUT)
[20:05:29][C][spi:082]:   MOSI Pin: GPIO23 (Mode: OUTPUT)
[20:05:29][C][spi:083]:   Using HW SPI: YES
[20:05:29][C][logger:175]: Logger:
[20:05:29][C][logger:176]:   Level: VERBOSE
[20:05:29][C][logger:177]:   Log Baud Rate: 115200
[20:05:29][C][logger:178]:   Hardware UART: UART0
[20:05:29][C][ili9341:029]: ili9341
[20:05:29][C][ili9341:029]:   Rotations: 0 °
[20:05:29][C][ili9341:029]:   Dimensions: 240px x 320px
[20:05:29][C][ili9341:030]:   Width: 240, Height: 320,  Rotation: 0
[20:05:29][C][ili9341:031]:   Reset Pin: GPIO18 (Mode: OUTPUT)
[20:05:29][C][ili9341:032]:   DC Pin: GPIO21 (Mode: OUTPUT)
[20:05:29][C][ili9341:034]:   Update Interval: 1.0s
[20:05:29][C][ota:029]: Over-The-Air Updates:
[20:05:29][C][ota:030]:   Address: 192.168.1.138:3232
[20:05:29][C][ota:032]:   Using Password.
[20:05:29][C][mqtt:051]: MQTT:
[20:05:29][C][mqtt:053]:   Server Address: 192.168.1.78:1883 (192.168.1.78)
[20:05:29][C][mqtt:054]:   Username: 

[redacted]
[20:05:29][C][mqtt:055]:   Client ID: 

[redacted]
[20:05:29][C][mqtt:057]:   Discovery prefix: 'homeassistant'
[20:05:29][C][mqtt:058]:   Discovery retain: YES
[20:05:29][C][mqtt:060]:   Topic Prefix: 'esp_wrover_kit_v4_1'
[20:05:29][C][mqtt:062]:   Log Topic: 'esp_wrover_kit_v4_1/debug'
[20:05:29][C][mqtt:065]:   Availability: 'esp_wrover_kit_v4_1/status'
[20:05:29][C][api:095]: API Server:
[20:05:29][C][api:096]:   Address: 192.168.1.138:6053

Anything else I can send.

mvturnho, 2019-08-08:
So your led is not lighting up?
I added the ledpin to the dump_config, so you should see your led pin now. So do a git pull and reflash to see the led_pin. The led pin is handled as active high so if yours is active low it will remain off.

Also (as a workaround) to make the backlight work you can add it to your config as a simple monochrome light.

Can you post your yaml file here?

tonyek, 2019-08-08:
Here is the yaml file

esphome:
  name: esp_wrover_kit_v4_1
  platform: ESP32
  board: esp-wrover-kit
  platformio_options:
    board_build.mcu: esp32
    board_build.f_cpu: 240000000L
# Enable logging
logger:
  level: VERBOSE
    
wifi:
  ssid: "ASUS"
  password: "g8Way4thnet"
  manual_ip:
    static_ip: 192.168.1.138
    gateway: 192.168.1.1
    subnet: 255.255.255.0  

mqtt:
  broker: "192.168.1.78"
  username: tonyqt
  password: qtyn0t
  
# Enable Home Assistant API
api:
  password: '1i0ta'
ota:
  password: '1i0ta'
  
spi:
  clk_pin: 19
  mosi_pin: 23
  miso_pin: 25

#binary_sensor:
#  !include includes/m5_buttons.yaml
#  - platform: display
#    name: "Button"
#    top_left: 10,10
#    size: 100,10

globals:
   - id: my_global_int
     type: int
     restore_value: no
     initial_value: '0'

font:
  - file: "fonts/OpenSans-Regular.ttf"
    id: my_font
    size: 20
  - file: "fonts/OpenSans-BoldItalic.ttf"
    id: bold_italic
    size: 30

display:
  - platform: ili9341
    model: TFT_2.4
    cs_pin: 22
    dc_pin: 21
    led_pin: 5
    reset_pin: 18
    rotation: 0
    lambda: |-
      //function
      if(id(my_global_int) < 1 ) {
        it.rectangle(0,0,318,238,ili9341::WHITE);
        it.rectangle(1,1,316,236,ili9341::WHITE);
        it.horizontal_line(2,50,316,ili9341::BLUE);
        it.horizontal_line(2,51,316,ili9341::BLUE);
        it.print(120, 10, id(bold_italic), ili9341::GREEN,"ili9341");
        it.printf(30, 100, id(my_font), ili9341::YELLOW, "Counter value:");
        it.printf(30, 170, id(my_font), ili9341::YELLOW, "Wifi connection status:");
      } else {
        it.filled_rectangle(185,100,40,30,ili9341::BLACK);
        it.printf(190, 100, id(my_font), ili9341::RED, "%1d", id(my_global_int));
      }
      if(wifi_wificomponent->is_connected()) {
        it.filled_circle(270, 185, 10, ili9341::GREEN);
      } else {
        it.filled_circle(270, 185, 10, ili9341::RED);
      }
      id(my_global_int) += 1;

Sorry but what do you want me to git pull

tonyek, 2019-08-08:
Just to let you know my Esphome is in a container on Qnap nas.

tonyek, 2019-08-09:
Hi
Got it going last night with your workaround. I've got GPIO 5 setup as a switch and can turn the backlight on and off. I'll just need to get my head around the lambda rendering now.

Thanks again for your help
Tony

mvturnho, 2019-08-09:
Yes I checked the schematics and the backlight led turnson when your pin 5 is low.

so in the config you colud use the `inverted `option for the `led_pin`:

```yaml
display:
  - platform: ili9341
    model: TFT_2.4
    cs_pin: 22
    dc_pin: 21
    led_pin:
      number: 5
      inverted: yes
    reset_pin: 18
    rotation: 0
    lambda: |-
      //function
      if(id(my_global_int) < 1 ) {
        it.rectangle(0,0,318,238,ili9341::WHITE);
        it.rectangle(1,1,316,236,ili9341::WHITE);
        it.horizontal_line(2,50,316,ili9341::BLUE);
        it.horizontal_line(2,51,316,ili9341::BLUE);
        it.print(120, 10, id(bold_italic), ili9341::GREEN,"ili9341");
        it.printf(30, 100, id(my_font), ili9341::YELLOW, "Counter value:");
        it.printf(30, 170, id(my_font), ili9341::YELLOW, "Wifi connection status:");
      } else {
        it.filled_rectangle(185,100,40,30,ili9341::BLACK);
        it.printf(190, 100, id(my_font), ili9341::RED, "%1d", id(my_global_int));
      }
      if(wifi_wificomponent->is_connected()) {
        it.filled_circle(270, 185, 10, ili9341::GREEN);
      } else {
        it.filled_circle(270, 185, 10, ili9341::RED);
      }
      id(my_global_int) += 1;
```

tonyek, 2019-08-09:
Yeah invert worked thanks

tonyek, 2019-08-12:
Just a heads up for you.
I added the line " esp32_ble_tracker: " to my yaml which I need to get the data from my temp/humidity sensors . This work fine before on this board before changes. Now the wifi will not reconnect to router.
I.E. bluetooth works fine but wifi won't connect. 

Checked out and as soon as I removed the following lines

platformio_options:
    board_build.mcu: esp32
    board_build.f_cpu: 240000000L
it started working.

Must be some sort of conflict in that build.

Thanks again for your help

tonyek, 2019-08-22:
I'm still having some issues with the display. I've got it printing some data from a sensor.
At first it's fine but when the data changes the new figures are printed on top of the old ones.
It eventually clears but takes some time . Not sure if this is the refresh rate or how I'm doing it. 
I've tried a few ways but the same thing happens.

esphome:
  name: esp_wrover_kit_v4_1
  platform: ESP32
  board: esp-wrover-kit
 
# Enable logging
logger:
  #level: VERBOSE
    
wifi:
  ssid: "ASUS"
  password: "g8Way4thnet"
  manual_ip:
    static_ip: 192.168.1.138
    gateway: 192.168.1.1
    subnet: 255.255.255.0  

web_server:
  port: 80
  
mqtt:
  broker: "192.168.1.78"
  username: tonyqt
  password: qtyn0t
  
# Enable Home Assistant API
api:
  password: '1i0ta'
ota:
  password: '1i0ta'
  
esp32_ble_tracker: 

time:
  - platform: homeassistant
    on_time:
      - seconds: 0
        minutes: 0
        hours: 4
        days_of_week: MON-SUN
        then:
          - switch.toggle: Restart
switch:
  - platform: restart
    name: "test_temperature Restart"
    id: Restart
    
sensor:
  - platform: xiaomi_mijia
    mac_address: 58:2D:34:31:94:DD
    temperature:
      name: "Sensor_2 Temperature"
      id: sen2temp
    humidity:
      name: "Sensor_2 Humidity"
      id: sen2himid
    battery_level:
      name: "Sensor_2 Battery Level"
      id: sen2batlev
      
    
    
  - platform: xiaomi_mijia
    mac_address: 58:2D:34:31:99:23
    temperature:
      name: "Sensor_3 Temperature"
      id: sen3temp
      
    humidity:
      name: "Sensor_3 Humidity"
      id: sen3himid
    battery_level:
      name: "Sensor_3 Battery Level"
      id: sen3batlev
      
      
text_sensor:
  - platform: mqtt_subscribe
    name: "Data from topic"
    id: mysensor
    topic: esp_wrover_kit_v4_1/sensor/sensor_2_temperature/state      
      
spi:
  clk_pin: 19
  mosi_pin: 23
  miso_pin: 25
  

globals:
   - id: my_global_int
     type: int
     restore_value: no
     initial_value: '0'

font:
  - file: "fonts/OpenSans-Regular.ttf"
    id: my_font
    size: 20
  - file: "fonts/OpenSans-BoldItalic.ttf"
    id: bold_italic
    size: 30
    

display:
  - platform: ili9341
    model: TFT_2.4
    
    cs_pin: 22
    dc_pin: 21
    led_pin:
      number: 5
      inverted: yes
    reset_pin: 18
    rotation: 90
    
    lambda: |-
      
      it.printf(30, 120, id(my_font), ili9341::RED, "The data is: %s", id(mysensor).state.c_str());
      
      if (id(sen2temp).has_state()) {
        it.printf(127, 60, id(my_font), ili9341::RED, "%.1f°", id(sen2temp).state);
      }

mvturnho, 2019-08-22:
The display only draws what is in your lambda code.
So the display is never cleared. I use `it.filled_rectangle(185,100,40,30,ili9341::BLACK);` to clear the text first and then draw the new text on top.

```cpp
        it.filled_rectangle(185,100,40,30,ili9341::BLACK);
        it.printf(190, 100, id(my_font), ili9341::RED, "%1d", id(my_global_int));
```

Te reason for this is that the driver is extremely slow. We have not enough memory to use a buffer and we do not use DMA for the datatransfer. 
So this is all experimental code and far from production ready.

tonyek, 2019-08-22:
Thanks I'll give that a try.

benv666, 2019-08-23:
Just got this up and running on M5Stack (core), wanted to point out that the colors are completely different from what I expect. I.e. black is white, blue shows as yellow, red shows as cyan, etc.
Currently testing with this snippet:
```
display:
  - platform: ili9341
    model: M5Stack
    update_interval: 10.0s
    cs_pin: 14
    dc_pin: 27
    led_pin: 32
    reset_pin: 33
    rotation: 0
    lambda: |-
      it.fill(ili9341::BLACK);
      it.print(0, 0, id(inconsolata), ili9341::RED, "M5Stack test");
      it.filled_rectangle(0,20,240,20,ili9341::BLACK);
      it.strftime(0, 20, id(inconsolata), ili9341::YELLOW, "%Y-%m-%d %H:%M:%S", id(sntp_time).now());

```

Electronlibre2012, 2019-08-24:
> Yes I've been working on this a bit - the ili9341 driver itself is not that complicated, but the display base changes are a bit more work:
>
>     * Color support for all display calls
>
>     * Color [...]

Hello @OttoWinter , i have a project that work on Arduino with a Wemos D1 mini and a TFT 2.4 (or 2.8) : weather station color : make of this : [https://thingpulse.com/product/esp8266-wifi-color-display-kit-2-4/](url) and it work with a Wemos D1 mini instead of a mini pro, or this : [https://learn.adafruit.com/wifi-weather-station-with-tft-display/software](url)

So i think an esp8266 (or Wemos D1 mini) is not a problem, and its cheapest than a ESP32...

It would be great if it can work on esphome with a Wemos D1 mini...

mvturnho, 2019-08-24:
Then I think you have another display type in your M5Stack.
We need to add a color_mode setting like with the fastled neopixel has been done. You can supply the color_mode.
RGB
BGR
...

Electronlibre2012, 2019-08-24:
Hello @mvturnho ,

how to install display.ili9341 on hassio please?

i cant type this command for hassio i think : ```
git clone https://github.com/mvturnho/esphome.git
cd esphome
git checkout ili9341
pip install -e .
```

thanks

Alan-K2, 2019-08-30:
@ Electronlibre2012 
I used pip3 install -e .
which seemed to work although when I compile mvturnho  yaml example for m5stack I get "Platform not found: 'display.ili9341'." so still not correct. I have two m5stack modules I would love to make use of. Please could someone clarify which files need to be in which directory for the display to be recognised?
Thanks
Alan

Electronlibre2012, 2019-08-31:
> pip3 install -e .

Hello @Alan-K2 , thanks but this command doesnt work on Hassio, are you on HA under Raspbian or other?

Nobody seem interested to make colors screen works on HA...

Nextion cost a lot... :(

mvturnho, 2019-08-31:
> @ Electronlibre2012
> I used pip3 install -e .
> which seemed to work although when I compile mvturnho yaml example for m5stack I get "Platform not found: 'display.ili9341'." so still not correct. I [...]

Did you forget to do

`git checkout ili9341`

After the git clone ......

Alan-K2, 2019-08-31:
> > pip3 install -e .
> 
> Hello @Alan-K2 , thanks but this command doesnt work on Hassio, are you on HA under Raspbian or other?
> 
> Nobody seem interested to make colors screen works on HA...
> 
> Nextion cost a lot... :(

No I've got a standard install of HA, pip didn't work for me but pip3 did - not sure why!

Alan-K2, 2019-08-31:
Hi mvturnho
> 
> Did you forget to do
> 
> `git checkout ili9341`
> 
> After the git clone ......

I was sure I had but I've just run it all again to be 100% sure and still get the same result. As it ran I got quite a lot of "requirement already satisfied" then the last few lines were:
Installing collected packages: voluptuous, paho-mqtt, colorlog, tornado, protobuf, pyserial, bottle, colorama, semantic-version, platformio, pyaes, ecdsa, 
esptool, esphome
  Running setup.py install for voluptuous ... done
  Running setup.py install for paho-mqtt ... done
  Running setup.py install for tornado ... done  Running setup.py install for platformio ... done
  Running setup.py install for pyaes ... done
  Running setup.py install for esptool ... done
  Running setup.py develop for esphome
Successfully installed bottle-0.12.17 colorama-0.4.1 colorlog-4.0.2 ecdsa-0.13.2 esphome esptool-2.7 paho-mqtt-1.4.0 platformio-4.0.3 protobuf-3.7.1 pyaes-1.6.1 pyserial-3.4 semantic-version-2.8.1 tornado-5.1.1 voluptuous-0.11.7

Anything else I can check?
Thanks
Alan

Electronlibre2012, 2019-09-01:
Hello,

i have install hassio in a docker env,

make like this : 
```
git clone https://github.com/mvturnho/esphome.git
cd esphome
git checkout ili9341
pip install -e .
```

and i still have : "Platform not found: 'display.ili9341'."

what i am missing? Is ther more package to install?

Thanks!

andrelung, 2019-09-18:
Quick and maybe dumb question regarding the display on the M5Stack: is there a way to access the native interface from esphome (like `M5.Lcd.setLight()`)? I guess it is included, since we defined the platform, but I do not know how to access it from esphome or the lambda...

@Electronlibre2012 - I get this error, when I am using a different version (like current dev). Maybe try removing esphome with `pip uninstall esphome` and make sure it is not active somehow (running `esphome` should throw an error). After that, try again.

mvturnho, 2019-09-24:
> Quick and maybe dumb question regarding the display on the M5Stack: is there a way to access the native interface from esphome (like `M5.Lcd.setLight()`)? I guess it is included, since we defined [...]

No the esphome has its own ili9341 driver and uses no M5Stack software. So treat the lcd backlight as a normal light.

mvturnho, 2019-09-24:
For all who are interested, I added an 8 bit framebuffer. The standard esp32 (like M5Stack) can handle that and it speeds things up. Also the driver now uses a drawing window. So only the parts that have been updated are flushed from the framebuffer to the display. the 8 bit framebuffer allows the use of 256 colors. More colors means bigger framebuffer.

enjoy.

twigs, 2019-11-12:
Thanks for developing this support for this display! Are there any plans to merge this back into esphome?

Electronlibre2012, 2019-11-17:
> 
> 
> Thanks for developing this support for this display! Are there any plans to merge this back into esphome?

Yes it will be awesome! we can make cheap touch buttons to control any device in home assistant...

Any chance @OttoWinter please?

Alan-K2, 2019-11-19:
Agreed - if its possible then native support would be very much appreciated!

brandond, 2019-11-21:
I finally got around to trying this on my ArduiTouch board. It builds fine, and the logs look clean and the backlight comes on, but I can't get the display to draw anything. There are no errors in the logs, so I have no idea if it's talking to the board or not.

Here's the layout for the board: https://www.hwhardsoft.de/app/download/11468539297/ArduiTouch+V01-02-00.pdf

I've stripped it down to just drawing a couple lines but still get nothing. Here's the YAML I'm using:
```yaml
spi:
  clk_pin: 18
  mosi_pin: 23
  miso_pin: 19

display:
- platform: ili9341
  model: TFT_2.4
  cs_pin: 5
  dc_pin: 4
  reset_pin: 22
  led_pin:
    number: 15
    inverted: true
  lambda: |-
    it.rectangle(0,0,50,50,ili9341::WHITE);
    it.horizontal_line(2,50,100,ili9341::BLUE);
    it.horizontal_line(2,51,101,ili9341::BLUE);
```

raberrio, 2020-01-07:
@Electronlibre2012 Hi, did you install it successfully? I am also running hassio on docker env, using a PC with ubuntu server.

Don't know how to install it. I have a TTGO T4 board.

Please if you can help me,

Thanks

Electronlibre2012, 2020-01-08:
> 
> 
> @Electronlibre2012 Hi, did you install it successfully? I am also running hassio on docker env, using a PC with ubuntu server.
> 
> Don't know how to install it. I have a TTGO T4 board.
> 
> Please if you can help me,
> 
> Thanks

Hello @raberrio ,

any succes to install it for me...

I am on hassio with RPi4 now and there is no integration in ESPHome for now for such a screen...

mvturnho, 2020-01-08:
Hi @raberrio 

I had one (or two ;-) of these TTGO T4 device available and tested the code there. And it works, check [this video](https://youtu.be/B4lbVLfxVzs); 

You have to pay attention to the pin layout, I added my yaml file here for you to try this.

```yaml
esphome:
  name: ttgot4
  platform: ESP32
  board: nodemcu-32s
  platformio_options:
    upload_speed: 921600
    monitor_speed: 115200

logger:
  level: VERBOSE
  
wifi: !include includes/esp32_wifi.yaml
mqtt: !include includes/mqtt_iot.yaml
ota:  !include includes/ota.yaml
spi:
  clk_pin: 18
  mosi_pin: 23
  miso_pin: 12

globals:
   - id: my_global_int
     type: int
     restore_value: no
     initial_value: '0'

font:
  - file: "fonts/OpenSans-Regular.ttf"
    id: my_font
    size: 20
  - file: "fonts/OpenSans-BoldItalic.ttf"
    id: bold_italic
    size: 30

display:
  - platform: ili9341
    model: TFT_2.4
    cs_pin: 27
    dc_pin: 26
    led_pin: 4
    reset_pin: 5
    rotation: 270
    lambda: |-
      //function
      if(id(my_global_int) < 1 ) {
        it.rectangle(0,0,318,238,ili9341::WHITE);
        it.rectangle(1,1,316,236,ili9341::WHITE);
        it.horizontal_line(2,50,316,ili9341::BLUE);
        it.horizontal_line(2,51,316,ili9341::BLUE);
        it.print(120, 10, id(bold_italic), ili9341::GREEN,"ili9341");
        it.printf(30, 100, id(my_font), ili9341::YELLOW, "Counter value:");
        it.printf(30, 170, id(my_font), ili9341::YELLOW, "Wifi connection status:");
      } else {
        it.filled_rectangle(185,100,40,30,ili9341::BLACK);
        it.printf(190, 100, id(my_font), ili9341::RED, "%1d", id(my_global_int));
      }
      if(wifi_wificomponent->is_connected()) {
        it.filled_circle(270, 185, 10, ili9341::GREEN);
      } else {
        it.filled_circle(270, 185, 10, ili9341::RED);
      }
      id(my_global_int) += 1;

```

raberrio, 2020-01-09:
Thanks @mvturnho :) i will try it today.
When you pu “cd esphome”, this is the esphome folder inside hassio docker directory? In my case it is
/home/user/docker/hassio/homeassistant/esphome

Sorry to bother you, this is my first “custom” component.

Thanks again.

raberrio, 2020-01-09:
I am getting this:

`Obtaining file:///home/raberrio/esphome
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/home/raberrio/esphome/setup.py", line 6, in <module>
        from esphome import const
      File "esphome/const.py", line 6
        __short_version__ = f'{MAJOR_VERSION}.{MINOR_VERSION}'
                                                             ^
    SyntaxError: invalid syntax
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /home/raberrio/esphome/ `

raberrio, 2020-01-09:
I am getting this:
```
Obtaining file:///home/raberrio/esphome
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/home/raberrio/esphome/setup.py", line 6, in <module>
        from esphome import const
      File "esphome/const.py", line 6
        __short_version__ = f'{MAJOR_VERSION}.{MINOR_VERSION}'
                                                             ^
    SyntaxError: invalid syntax
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /home/raberrio/esphome/
```

mvturnho, 2020-01-09:
ok, you have python 3 installed? check

raberrio, 2020-01-09:
Yeah, i have it. Ok i did it, used pip3 instead pip.

raberrio, 2020-01-09:
Ok i need some further configuration? I am
getting an error msg on esphome when i put this lines

```
display:
  - platform: ili9341
```

mvturnho, 2020-01-09:
Did you forget to do

git checkout ili9341

raberrio, 2020-01-09:
Yeah. Surely i am doing something wrong. Do i need to compile outside of esphome environment on hassio?

[67 more replies]

closed 2021-01-11 by glmnet