esphome/feature-requests #56: Allow Device Symlinks I.E. By-ID To Be Mapped In Docker Containers

danieledwardgeorgehitchcock, 2019-02-10:
**Is your feature request related to a problem/use-case? Please describe.**

At present, the Docker container for ESPHomelib does not seem to support device symlinking i.e. devices in /dev/serial/by-id.

**Describe the solution you'd like:**

Device symlinking has been an allowed feature of Docker for quite a while and allowing this method of mapping devices can resolve a lot of issues for machines that have multiple devices occupying the serial bus for which, the generic /dev/TTYUSB0 can change on reboot.

**Additional context:**

N/A

OttoWinter, 2019-02-10:
ESPHome uses pyserial's mechanism for listing serial devices. (see https://github.com/pyserial/pyserial/blob/master/serial/tools/list_ports.py). That functionality would need to be added in pyserial's repo.

Also you can totally use `/dev/serial/by-id` paths in ESPHome, the only thing that's not working is listing them.

danieledwardgeorgehitchcock, 2019-07-29:
@OttoWinter according to the following docs, this can be done by adding `include_links=True` to the the parameters of the list_ports module:

[List Ports](https://pyserial.readthedocs.io/en/latest/tools.html#module-serial.tools.list_ports)

danieledwardgeorgehitchcock, 2019-08-16:
For anyone stumbling across this issue in the meantime, there is a workaround using docker-compose bind style device mappings (could probably be ported over to a command line argument):

`    devices:
      - /dev/serial/by-id/usb-FTDI_FT230X_Basic_UART_DO00KGTH-if00-port0:/dev/ttyUSB0
`

OttoWinter, 2019-08-27:
@danieledwardgeorgehitchcock Thanks! That parameter might solve it.

closed 2019-08-31 by OttoWinter