esphome/feature-requests #57: Allow esphome nodes to talk directly to eachother

erazor666, 2019-02-10:
I've asked about this in discord once, perhaps this is more the correct place for it.
Not being an able coder of any sort myself i've embraced esphome and loving it.

I'll try to illustrate this from my perspective, there are probably many use cases for this:

I've made some monochromatic lights and i'm controlling them with binary sensors on another node via home assistant automation/scenes and this works great and as expected.
However, if something should happen to my HA installation it would definitely be very bad and render me without control of the lights. 

If I want to make a very small installation in a place that its not practical or economical to have a HA instance running. 
Say i want to make a simple installation of a couple of lights and a switch for my parents and having the robustness and easy deployment of esphome nodes would be very beneficial.

I don't know how tall of an order this request is, I would be happy with this being some sort of 'expert option' that needed you to address the nodes with their ip address and functions. It wouldn't have to be all 'automagic'. Of course a smart implementation with lots of automagics would be sweet, but then nodes would need some kind of registry of eachother and complications could arise.

I'm guessing this is not possible as esphome is currently, at least i haven't spotted it in documentation.
Not looking for a world without HA but as a fallback, fail safe mode. The Node would have to go into this mode if it doesn't detect the HA instance for a given amount of time or set up as standalone nodes if there never will be a HA connection.

**TL;DR:**
_If a lightswitch node could turn on/off lights on a remote node(monochromatics at least) directly without HA being alive.
If a sensor node could publish to a remote node with a VFD or OLED display (or the display node could pull this info from the sensor node)_

SiliconAvatar, 2019-02-23:
I second this request, along with #52 which would also suit my use case.

The MQTT topic subscription component gets close to this, taking HA out of the loop, but I want a truly independent system that only requires WiFi to be working for one node to trigger an action in another node.

OttoWinter, 2019-02-24:
I'm closing this for https://github.com/esphome/feature-requests/issues/52 (they look essentially the same for me)

As for the feature, yes that is a future plan for ESPHome's native API - but it probably won't happen soon because quite a things need to be done first to allow that.

closed 2019-02-24 by OttoWinter