esphome/feature-requests #58: Emulate WeMo device

highground88, 2019-02-12:
Fantastic work on the current state of esphomeyaml!!! Have just started playing with it this week and it's a dream in most respects. But.. onto the topic at hand: Missing WeMo support

Currently I'm using Tasmota and a mixture of others for my ESP's, however I would love to fully move over, but unless I'm mistaken the ability to emulate a WeMo device (or similar), to allow discovery in Alexa and Google Home, is missing.

Adding this ability into the system would provide voice control for switches etc, and potential simple binary sensors to communicate and work within these AI systems (routines etc..)

I

brandond, 2019-02-12:
I am using the Home Assistant Hue bridge emulation with Google Home and can vouch for that. I haven't tried it with Alexa yet.

randyarrasmith, 2019-04-09:
I am using Home Assistant and it doesn't emulate Wemo devices.  It has support for emulated Wemo devices, which is what I believe this request is about, which would be redundant to HA since HA has ESPHome as an integration.  However, it would greatly increase the ease of using Alexa with ESPHome for users that don't have Alexa working with HA because they are not using HA.cloud or have an Amazon developer setup with an endpoint.  IMO, this is a big happy factor for a lot of people and even opens possibilities of using ESPHome independent of HA.
DISCLAIMER:  I have not tried the Hue bridge emulation but from what I read, i am not looking for something that exposes everything (with the option to exclude) but rather have the opt-in for esp8266 / 32 devices via esphome.

jhall39, 2019-08-21:
FWIW, I just tried to setup the emulated hue option in HA with Alexa and ran into a roadblock. The Alexa skill for Hue requires you to link to a Hue account but that account requires you to select your Hue bridge before it will work. It doesn't detect the emulated bridge in HA, so I was unable to complete the link. On the other hand, my Wemo devices simply work with Alexa without fuss. It would be great if esphome could be configured to look like a Wemo light to Alexa.

highground88, 2019-09-20:
Not sure if this issue is still being monitored or considered, but this new code may be of interest and is only 80 lines - unsure of port usage, but this would provide voice support to simple devices.

https://github.com/philbowles/weenymo

prabbit237, 2019-10-28:
> It would also be difficult because the web_server component uses the same port too (so using that library would probably not be possible)

The port in the web_server can be changed so I don't see this as impossible (or even difficult. Maybe just throw a compile error if it's left to the default of 80 and if the fauxmo is also enabled. Could be done with a few #ifdef statements.)

And one benefit of the Wemo/fauxmo is that it doesn't require an external server at all. It's all done directly between the Alexa and the Wemo device. So if a server goes down somewhere (other than the Alexa server) it'd still work.

jamman9000, 2019-12-17:
I LOVE the GUI for ESPHome in HA. I love how simple it is to get all the sensors and anything I add into the sketch right into HA and it all just WORKS! So awesome, and thank you!

I have to add to the list of hands raised saying easier Alexa integration via this method would be the tipping point. I can't run routines in Alexa on my ESPHome entities, shoot, I can't even get the entities exposed and working properly from HA into Alexa. The Emulated Hue integration has been busted for months, thankfully just got a ton fixed but still a ways to go (maybe arrive to all in an HA update or two in the future)... switched to trying Node-Red.

There are two pallets being maintained, one uses a third-party cloud server to match up the alexa entities with the HA entities and translate messages... the other keeps as much as we can local. The local one is node-red-contrib-amazon-echo. Unfortunately this pallet has very little documentation on how to actually use it in various ways. The furthest I can figure out is that I have to create fake entities for Alexa to recognize, sniff out the unique IDs those were given, then endlessly parse messages to both match up those IDs with HA entity names, changing json values based on the matching, then again parse what the action is depending on if it's an ON/OFF action or just a Dimmer and don't get me started on colors and..ugh.

TL;DR - Search for "ESPHome devices in Alexa" and it's a big hole of why doesn't this exist? Thankfully this github thread came up, and it also says it doesn't exist.

I'm stuck in Tasmota until I can get Alexa to discover and properly command ESPHome devices. What about Hue Bridge? We have a HA API component we just call in the ESPHome yaml... WeMo and Hue Bridge have been standard on Tasmota for a long time. Adding a voice assistant integration method seems like a no-brainer - ESPHome sensors to trigger routines or automations in Alexa is another huge reason.

prabbit237, 2019-12-18:
Take a look at [https://github.com/custom-components/alexa_media_player](https://github.com/custom-components/alexa_media_player)

It has ways of sending commands to Alexa to start/stop media play (such as Pandora or such) but it also has ways of sending strings to Alexa to be ran through its TTS engine as well as ways to trigger Alexa routines. So this may resolve at least some of your issues.

sheminasalam, 2021-03-31:
any development in adding the emulation to esphome?

closed 2022-07-01 by nagyrobi