esphome/feature-requests #61: Servo Support

thanatars, 2019-02-14:
**Describe the problem you have/What new integration you would like**
It would be great to have support for analog or digital servos such as the Futaba S3003.

**Please describe your use case for this integration and alternatives you've tried:**
Servos are commonly used in window blind projects, but existing solutions could be improved. 

**Additional context**
I would use the new capability to finally automate my blinds through Home Assistant and locally using a button on the blinds.

OttoWinter, 2019-02-20:
Servos are just things that use PWM duty cycles to encode data - It will be added to esphome at some point, but until then you can use this simple config that I created some time ago too: https://pastebin.com/8b8vQPxc

335iguy, 2019-03-10:
> Servos are just things that use PWM duty cycles to encode data - It will be added to esphome at some point, but until then you can use this simple config that I created some time ago too: https://pastebin.com/8b8vQPxc

Does this go straight into the yaml file?

OttoWinter, 2019-03-17:
Added in 1.12

closed 2019-03-17 by OttoWinter