esphome/feature-requests #63: Esphome forum

erazor666, 2019-02-17:
**Describe the problem you have/What new integration you would like**
A dedicated forum for esphome

**Please describe your use case for this integration and alternatives you've tried:**
Discord, github and the documentation are all good places for information and help, but i dont feel any of them fully replace a forum. For looking up solutions, examples and better search functions they usually provide.

**Additional context**
There is of course some work in deploying a forum, getting moderators and the works. The question i guess is, is this something that has been thought of and discarded or other reason there isn't one?

OttoWinter, 2019-02-20:
Well I wanted to add one some time ago, but all solutions either turned out to be bad or too expensive.

But ... there is already a forum for esphome on community.home-assistant.io - it's not the perfect place but until a better solution is found I think it is good enough. https://community.home-assistant.io/c/third-party/esphomelib

erazor666, 2019-02-20:
@OttoWinter
Yes, cost is also a factor. Is there/will be a donation option? For hardware, development costs or beer at least?

nagyrobi, 2022-06-17:
Join Discord please: https://discord.gg/3YFPbkkJ
People are friendly there and Threads functionality allows to discuss stuff in detail, also to look for things discussed in the past.

closed 2022-06-17 by nagyrobi