esphome/feature-requests #64: Add Minimum Power setting to Output

mtl010957, 2019-02-18:
**Describe the problem you have/What new integration you would like**

Add a Minimum Power configuration value to the [Output Compoent](https://esphome.io/components/output/), complementary to the existing Maximum Power setting. Make the [ESP8266 Software PWM](https://esphome.io/components/output/esp8266_pwm) and others as appropriate handle both minimum and maximum power settings.

**Please describe your use case for this integration and alternatives you've tried:**

Having both a minimum and maximum setting makes the PWM output capable of driving standard servos (for example, [MS90S](https://engineering.tamu.edu/media/4247823/ds-servo-mg90s.pdf)). These servos require a PWM signal at 50 Hz. with duty cycle of 10% to 20% to operate properly. The Minimum Power setting would make the full range of input values (0.0-1.0) available to span the 10%-20% operating range. Currently the PWM output does work with the Maximum Power set to 20% but the lower value of 0% makes the first half of the operating range unusable. For example, using the Monochromatic platform light to drive a servo using PWM works, but nothing happens until the brightness setting hits at least 50%, and the entire servo operation (180 degrees of rotation) is only controllable in the upper 50% of the brightness adjustment.

**Additional context**

This small change would help get a workable Servo capability (reference feature request #61) without creating an entirely new output component platform for motors at this time.

mtl010957, 2019-02-19:
I'm currently working on implementing this, I hope to get a PR submitted int the next few days.

OttoWinter, 2019-02-20:
I agree with the `min_power` option, but I would not recommend adding steppers through lights, that feels very hacky

mtl010957, 2019-02-21:
Thanks for the feedback and this excellent project! I agree that lights are not the right way to handle servos or steppers, it was just the easiest way to test the behavior. I plan to put in another enhancement to extend the Covers base component to include the tilt related messages from the HA MQTT Cover component. My use case is to drive window blinds with servos.

closed 2019-02-26 by OttoWinter