esphome/feature-requests #65: Scan Bluetooth (non BLE) devices

abmantis, 2019-02-20:
**Describe the problem you have/What new integration you would like**

It would be great to be able to scan classic bluetooth (not BLE) devices, like phones.

**Please describe your use case for this integration and alternatives you've tried:**

A device tracker for phones.

**Additional context**

I'm not sure if there's a high-level API for this yet, but it is possible: https://techtutorialsx.com/2018/03/04/esp32-arduino-bluetooth-classic-getting-started/

The PR mentioned on that article is already merged, so it should be simpler.

abmantis, 2019-02-22:
Quick question: are you guys using (or OK with using) the espidf framework from platformio? I'm working on a classic BT device tracker for the ESP32, which could probably be included in esphome, but it requires that framework.

OttoWinter, 2019-02-22:
ESPHome uses the arduino framework, but that uses esp-idf internally, so yes that can be used (see ble_tracker implementation for example)

abmantis, 2019-02-22:
@OttoWinter cool, thanks! I'll try to do something then!

runevad, 2019-03-10:
If this would be implemented, does it mean we can use an ESP32 and Esphome to set up Room tracking like: https://www.home-assistant.io/components/sensor.mqtt_room/  ? 
That would be awesome :)

abmantis, 2019-03-10:
@runevad yes, that was my plan. Unfortunately I'm finding it hard to do without having an impact on the phone's baterry, since the classic bluetooth discovery APIs from espressif are incomplete at the moment.

GeoffAtHome, 2019-06-06:
Would be great if you could detect regular Bluetooth devices as easily as Bluetooth LE device.

nmsoares, 2019-08-22:
Any developments on this ?

deex2, 2019-10-08:
Hi,

Any developments on this please?

martydingo, 2022-11-03:
Any progress? 

It's been years. And I have a device that requires pairing before BLE can work. I am seeing a status=133 error code, and I can't seem to get the ESP32 to pair with the BLE device.