esphome/feature-requests #66: Add option to periodically rescan for optimum WiFi channel (in case of mesh wifi)

xlfe, 2019-02-20:
I have multiple Wifi Access points operating in a mesh - this means they are broadcasting the same SSID but on different channels. 

If an ESPHome  device connects to one AP it seems that it will never rescan (which makes sense in most scenarios) to look for an AP that is operating on a different channel but is closer and would provide a more reliable connection for the same SSID.

In the case of a power failure where the further AP boots before the closer AP, all ESPHome devices connect to the one AP and then don't migrate (even after a few days).

It would be good if ESPHome Wifi component could add an option to rescan for the best wifi channel/AP every (configurable) 6, 12, 24 hours. This would presumably be a short disconnect so having the ability to configure it (rather than default) would make sense.

OttoWinter, 2019-02-20:
The simple solution for that would be to have an automation trigger a reboot every 6/12/24 hours - that will effectively have the same effect.

But why would this be necessary? I mean as long as the ESPs have a connection all is well right? And if they lose connection because of too poor wifi they _will_ do a scan and find the new best one.

xlfe, 2019-02-20:
Good point. Thanks!

closed 2019-02-20 by xlfe