esphome/feature-requests #69: Upgrade ESP8266 arduino core to 2.5.0

OttoWinter, 2019-02-20:
**Describe the problem you have/What new integration you would like**
Arduino ESP8266 core 2.5.0 has recently been released (https://github.com/esp8266/Arduino/releases/tag/2.5.0) and today platformio shipped the update in the espressif8266 package (https://github.com/platformio/platform-espressif8266/releases/tag/v2.0.0).

Among those fixes appear to be quite a few network-related ones (and apparently the sleep mode issue has been resolved, so no more `power_save_mode: none`)

Todo:
- Evaluate how stable 2.5.0 is. You can help with this! just put this in your config (and report if things are less stable, the same or better):

```yaml
esphome:
  arduino_version: espressif8266@2.0.0
```

- Fix breaking changes (if any). With a quick compile I could not find any (yet)

brandond, 2019-02-21:
First regression is flash/spiffs size is wrong; that appears to be due to the version not being detected properly here: https://github.com/esphome/esphome/blob/dev/esphome/writer.py#L359

```
Linking .pioenvs/powerstrip/firmware.elf
Retrieving maximum program size .pioenvs/powerstrip/firmware.elf
Checking size .pioenvs/powerstrip/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [====      ]  38.9% (used 31848 bytes from 81920 bytes)
Building .pioenvs/powerstrip/firmware.bin
PROGRAM: [======    ]  61.5% (used 468744 bytes from 761840 bytes)
```

brandond, 2019-02-21:
After fixing that so the partition size is correct, the resulting image is still 100k+ larger:

```
# 2.5.0
DATA:    [====      ]  38.3% (used 31384 bytes from 81920 bytes)
PROGRAM: [=====     ]  45.7% (used 468432 bytes from 1023984 bytes)

# 2.4.2
DATA:    [====      ]  40.5% (used 33208 bytes from 81920 bytes)
PROGRAM: [===       ]  34.5% (used 353284 bytes from 1023984 bytes)
```

brandond, 2019-02-21:
Boot loop at wifi:

```
[17:06:04]
[17:06:04] ets Jan  8 2013,rst cause:4, boot mode:(3,7)
[17:06:04]
[17:06:04]wdt reset
[17:06:04]load 0x4010f000, len 1384, room 16
[17:06:04]tail 8
[17:06:04]chksum 0x2d
[17:06:04]csum 0x2d
[17:06:04]v951aeffa
[17:06:04]~ld
[17:06:04][I][logger:122]: Log initialized
[17:06:04][C][ota:480]: There have been 5 suspected unsuccessful boot attempts.
[17:06:04][I][application:055]: Running through setup()...
[17:06:04][C][status_led:018]: Setting up Status LED...
[17:06:04][C][switch.gpio:023]: Setting up GPIO Switch 'Smart Power Strip Power'...
[17:06:04][D][switch:035]: 'Smar Power Strip Power' Turning OFF.
[17:06:04][D][switch:055]: 'Smart Power Strip Power': Sending state OFF
[17:06:04][D][switch:035]: 'Smart Power Strip Power' Turning OFF.
[17:06:04][C][wifi:029]: Setting up WiFi...
[17:06:13]
[17:06:13] ets Jan  8 2013,rst cause:4, boot mode:(3,7)
[17:06:13]
[17:06:13]wdt reset
[17:06:13]load 0x4010f000, len 1384, room 16
[17:06:13]tail 8
[17:06:13]chksum 0x2d
[17:06:13]csum 0x2d
[17:06:13]v951aeffa
[17:06:13]~ld
[17:06:13][I][logger:122]: Log initialized
[17:06:13][C][ota:480]: There have been 6 suspected unsuccessful boot attempts.
[17:06:13][I][application:055]: Running through setup()...
[17:06:13][C][status_led:018]: Setting up Status LED...
[17:06:13][C][switch.gpio:023]: Setting up GPIO Switch 'Smart Power Strip Power'...
[17:06:13][D][switch:035]: 'Smar Power Strip Power' Turning OFF.
[17:06:13][D][switch:055]: 'Smart Power Strip Power': Sending state OFF
[17:06:13][D][switch:035]: 'Smart Power Strip Power' Turning OFF.
[17:06:13][C][wifi:029]: Setting up WiFi...
```

brandond, 2019-02-21:
Tried to turn up the log level from DEBUG to VERY_VERBOSE and the compile failed:
```
In file included from .piolibdeps/esphome-core/src/esphome/wifi_component.cpp:16:0:
.piolibdeps/esphome-core/src/esphome/wifi_component.cpp: In static member function 'static void esphome::WiFiComponent::wifi_event_callback_(System_Event_t*)':
.piolibdeps/esphome-core/src/esphome/wifi_component.cpp:753:36: error: could not convert 'it.Event_StaMode_Got_IP_t::ip' from 'ip4_addr' to 'ip_addr'
format_ip_addr(it.ip).c_str(), format_ip_addr(it.gw).c_str(), format_ip_addr(it.mask).c_str());
^
.piolibdeps/esphome-core/src/esphome/log.h:92:126: note: in definition of macro 'esph_log_v'
#define esph_log_v(tag, format, ...) esp_log_printf_(ESPHOME_LOG_LEVEL_VERBOSE, tag, ESPHOME_LOG_FORMAT(tag, V, format), ##__VA_ARGS__)
^
.piolibdeps/esphome-core/src/esphome/wifi_component.cpp:752:7: note: in expansion of macro 'ESP_LOGV'
ESP_LOGV(TAG, "Event: Got IP static_ip=%s gateway=%s netmask=%s",
^
.piolibdeps/esphome-core/src/esphome/wifi_component.cpp:753:67: error: could not convert 'it.Event_StaMode_Got_IP_t::gw' from 'ip4_addr' to 'ip_addr'
format_ip_addr(it.ip).c_str(), format_ip_addr(it.gw).c_str(), format_ip_addr(it.mask).c_str());
^
.piolibdeps/esphome-core/src/esphome/log.h:92:126: note: in definition of macro 'esph_log_v'
#define esph_log_v(tag, format, ...) esp_log_printf_(ESPHOME_LOG_LEVEL_VERBOSE, tag, ESPHOME_LOG_FORMAT(tag, V, format), ##__VA_ARGS__)
^
.piolibdeps/esphome-core/src/esphome/wifi_component.cpp:752:7: note: in expansion of macro 'ESP_LOGV'
ESP_LOGV(TAG, "Event: Got IP static_ip=%s gateway=%s netmask=%s",
^
.piolibdeps/esphome-core/src/esphome/wifi_component.cpp:753:100: error: could not convert 'it.Event_StaMode_Got_IP_t::mask' from 'ip4_addr' to 'ip_addr'
format_ip_addr(it.ip).c_str(), format_ip_addr(it.gw).c_str(), format_ip_addr(it.mask).c_str());
^
.piolibdeps/esphome-core/src/esphome/log.h:92:126: note: in definition of macro 'esph_log_v'
#define esph_log_v(tag, format, ...) esp_log_printf_(ESPHOME_LOG_LEVEL_VERBOSE, tag, ESPHOME_LOG_FORMAT(tag, V, format), ##__VA_ARGS__)
^
.piolibdeps/esphome-core/src/esphome/wifi_component.cpp:752:7: note: in expansion of macro 'ESP_LOGV'
ESP_LOGV(TAG, "Event: Got IP static_ip=%s gateway=%s netmask=%s",
^
.piolibdeps/esphome-core/src/esphome/wifi_component.cpp:787:69: error: could not convert 'it.Event_SoftAPMode_Distribute_Sta_IP_t::ip' from 'ip4_addr' to 'ip_addr'
format_mac_addr(it.mac).c_str(), format_ip_addr(it.ip).c_str(), it.aid);
^
.piolibdeps/esphome-core/src/esphome/log.h:92:126: note: in definition of macro 'esph_log_v'
#define esph_log_v(tag, format, ...) esp_log_printf_(ESPHOME_LOG_LEVEL_VERBOSE, tag, ESPHOME_LOG_FORMAT(tag, V, format), ##__VA_ARGS__)
^
.piolibdeps/esphome-core/src/esphome/wifi_component.cpp:786:7: note: in expansion of macro 'ESP_LOGV'
ESP_LOGV(TAG, "Event: AP Distribute Station IP MAC=%s IP=%s aid=%u",
^
```

OttoWinter, 2019-02-21:
That firmware size increase is not good :( Thanks for spotting it!

I did some digging and I think I have found some options to get it down to a comparable level (mostly based on these settings [here](https://github.com/arendst/Sonoff-Tasmota/blob/development/platformio.ini#L69)):

`-fno-exceptions`: ESP8266 arduino core added support for exceptions in 2.5.0 - their changelog says it's disabled by default but adding this option saves around 80kb!

`-DNDEBUG` saves 3kb, we could consider disabling this when logger is not in verbose mode (`-DNDEBUG` disables ESP-internal logging)

We could also look into switching the [lwip variant](http://docs.platformio.org/en/latest/platforms/espressif8266.html#lwip-variant), default appears to be LOW_MEMORY. I tries HIGH_BANDWIDTH but I did not see any runtime difference - I think this only increases the LWIP socket buffer sizes, which could result in fewer MQTT missed publish packets and less "Message could not be transmitted because it did not fit into TCP buffer" messages.

`-mtarget-align` does not appear to do anything
`-DVTABLES_IN_FLASH` and `-lstdc++` are on by default.

So with these optimizations I got it down to at least not increase firmware size too much (anything that allows OTA for 1mb devices and a few components is fine I think)

```
# 2.4.0
DATA:    [====      ]  39.9% (used 32712 bytes from 81920 bytes)
PROGRAM: [===       ]  31.6% (used 329856 bytes from 1044464 bytes)
# 2.5.0 default:
DATA:    [====      ]  37.9% (used 31024 bytes from 81920 bytes)
PROGRAM: [====      ]  41.8% (used 436892 bytes from 1044464 bytes)
# 2.5.0 optimized:
DATA:    [====      ]  37.2% (used 30492 bytes from 81920 bytes)
PROGRAM: [===       ]  34.2% (used 357336 bytes from 1044464 bytes)
```

Of course if the 2.5.0 doesn't appear to improve wifi performance there's little reason for us to upgrade.

------------

`[17:06:04]wdt reset`

Oh noes, that sounds a lot like this issue https://github.com/esphome/issues/issues/58#issuecomment-465320220 (you commented there, did you try the instructions there?). I have not seen this issue myself, will try later

------------

Re verbose level: Ah yes that is because of lwip upgrade (and ipv6 support). I think I will just remove the offending lines as I did with the ESP32

OttoWinter, 2019-02-21:
I think I'll cherry-pick the fixes for these issues into 1.11.0, so that if someone wants to test 2.5.0 they don't have to use the dev version.

OttoWinter, 2019-02-21:
Ok, just found https://github.com/esp8266/Arduino/issues/5736 and https://github.com/esp8266/Arduino/issues/5513 - so 2.5.0 upgrade is probably not worth it right now.

Also experiencing the wdt reset issues now, after some quick debugging it looks like the offending line is this one: https://github.com/esphome/esphome-core/blob/dev/src/esphome/wifi_component.cpp#L41 (at least uncommenting it seems to push the wdt reset to a later point). It looks like delay is entirely broken in this build - or esphome is using the internal wifi methods wrong (but I didn't see any related changes in the ESP8266WiFi for this)

OttoWinter, 2019-03-05:
Ok so the `delay` issue appears to have been an issue with the new toolchain that platformio bundled with the release - no idea why that could impact things.

2.5.0 appears to be quite stable now with dev and works quite well too: https://github.com/esphome/esphome/pull/469

I'll ask more users to try 2.5.0 and if it works I think we can enable it for 1.13.0

OttoWinter, 2019-03-24:
Closing because 2.5.0 can be safely used now, but early data also shows that 2.5.0 is worse than 2.4.0 in certain places, so we will have to wait for another release.

closed 2019-03-24 by OttoWinter