esphome/feature-requests #73: Allow opt-in storing restore data in flash on ESP8266

OttoWinter, 2019-02-21:
**Describe the problem you have/What new integration you would like**

ESPHome stores all restore data in RTC memory, which is great because it doesn't wear out flash. but unfortunately RTC memory doesn't persist across power cycles.

Proposal: Add an option in `esphome:` config that allows users to enable storing this restore data into flash. It's good not to enable it by default because it can wear out flash, but if the user is aware of the issue and enables it by themselves then it's more ok. Optionally this option could also be given an "offset" to store things in the flash, so that if the first block wears out (if it ever does), the user can move to the next block.

API wise this is probably not too hard, just extend the appropriate methods in `esppreferences.h/.cpp` to store to flash.

Things to do:

- Figure out how the flash API works - or alternatively just use EEPROM utility
- How does this work with ESPHome disabling SPIFFS? Any special offset that can still be written?

markcarline, 2019-02-21:
+1 from me for this. Would be amazing to have it.

From what I have read flash can be written to around 100,000 times so based on say 10 updates a day (lights on / lights off) that's still 28 years usage.

MrPasty, 2019-02-21:
+1 from me on this too

OttoWinter, 2019-02-23:
> From what I have read flash can be written to around 100,000 times so based on say 10 updates a day (lights on / lights off) that's still 28 years usage.

Yes with 10 updates per day that would not be a problem. The problem only starts when people use some of the restoring components (like switch, light etc) for internal automations, for example a custom light effect - if not configured correctly - could result in many hundred write cycles per day while it's active. That's why it won't be the default, I don't want to damage your HW without your explicit permission.

For implementation (more of a note-to-self):
Looks like [EEPROM](https://github.com/esp8266/Arduino/blob/master/libraries/EEPROM/EEPROM.cpp) is just a very thin wrapper around `spi_flash_*` functions. So I will probably just use `spi_flash_*` myself (not through the wrapper); that will also be easier since ESPPreferences and SPI works with uint32_t whereas EEPROM works with uint8_t.

Only potential problem is where to store the data buffer. We need to store a buffer for reading/writing somewhere, but how do we know its size? I think one hacky but cool solution would be to store everything in RTC memory as before, but just add `spi_flash_read/write` calls after each access (minimal work)

markcarline, 2019-02-23:
All looks very good would love to help to test out when you have something. Watching this closely. 

keep up the great work I just migrated all my ESP devices to esphome and I'm using the on_boot work around for turn in & colour for now, when this feature comes in it will be brilliant

closed 2019-02-26 by OttoWinter

markcarline, 2019-03-04:
This is great, any ideas when this will be part of the logical release? 

Thanks for the work with esphome - it's brill!