esphome/feature-requests #74: Set default colour or effect for neopixel light?

MrPasty, 2019-02-21:
Is there any way we can specify what colour the light will default to when it is turned on? at the moment mine seems to turn on to a yellowish colour but i would like it to be a specific colour or effect?  I expect this can be done with an automation in HA but i would like to just be able to toggle the on off state and it be the correct colour.

Great project - keep up the good work.

Thanks

markcarline, 2019-02-21:
If this one is delivered and it "remembers" the light / colour state then guess it would do, yes?....

https://github.com/esphome/feature-requests/issues/73

OttoWinter, 2019-02-21:
If your colors are a bit off you can set a color correction (see docs).

Otherwise I would agree https://github.com/esphome/feature-requests/issues/73 is the real issue here. Else this should be solved in HA or esphome custom code.

closed 2019-02-21 by OttoWinter