esphome/feature-requests #75: Move component+platform docs

OttoWinter, 2019-02-21:
**Describe the problem you have/What new integration you would like**

In some cases esphome uses a component+platform config schema, for example like this:

```yaml
dallas:

sensor:
- platform: dallas
  # ...
```

Currently, the docs usually have it split up into two docs pages. That is confusing and taking up space in the front index. Let's merge each of these docs into one page each.

Affected docs:

- [ ] Remote Transmitter
- [ ] Remote Receiver
- [ ] PCA9685
- [ ] ESP32 Touch
- [ ] PN532
- [ ] RDM6300
- [ ] MY9231
- [ ] ADS1115

Other components that should be merged:

- [ ] SSD1306 SPI+i2c
- [ ] LCD display GPIO+PCF9574
- [ ] FastLED clockless+spi

FrengerH, 2019-02-26:
I created a PR that merges the documentation of the Dallas component. 
[https://github.com/esphome/esphome-docs/pull/179](url) 
If this is like you had in mind I can do the other components as well.

closed 2019-02-26 by OttoWinter

FrengerH, 2019-02-26:
I'll merge the other components as well. @OttoWinter Do you prefer a PR per component, or one big PR with all components merged?

OttoWinter, 2019-02-26:
@FrengerH Uhm if it's basically only a reshuffle of the existing stuff like here a single PR is fine 👍

FrengerH, 2019-02-26:
Just filed a PR for the other components mentioned in the request.
[https://github.com/esphome/esphome-docs/pull/183](url)