esphome/feature-requests #77: More displays support, epaper and tft

erazor666, 2019-02-23:
**Describe the problem you have/What new integration you would like**
Wondering if its possible to support displays like these:
https://www.aliexpress.com/item/1-8-inch-Full-Color-128x160-SPI-Full-Color-TFT-LCD-Display-Module-ST7735S-3-3V/32880822720.html
epaper:
https://www.aliexpress.com/item/TZT-1-54-Inch-Epaper-Module-E-paper-E-Ink-EInk-Display-Screen-SPI-Support-Global/32870097847.html

**Please describe your use case for this integration and alternatives you've tried:**
N/A

**Additional context**
From dealing with some displays the last few days i'm guessing much of the basis for supporting this type of display is already there with some adaption. These are cheap pixel based displays that can do color, seems interesting. The last one may already be almost supported, at least in 1.10 epaper screen was missing color support.

OttoWinter, 2019-02-23:
Each of these needs their own integration (most of them aren't hard to write, the drawing API already exists).

ST7735S is tracked here: https://github.com/esphome/feature-requests/issues/1

nagyrobi, 2022-06-17:
https://esphome.io/components/display/st7735.html