esphome/feature-requests #80: Support for ESP-01S 1MB and ESP-01 4MB (1MB IC manually replaced by 4MB IC)

PjotrGrn, 2019-02-23:
**Describe the problem you have/What new integration you would like**
 Currently, only the ESP-01 512KB seems to supported in ESPHome. Could the ESP01S (1MB) or even a ESP-01 with manually replaced (upgraded) memory IC (e.g. 4M instead of 512K or 1 M) also be supported in the future? Or can I modify (yaml file?) ESPHome myself to do this (yes, I am a novice in the ESPHome scene.. ;-)) 

**Please describe your use case for this integration and alternatives you've tried:**
 Currently, only hardware support for ESP 8266 ESP-01 512KB

**Additional context**

OttoWinter, 2019-02-23:
The 1mb option is called the "Generic ESP8266" - for 4mb there's probably also another board (see platformio website) and replace that in the config file under board. Otherwise you can use nodemcu as board because that has 4mb (board option is mainly for pin aliases)

OttoWinter, 2019-03-05:
I will close this feature request as this is already supported

closed 2019-03-05 by OttoWinter