esphome/feature-requests #84: New Component - Timer

jchambers2012, 2019-02-25:
**Describe the problem you have/What new integration you would like**
I would like a timer sensor.  This would be triggered by automation or Home Assistant and after the timer runs out, it would trigger an event on the ESP.  The timer would run on the ESP itself for internal automation and call back without the need to talk to Home Assistant. 

**Please describe your use case for this integration and alternatives you've tried:**
When a cabinet door is open; the ESP turn on an LED light strip. After the door is closed; the ESP turn off the LED strip. I am looking to have a fail safe to turn off the LEDs after X hours/minutes.  I want this timer to run on the ESP itself so it can still perform the "turn off" automation if the HA crashes or goes offline.

**Additional context**

Possible config:
time:
  - platform: timer
    seconds: <number>
    minutes: <number>
    hours: <number>
    self_reset TRUE(Default, Optional) <- would reset the counter after it expires
    self_stop TRUE(Default, Optional) <- if the self_reset is set to true, TRUE: set the timer to a stop state FALSE: time would keep running
    disable_switch: switch.myswitch (Default, Optional) <- switch exposed to Home Assistant for disabling the timer, could also be allowed to accept a GPIO or other binary components
    on_expire:
        then: (actions) 
        - light.turn_off: 
            id: dining_room_cabinet_inside_a
        - light.turn_off:
            id: dining_room_cabinet_inside_b
          - timer.reset_stop: my_timer

Automation functions:
-timer.start <-start the countdown counter
-timer.stop <- stop the countdown counter
-time.reset <- reset the countdown time to being full; counter would remain in a start or stop state
-timer.reset_start  <- reset the countdown time to being full; counter would go to start state
-timer.reset_stop  <- reset the countdown time to being full; counter would go to start stop
	
Real world example:
Switch:
  - platform: template
    name: "Dining Room Cabinet Auto Light Turnoff"
    optimistic: true
    id: dining_room_cabinet_automation_autoturnoff
time:
  - platform: timer
    id: my_timer
    name: “Auto Lights turn off”
    disable_switch: switch.dining_room_cabinet_automation_autoturnoff
    hours: 2
    on_expire:
        then:
        - light.turn_off: 
            id: dining_room_cabinet_inside_a
        - light.turn_off:
            id: dining_room_cabinet_inside_b
        - timer.reset_stop: 
            id: my_timer
binary_sensor:
  - platform: gpio
    name: "Dining Room Cabinet Left Door"
    pin:
     number: D2
    device_class: door
    id: dining_room_cabinet_door_left
    filters:
      - heartbeat: 60s
    on_release:
      then:
        - light.turn_off:
            id: dining_room_cabinet_inside_a
        - light.turn_off:
            id: dining_room_cabinet_inside_b
        -timer.stop: 
            id: dining_my_timer_cabinet_inside_b
    on_press:
      then:
        - light.turn_on:
            id: dining_room_cabinet_inside_a
            brightness: 80%
        - light.turn_on:
            id: dining_room_cabinet_inside_b
            brightness: 80%
        -timer.reset_start: 
            id: dining_my_timer_cabinet_inside_b

OttoWinter, 2019-03-05:
I believe scripts can do exactly that.

```yaml
script:
  id: my_timer
  then:
  - # on_start
  - delay: 30min  # timer length
  - # on_end

# start/reset timer
- script.stop: my_timer
- script.execute: my_timer

# stop timer
- script.stop: my_timer
```

I definitely don't like the suggested config. It's way to complicated for normal setups, and feels a lot like the HA style of configs (with way too many options that thus confuse users).

jchambers2012, 2019-03-05:
Should work for what I need. I was just building out the new component config with a full feature set nothing was required.

2 QQ: about the Script Component, not 100% yet on how automation runs under the hood:
-> if the script.execute function is called twice, 10 minutes apart, does the old script get canceled or does it start a second "timer"? 
-> Are there any DOC on the script option or is it just being built out?

OttoWinter, 2019-03-05:
When a second script is started the old one still runs. However, if you call `.stop` before it's the same as having a single running instance.

> Are there any DOC

See automation page.

jchambers2012, 2019-03-05:
I see it now, nothing was showing up in the search function of the website

Thanks for the info!

closed 2019-03-05 by jchambers2012

glmnet, 2019-05-07:
@ottowinter in the need of a timer I stumbled upon this. 
I needed a way to know if a timer was running, so I could have created a global and set/reset but I found an undocumented `script_is_running()` which nailed it. 
Do you think this can be documented and hinted to be used as timer?

OttoWinter, 2019-05-07:
@glmnet That has been added in dev - it doesn't exist for 1.12. Docs will be part of 1.13 docs writing.