esphome/feature-requests #85: [Feature Request] Support CD74HC4067 16 Channel Analog / Digital Multiplexer...

allnerag, 2019-02-26:
**Describe the problem you have/What new integration you would like**

It would be great to be able to add the multiplexer to Analog or digital and then have more ports to use for integration. By doing this we can even use D1 Mini with expander boards to be able to expand even more.  This can bring the count down of connected devices on your network.

**Please describe your use case for this integration:**

I want to use this for alarm port monitoring (Digital for new alarm system only using This, or connecting to an analog port for piggy back on Alarm systems with resistors)

With homes that you want to use presence detection, many homes have large installations of Alarm systems and we can use the PIR, Door Magnets to do automation with.  Why add more devices if you can use existing once.

I have an IDS x64 that can do 64x zones and unable to connect directly to it, so piggy back is the only way.

**Additional context**

Here is an Arduino Library for this;
https://github.com/waspinator/CD74HC4067

Thank you,
Andrew

allnerag, 2019-03-03:
here is more info;

Have you look for this feature in other issues and in the wiki?
Yes, but did not see any mention of it.

Is your feature request related to a problem? Please describe.
Yes, Certain Devices only have a specific amount of ports and most of them only have one Analog port, except the ESP32. The problem is that if you want to connect multiple digital devices on one ESP we can do it to a specific amount, D1 Mini only about 8. If you go to the Analog side of things, we can only go to ESP32 and it is not fully supported yet on TASMOTA only on ESPHOME. The problem that I have and I assume more people have this issue is that if you need to use a Voltage follower and a comparetor, you need to use 1x voltage follower and one comparetor for each Analog or digital line you connect directly to an ESP. If you use a Multiplexer like "CD74HC4067 16 Channel Analog / Digital Multiplexer" you only need one of each and use the multiplexer ability of channels through one of the ports. By doing this you also bring the count down of connected devices on your network.

I want to use this for alarm port monitoring (Digital for new alarm system only using This, or connecting to an analog port for piggy back on Alarm systems with resistors)
With homes that you want to use presence detection, many homes have large installations of Alarm systems and we can use the PIR, Door Magnets to do automation with. Why add more devices if you can use existing once.
I have an IDS x64 that can do 64x zones and unable to connect directly to it, so piggy back is the only way.

Describe the solution you'd like
Add 2x Options to the Selection of Devices to enable the feature to use a Multiplexer. Only enable Analog Port 0 and Digital Port 0 for the selection. This means that you can do Analog Read if it is on Analog pin and only digital if you select it on the digital port.
Because it is a 16x Multiplexer you need 4x Digital port to address each of the ports depending on what port are pulled high. This is for the same if you use the Multiplexer for Digital or Analog. The pin on D0 or A0 is called the Common Port, and depending what that port is you can do certain things.
If it is on Digital port it will just show "High or Low". If it is on Analog port it can only do Analog read). These ports can be set for Input or Output.

So the options on D0 could be;

CD74HC4067 16 Channel Digital Multiplexer Input
CD74HC4067 16 Channel Digital Multiplexer Output
This will automatically change D1-D4 to Digital Out Only to use with the Multiplexer. This will Allow only Digital In or Out.
So the options on A0 could be;

CD74HC4067 16 Channel Analog Multiplexer Input
CD74HC4067 16 Channel Analog Multiplexer Output
This will automatically change D1-D4 to Digital Out Only to use with the Multiplexer. This will Allow only Analog Read and write.
The info per Multiplexer port will come through channels (1-16). This can be be set using MQTT topics per channel.

Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.
I did look at using each digital port on ESP8266 (Node MCU), this will mean I have to use 8x ESP8266 Node MCU's to be able to do the same thing and Will also have to use 32x Voltage followers and 32x Comparators to do the same. for Analog ports I have to use 4x ESP32's with the same amount of voltage followers and Comparators as the digital one. I cannot use the ESP32 with TASMOTA as it does not have Analog shipped to it yet, I can only use ESPHOME for the Analog side. I can use an Arduino Mega but then I loose the easy integration via TASMOTA or ESPHOME, or I have to Use RX/TX to an ESP-01 or ESP8266 / ESP32 for that. That will also mean I have to use custom Arduino Code for that.

Additional context
Here is an Arduino Library for this;
https://github.com/waspinator/CD74HC4067

(Please, remember to close the issue when the problem has been addressed)

rmorenoramos, 2020-10-05:
Hello,

I have created the CD74HC4067 component to work as an analog sensor multiplexer for a project of mine. I just tried it and works, but it might need a bit of cleaning and adaptation to some other platforms.

Let me know how I can be of any help.

Cheers

[cd74hc4067.zip](https://github.com/esphome/feature-requests/files/5325788/cd74hc4067.zip)

toomuchcookies, 2021-01-24:
@rmorenoramos first of all thanks for the code. I tried it out but i don't seem to be getting the entities for the sensors. Could you please check the bug report i posted above? Maybe you can help?

rmorenoramos, 2021-01-28:
Hi @toomuchcookies , 

I guess it was my mistake not to include any example on how to use it...

Can you try with this configuration?

cd74hc4067:
  s0_pin: D1
  s1_pin: D2
  s2_pin: D3
  s3_pin: D4     
  adc_pin: A0 

sensor:
  - platform: cd74hc4067
    number: 0
    id: LivingRoom
    update_interval: 1s
  - platform: cd74hc4067
    number: 1
    id: Kitchen
    update_interval: 1s

nickizgr8, 2021-03-28:
Hi all, I have made a 10 slot battery capacity tester for charging and discharging 18650 batteries using some tp4056 chargers and some capacity test modules like these [https://www.amazon.com/gp/product/B0824HR66Z/ref=ppx_yo_dt_b_asin_image_o09_s01?ie=UTF8&psc=1](url) . The modules display voltages but I want to add some sensors to the project that will measure temperature and voltage so I can graph their relationships in home assistant at different stages while charging/discharging along with set automations to warn me if any cells get to hot). 
[image]
So far I have the temperature side successfully setup reporting values from 10 separate ds18b20 sensors that I'll put in a small drilled out hole on the bottom of each cell holder so the sensors will be touching the cells for a good temp measurement. The sensors are connected in parallel to each other via pin D3, 3.3v and GND on a Wemos D1 Mini with a 4.7k pullup resister between data and 3.3v; all running on ESPHome. I used InfluxDB and Grafana to get these great graphs by following this tutorial here [https://www.youtube.com/watch?v=VkBEU3DmtjY](url)
[image]
The sensors are all next to each other on a breadboard and giving slightly different temp readings so they'll need some calibrating after they're placed in their final positions in the cell holders but getting to this point has taken me over 2 days (I am very much a novice). For now I want to move on to the voltage part which is stumping me...
My plan: to measure the cell voltages I'll create voltage divider circuits for each battery and connect the lowered voltages to a cd74hc4067 16-channel analog multiplexer module like this one [https://www.amazon.com/gp/product/B07K7JF3HX/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1](url) all wired to the same D1 Mini as the temp sensors and then map the values to typical voltages of 18650 cells. I found @rmorenoramos 's zip file above and think this is what I need to read the 10 voltages but how to I go about adding this file into home assistant (OS)? I have worked with my configuration.yaml directly before (a little) but do not see a "custom" folder.  I have samba share setup, so I'm wondering if I can create the required folder there and just drag the extracted zip file into it? 
[image]. Is there a step by step for adding the CD74HC4067 component to my home assistant? 
IF I can get the multiplexer part setup how do I go about telling ESPHome which pin is which? The 16 pins on the multiplexer module are labeled C0 thru C15 but in the example given by @rmorenoramos above there is no pin given for each sensor, just a 'number'.  I'm assuming the numbers correspond to the C# pins on the multiplexer module? Also, all the ds18b20 sensors are connected to D3 so in the setup for the CD74HC4067 could I designate a different pin for the s2_pin, which in the code above is using D3? If not I can change that on the temp sensor side in ESPHome I suppose, it shouldn't mess anything up for the temp sensors, I hope. 
It's probably obvious how much if a noob I am. posting questions is a bit intimidating but I've spent a fair amount of time researching and can't seem to get this part. I saw there was recent comments and a related discussion here [https://github.com/esphome/issues/issues/1785](url) opened by @toomuchcookies but where this discussion is more about the install part I thought I might try my luck here.
I greatly appreciate any and all help or recommendations with getting the voltage reading part running. Unrelated but if anyone knows how to map the voltage values and/or has insight on the best resistor combo I should use in the voltage divider to measure an 18650's voltage that would be helpful too. I know equal values will cut voltage in half but would 2x220 ohm be better then 2x10k ohm?

Thank you,
-Nick

p.s. I understand the shortcomings of this tester as well, not being able to apply a constant current load, i.e. as the battery voltage decreases so will the current, will result in not providing the best test results but I'm ok with that. Additionally I understand there are probably many voltage/power measuring modules out there but I would like to do this with voltage divider circuitry as it is a better learning experience and more fun for me.

asoehlke, 2021-04-11:
Hi,
@rmorenoramos, thank for the code as well! I also tried it out and had the same issue as @toomuchcookies with the entities not showing up in Home Assistant. I added the code and a fix for that in my fork (git@github.com:asoehlke/esphome.git). The problem was that each of the sensors needs to have its own unique ID. So it works for me now:
[image]

My usecase is a set of SCT-013-050 power clamps, so I also added some code to average the measurements. 

@nickizgr8: Bringing this into the HA plug-in is possible using the esphome_version configuration as described in the plug-in doc. I opened the pull request above, so setting esphome_version: asoehlke:add-CD74HC4067-multiplexer should include the code in the plug-in.

The PR is still work in progress, though. The linter still has some complaints :-)
Most are very minor, but there is one serious issue: components should not call analogRead directly. It is working fine and also done so by the ADC converter component. It seems to be a design issue: Only the ADC component may use analogRead directly, other components should use this sensor instead of duplicating the code.  This is the way it is done for e.g. the [resistance sensor](https://esphome.io/components/sensor/resistance.html?highlight=resistance) and [ct clamp sensor](https://esphome.io/components/sensor/ct_clamp.html?highlight=clamp). @nickizgr8: the resistance sensor documentation also explains some of your questions about the voltage reading.

So getting this component officially into esphome requires a mayor change to the code. 
I am still not quite certain how to do this best, ideally the component would have a set of sensors attached and would multiplex their output. 
For nickizgr8's case this would be not too difficult. But in my case, I would need to switch between ct clamps sensors, which already use a more complex logic in the loop() function to do the averaging of the readings. 
This would solve also another issue with the current implementation, which is that the update() function should not block. Currently, the switching the multiplexer and doing the measurements takes too long for a single update call. It works fine for me, but could be an issue in a more complex set-up.

rbonat, 2021-11-17:
I'm also waiting for merge of this component in next release!
Meanwhile I've created simple configuration, that I suppose can support any analog multiplexer, not only CD74HC4067 (in my case I've tested it with CD4051). It is using only lambda filter, no additional components, no installation, just this yaml.
So I share it in case anyone is interested.
In this example there are 8 analog inputs selected by 3 adress lines, but it can be easily modified to support whatever you need.
So maybe with a little effort this (CD74HC4067) component can be generalized as well to support any simple analog multiplexer/switch?
```
substitutions:
  dev_name: test-board

logger:
  logs:
    adc: ERROR

# define outputs connected to multiplexer controll lines
output:
  - platform: gpio
    pin: 14
    id: ADR0
  - platform: gpio
    pin: 12
    id: ADR1
  - platform: gpio
    pin: 13
    id: ADR2
    
# define template sensors for each multiplexer input line
sensor:
  - platform: template
    name: "${dev_name} A0"
    id: "Analog_0"
  - platform: template
    name: "${dev_name} A1"
    id: "Analog_1"    
  - platform: template
    name: "${dev_name} A2"
    id: "Analog_2"    
  - platform: template
    name: "${dev_name} A3"
    id: "Analog_3"    
  - platform: template
    name: "${dev_name} A4"
    id: "Analog_4"    
  - platform: template
    name: "${dev_name} A5"
    id: "Analog_5"    
  - platform: template
    name: "${dev_name} A6"
    id: "Analog_6"    
  - platform: template
    name: "${dev_name} A7"
    id: "Analog_7"    

# define ADC sensor and apply lambda filter implementing multiplexer logic
  - platform: adc
    pin: A0
    name: "${dev_name} AnalogValue"
    update_interval: 10s
    internal: true
    filters:
      - lambda : |-
          static byte multiplex_counter = 0;
          if (multiplex_counter==0) id(Analog_0).publish_state(x);
          if (multiplex_counter==1) id(Analog_1).publish_state(x);
          if (multiplex_counter==2) id(Analog_2).publish_state(x);
          if (multiplex_counter==3) id(Analog_3).publish_state(x);
          if (multiplex_counter==4) id(Analog_4).publish_state(x);          
          if (multiplex_counter==5) id(Analog_5).publish_state(x);          
          if (multiplex_counter==6) id(Analog_6).publish_state(x);
          if (multiplex_counter==7) id(Analog_7).publish_state(x); 
          multiplex_counter++;
          if(multiplex_counter==8) multiplex_counter=0;
          if (multiplex_counter & 0b001) id(ADR0).turn_on(); else id(ADR0).turn_off();
          if (multiplex_counter & 0b010) id(ADR1).turn_on(); else id(ADR1).turn_off();
          if (multiplex_counter & 0b100) id(ADR2).turn_on(); else id(ADR2).turn_off();
          
          return 0;
      - delta: 100  # to prevent processing raw analog sensor  
```

closed 2022-01-11 by jesserockz