esphome/feature-requests #86: Enable fallback wifi AP by default

OttoWinter, 2019-02-27:
**Describe the problem you have/What new integration you would like**

ESPHome seems to be used with more Sonoff/Tuya devices now, and those are a pain to flash.
So when you enter wrong WiFi credentials or ESPHome's wifi connections have a bug, you will not be able to flash the device OTA again.

So ESPHome should probably create a fallback AP by default in the setup wizard so that it's at least possible to flash the ESP when WiFi does not connect.

The WiFi AP+STA mode probably will need to be changed for this a bit, ESPHome currently starts an AP immediately when STA does not connect. Likely having an AP active will decrease WiFi sta scan+connect performance, so the AP mode would need to get a cooldown timer. So that the AP only starts after a set time period (1min or so) of not being able to connect to STA.

Also just enabling an AP won't be the only thing that needs to happen. While the fallback AP allows you to connect to the ESP, you can't change anything - so the system running ESPHome would also need to connect to this AP. But as ESPHome often runs on Hass.io / docker setups it's not possible to connect *them* to the AP. 

Either we need to enable the `web_server` by default too (not a huge fan because it takes resources), or we need to get the OTA process to run on other systems too without having to install ESPHome - the best location would probably be ESPHome-Flasher.

dlashua, 2019-03-06:
Creating a fallback AP is a nice touch, but it's also not all that difficult to add it into a base/common configuration so that users who don't need this functionality don't have to include it.

The biggest problem is actually flashing the device once it's in AP mode. ESPHome-Flasher seems like it would handle this okay as long as the networking aspects are obvious (i.e. what IP address does the device have when it's in AP mode?). However, a smartphone App would make this REALLY simple.

But, as you stated, even more simple than that, would be the ability to configure ssid and password via a simple web interface. Then any device that could connect to the AP could likely configure it. Even if it was just an HTTP endpoint with no UI to keep the footprint low (.. i.e we hit http://10.1.4.92/api/wificonfigure?ssid=MySSID&pass=myPass and that sets those config elements and restarts the device)

ToRvaLDz, 2019-03-12:
In AP mode the ip address is 192.168.4.1, but the Web endpoint just to setup WiFi connection is the best and fastest solution.

dougmoscrop, 2019-04-16:
Would another alternative be BLE + companion app for setup?

OttoWinter, 2019-04-21:
@dougmoscrop Nah, for many reasons some of which:
- Would need a full app -> more work
- Only available on ESP32 -> most users are using ESP8266s
- BLE is quite unstable together with WiFi
- BLE takes a humongous amount of program space on the ESP32

emouawad, 2019-05-09:
Thanks for the great work you guys have done so far, ESPHome is making everything easier.

I believe there is still room for improvement in the Wifi configuration - changing network password shouldn't need a re-flashing - any method to achieve this target with conjunction to AP would add much needed value.

Would something like @dlashua proposed work?
- "configure ssid and password via a simple web interface"
OR
- "just an HTTP endpoint with no UI /  http://10.1.4.92/api/wificonfigure?ssid=MySSID&pass=myPass"

Thanks

rp3tya, 2019-05-09:
The ability to change wifi ssid/password without reflash is indeed missing. What do you say about [SmartConfig](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/network/esp_smartconfig.html)? Mobile app is not a problem, nor memory footprint as functions are already compiled in the wifi library.

OttoWinter, 2019-05-11:
There's a big problem with changing SSID/PSK without reflashing:

**How should the device know if it should use the values in the binary or from flash?**

That is what has kept this from going forward. Sure yes the ESP can store WiFi info in the flash area - but the problem is that we don't know if we should use the values given in the compiled firmware or the ones stored in some flash section.

Any web server UI, app etc solution will suffer from this.

When the user uploads a binary, the wifi settings given in there should *always* override anything set previously.
When the ESP can't connect and creates an AP for user config, the device will need to write the new WiFi SSID/PSK somewhere. Until the device is re-flashed that data must be used.
But when the user re-flashes the device, the temporarily set SSID/PSK needs to be cleared again.

If that can be overcome, the way how the wifi settings should be entered is pretty clear to me: A simple captive portal. Sure apps etc are also possible, but a captive portal is already not much work and I wouldn't want to have to install an app just to configure my ESP.

> What do you say about SmartConfig?

It's one possibility, but the problem is not there. Building a captive portal is a simple task. The problem is in storing/retrieving the wifi settings (and single source of truth)

> nor memory footprint as functions are already compiled in the wifi library.

Haven't checked, but I think it's likely the linker removes them if the symbols are unused.

rp3tya, 2019-05-11:
I get it now... And the 'when?' is also a problem: How should the device know if it should create an AP for user config? Would not like to expose my switches to passers-by in case of a router crash or restart, etc (don't know if there was a discussion on this already)

OttoWinter, 2019-05-11:
> How should the device know if it should create an AP for user config?

Just a user-configurable timeout.

When creating a new config, the dashboard would automatically set up the config with the name of the ESP as the SSID and a random string as the password.

closed 2019-06-09 by OttoWinter