esphome/feature-requests #87: eQ-3 Bluetooth Thermostat support

bbbenji, 2019-02-27:
**Describe the problem you have/What new integration you would like**
Support for forwarding bi-directionally data from/to the [eQ-3 Bluetooth Thermostat](https://www.eq-3.com/products/eqiva/bluetooth-smart-radiator-thermostat.html)

**Please describe your use case for this integration and alternatives you've tried:**
Been using [bt-mqtt-gateway](https://github.com/zewelor/bt-mqtt-gateway) which does exactly what the name states

**Additional context**
[Python library and command-line tool for EQ3 Smart Bluetooth thermostats
](https://github.com/rytilahti/python-eq3bt)

snizzleorg, 2019-03-03:
There is a esp firmware that does exactly this: 
https://github.com/softypit/esp32_mqtt_eq3

egon0, 2019-03-08:
This would be a great addition. I have esphome nodes with sensors in almost every room. Currently I have everywhere a second ESP32 for only controlling the eq3 thermostats. Would nice to merge the sensors-nodes with the eq3-controlling nodes

inforsgit, 2019-03-13:
Just for more context on this - there's also an ESP32 implementation here: https://github.com/softypit/esp32_mqtt_eq3

barthand, 2019-03-15:
I would love to have EQ3 integrated with esphome in order to easily expose it to Home Assistant.

Being not familiar with esphome architecture, does it take a lot of work to integrate aforementioned ESP32 implementation (https://github.com/softypit/esp32_mqtt_eq3) with it?

inforsgit, 2019-03-15:
I'm not 100% familiar with the inner workings of ESPHome but I think the integration I mentioned would more be for reference than directly porting it into ESPHome.

I was thinking that the [Xiaomi MiFlora implementation in ESPHome](https://esphome.io/components/sensor/xiaomi_miflora.html) (covered by the BLE Tracker code: https://github.com/esphome/esphome-core/blob/dev/src/esphome/esp32_ble_tracker.h) could serve as a starting point - that at least shows how a BLE device can be integrated in ESPHome. The major difference is that it only reports received values, whereas the EQ3s require two-way communication.

inforsgit, 2019-03-15:
Just to add even more background info (sorry for not having a solution as yet) - a [Home Assistant component exists for the EQ3 thermostats](https://www.home-assistant.io/components/climate.eq3btsmart/) but of course that only works if the home assistant server (e.g. an RPi 3) is in range of all of the heaters. I guess that, like me, that's not the case for you and the dream would be to have ESP32s around the house, each running ESPHome and acting as bridges to Home Assistant.

If so then I think we need to consider that ESPHome would present a "Climate" component to Home Assistant, in the same way that it can currently present a "Fan" component. So I assume we'd need to create a thermostat/TRV "Core" in ESPHome and then have eq3 TRVs as an extension of that...anyone think that's a good or bad idea? I'll keep thinking myself and try and flesh out the idea!

OttoWinter, 2019-03-15:
@inforsgit This integration would be a lot of work, first a new esp32_bt base would need to be created (for communicating with normal BT devices) with an API that several devices like the EQ3 could hook into.

I would not recommend creating a climate component in esphome - the HA model of a climate entity is seriously broken from an architecture standpoint (there are tons of properties that are improperly documented). Plus adding a new component to esphome is a whole bunch of work because the base component would need to be created, then MQTT and native API layers, then aioesphomeapi needs to be updated and finally HA - a whole bunch of work. So I would highly recommend exposing everything as individual switches/sensors/etc. If the user wants they can still make a template in HA that converts these to a climate entity.

animavitis, 2019-03-16:
Maybe just add custom component: **Custom BLE device**, this component will have **BLE read characteristic** and **BLE write characteristic** function. This can be enough to create eQ-3 (or any other ie. Sygonix) thermostat template

Ilpredatore, 2019-08-30:
Hi @OttoWinter, i saw that you inplement climate components in ESPHome, do you have intention of develop integration for eQ-3 TRV platform in the future?

Thank you very much.

OttoWinter, 2019-08-31:
@Ilpredatore tbh, I don't have any plans for this device in particular. I have no need for a bluetooth thermostat and also don't own this unit. If someone implements it, I would be happy to accept it though.

pieter, 2019-10-13:
I've started implementation of a library that can control the EQ3 thermostats here: https://github.com/pieter/ESP-EQ3

It's fairly simple (depends on Arduino & a BLE library) and incomplete, but should be easy to expand with the necessary functionality for ESPHome. [Here](https://github.com/pieter/ESP-EQ3/blob/master/src/eq3thermostat.h) is the header -- you initialise it with a MAC address, then call getTemperature/setTemperature and getMode/setMode.

I hope to create a custom component for this at some point once I find some more time, if nobody beats me to it.

ayufan, 2019-12-04:
You might look at my custom components. I spend last week on trying to get this working: https://github.com/ayufan/esphome-components.

It seems to be. So any feedback is welcome. Maybe we could upstream that if it proves to be fairly stable.

inforsgit, 2019-12-05:
Hi @ayufan - I'm really keen to test out your code, it looks really good. I'm struggling to get it running though - I've used the config entry on your link but it won't compile - always says `platform not found: climate_eq3`. I'm sure it's my mistake in applying it - I've got the files from the eq3_v2 directory in the same folder as the yaml with the config. Could you maybe add a bit more detail to describe how to install the component from scratch? Thank you!

ayufan, 2019-12-05:
I add installation section: https://github.com/ayufan/esphome-components#1-installation.

inforsgit, 2019-12-05:
Great...thanks - that's amazing - initial test has the basics working 👍 👍 👍. I'll try and run through some proper testing with it and then let you know on your github if I find bugs (and solutions!). Cheers!!

ayufan, 2019-12-05:
@inforsgit I still see some problem with time configuration. Not sure why, something is broken there.

ayufan, 2019-12-05:
I fixed a bug with time. Update is on GitHub.

konzolec, 2019-12-08:
Ideas whats wrong here? MAC from the valve is 00:1A:22:0E:14:73 and ESP32 is next to the valve

`[10:22:47][C][template.switch:058]: Template Switch 'Refresh Bedroom EQ3'
[10:22:47][C][template.switch:059]:   Restore State: NO
[10:22:47][C][template.switch:060]:   Optimistic: NO
[10:22:47][C][logger:175]: Logger:
[10:22:47][C][logger:176]:   Level: DEBUG
[10:22:47][C][logger:177]:   Log Baud Rate: 115200
[10:22:47][C][logger:178]:   Hardware UART: UART0
[10:22:47][C][ota:029]: Over-The-Air Updates:
[10:22:47][C][ota:030]:   Address: 192.168.1.80:3232
[10:22:47][C][mqtt:051]: MQTT:
[10:22:47][C][mqtt:053]:   Server Address: 192.168.1.10:1883 (192.168.1.10)
[10:22:47][C][mqtt:054]:   Username: [redacted]
[10:22:47][C][mqtt:055]:   Client ID: [redacted]
[10:22:47][C][mqtt:057]:   Discovery prefix: 'homeassistant'
[10:22:47][C][mqtt:058]:   Discovery retain: YES
[10:22:47][C][mqtt:060]:   Topic Prefix: 'eq3bt'
[10:22:47][C][mqtt:062]:   Log Topic: 'eq3bt/debug'
[10:22:47][C][mqtt:065]:   Availability: 'eq3bt/status'
[10:22:47][C][sntp:043]: SNTP Time:
[10:22:47][C][sntp:044]:   Server 1: '0.pool.ntp.org'
[10:22:47][C][sntp:045]:   Server 2: '1.pool.ntp.org'
[10:22:47][C][sntp:046]:   Server 3: '2.pool.ntp.org'
[10:22:47][C][sntp:047]:   Timezone: 'CET-1CEST-2,M3.5.0/2,M10.4.0/3'
[10:22:47][C][mqtt.switch:038]: MQTT Switch 'Refresh Bedroom EQ3': 
[10:22:47][C][mqtt.switch:039]:   State Topic: 'eq3bt/switch/refresh_bedroom_eq3/state'
[10:22:47][C][mqtt.switch:039]:   Command Topic: 'eq3bt/switch/refresh_bedroom_eq3/command'
[10:22:47][C][mqtt.sensor:024]: MQTT Sensor 'Bedroom EQ3 Valve State':
[10:22:47][C][mqtt.sensor:026]:   Expire After: 3660s
[10:22:47][C][mqtt.sensor:028]:   State Topic: 'eq3bt/sensor/bedroom_eq3_valve_state/state'
[10:22:47][C][eq3:272]: EQ3-Max Thermostat 'Bedroom EQ3'
[10:22:47][C][eq3:273]:   Update Interval: 1800.0s
[10:22:47][C][eq3:274]:   Mac Address: 1a220e1473
[10:22:47][C][eq3:275]:   Valve 'Bedroom EQ3 Valve State'
[10:22:47][C][eq3:275]:     Unit of Measurement: '%'
[10:22:47][C][eq3:275]:     Accuracy Decimals: 0
[10:22:47][C][eq3:275]:     Icon: 'mdi:percent'
[10:22:49][I][eq3:042]: Requesting update of 1a220e1473...
[10:22:50][E][eq3_cmd:144]: Clock source for 1a220e1473 is not valid.
[10:22:50][W][eq3:051]: Update of 1a220e1473 failed. Tries left: 1.
[10:22:53][I][eq3:042]: Requesting update of 1a220e1473...
[10:22:56][E][eq3_cmd:144]: Clock source for 1a220e1473 is not valid.
[10:22:56][W][eq3:056]: Update of 1a220e1473 failed. Too many tries.
[10:22:56][D][sensor:092]: 'Bedroom EQ3 Valve State': Sending state nan % with 0 decimals of accuracy
[10:24:32][I][ota:046]: Boot seems successful, resetting boot loop counter.`

ayufan, 2019-12-08:
It seems to be missing WiFi connection. Eq3 requires valid time source, and
this has to be taken from somewhere

konzolec, 2019-12-08:
Has to be specified somewhere else? wifi: not enough?

Complete YAML config > https://pastebin.com/EkR7WQDD

ayufan, 2019-12-08:
You seem to miss dns, because of manual ip.

konzolec, 2019-12-08:
Ok, let me try without that

konzolec, 2019-12-08:
Issue remains, looks like it cannot connect to my valve. Do i need to set up anything extra on the valve? MAC is correct ( 00:1A:22:0E:14:73) cuz it was working with this plugin https://github.com/softypit/esp32_mqtt_eq3. Wanted to test this solution

[13:42:08][I][eq3:062]: Requesting ID of 1a220e1473...
[13:42:13][W][eq3_cmd:052]: Cannot connect to 1a220e1473.
[13:42:13][W][eq3:071]: ID of 1a220e1473 failed. Too many tries.
[13:42:13][I][eq3:076]: Requesting Schedule of 1a220e1473...
[13:42:18][W][eq3_cmd:052]: Cannot connect to 1a220e1473.
[13:42:18][W][eq3:091]: Schedule of 1a220e1473 failed. Too many tries.
[13:42:18][I][eq3:042]: Requesting update of 1a220e1473...
[13:42:23][W][eq3_cmd:052]: Cannot connect to 1a220e1473.
[13:42:23][W][eq3:051]: Update of 1a220e1473 failed. Tries left: 2.
[13:42:26][I][eq3:042]: Requesting update of 1a220e1473...
[13:42:31][W][eq3_cmd:052]: Cannot connect to 1a220e1473.
[13:42:31][W][eq3:051]: Update of 1a220e1473 failed. Tries left: 1.
[13:42:34][I][eq3:042]: Requesting update of 1a220e1473...
[13:42:39][W][eq3_cmd:052]: Cannot connect to 1a220e1473.
[13:42:39][W][eq3:056]: Update of 1a220e1473 failed. Too many tries

ayufan, 2019-12-08:
It can be other problem. Can you look in logs in verbose? It might not be
supported maybe.

konzolec, 2019-12-09:
I tried to compile with verbose but couldn't. Im getting a weird error.

This kind of logging compiles fine
logger:
  level: DEBUG

But this one doesnt, might be a esphome error?

logger:
  level: VERBOSE

https://pastebin.com/9F3qZ9sL

ayufan, 2019-12-09:
I pushed a fix for `VERBOSE` mode.

konzolec, 2019-12-09:
Great, here is now the VERBOSE log. If you can get anything useful from it

https://pastebin.com/MkjMfWzz

ayufan, 2019-12-09:
No idea. Can you run `esp32_ble_tracker` to scan all devices with their
signal strength from ESP32?

konzolec, 2019-12-09:
Played with bit valve BLE ON/OFF and it seems it connected now but disconected afterwards

LOG > https://pastebin.com/VfVY5Ct3

ayufan, 2019-12-09:
This.: `State parse of 1a220e1473 failed. Invalid size: 6.`

You have older version of SW: `[18:30:48][I][eq3:255]: 'Bedroom EQ3': Version: 110`. I think that I run it against `Version: 120` only.

ayufan, 2019-12-09:
I pushed a small change that adds logging for the state when is not understood :)

konzolec, 2019-12-10:
I update the valve firmware via their official app to 1.20. And now it's looking promising

[08:05:16][I][eq3:042]: Requesting update of 1a220e1473...
[08:05:18][I][eq3:049]: Update of 1a220e1473 succeeded.
[08:05:18][I][eq3:168]: 'Bedroom EQ3': Valve: 82%. Target: 25.0. Mode: manual dst.
[08:05:18][I][eq3:173]: 'Bedroom EQ3': Window Open: 15 minutes. Temp: 12.0.
[08:05:18][I][eq3:179]: 'Bedroom EQ3': Temp: Comfort: 21.0. Eco: 17.0. Offset: 0.0.
[08:05:18][D][sensor:092]: 'Bedroom EQ3 Valve State': Sending state 82.00000 % with 0 decimals of accuracy
[08:05:18][D][climate:172]: 'Bedroom EQ3' - Sending state:
[08:05:18][D][climate:175]:   Mode: HEAT
[08:05:19][D][climate:186]:   Target Temperature: 25.00°C

OttoWinter, 2019-12-10:
@ayufan Thanks for working on this!

I've had a look at the code and it does look pretty clean (using c++ functional in a couple of places where it fits well etc).

However, I can't really comment much more because I don't quite understand the abstractions you created. Could you please update the code with some documentation for some classes/functions?

Also, from the logs from @konzolec it looks like the connection code is blocking, right? If a BLE connection takes more than a couple milliseconds it will need to be offloaded into another task in order not to block the main thread/event loop.

ayufan, 2019-12-10:
@OttoWinter 

Yes, it is blocking. It was done on a purpose, as concurrent access to WiFi and BLE just does not work well. This is why any interaction with WiFi (sending logs, or commands) during this short connect (~3s) causes node instability.

I tried multiple approaches, this seemed like an only one that was somehow stable (in conditions with logs within connection disabled). Otherwise, node would crash, would loose WiFi beacons, TCP would be blocked, web interface would not responsive for brief moments, etc.

pieter, 2019-12-16:
@ayufan tried out your plugin, seems to be working, great!

A few suggestions:

* Would be nice to be able to adjust retry count through config
* Is there any way we can make this work with esp32_ble_tracker at the same time?
* Would be nice to be able to point to a temperature sensor to provide current temperature, rather than the 0 it's set to now. So have a config option that points to the ID of another sensor component, and read to update the component state. Might take a look at this myself. Otherwise, setting `current_temperature = target_temperature` might be nicer than 0, and is what the built-in EQ3 home-assistant plugin also does.
* "auto" mode shows up as "Heat/Cool" in HomeAssistant, but I think this is probably more related to the HA side (@OttoWinter ?)

ayufan, 2019-12-16:
> Would be nice to be able to adjust retry count through config

Good idea.

> Is there any way we can make this work with esp32_ble_tracker at the same time?

This is made on purpose. No, it just doesn't work stable with the concurrent WiFi/BT access otherwise. Maybe it would be possible to run it, but I prefer to avoid that :)

> Would be nice to be able to point to a temperature sensor to provide current temperature, rather than the 0 it's set to now

Seems like a good idea, but otherwise it will behave as today. So, if there's no temp, to HA it will send information that there's no current temp support, but internally set the current to target.

> "auto" mode shows up as "Heat/Cool" in HomeAssistant

Maybe this is mine. Not sure. There's problem that Climate is quite limited and it is not easy to expose boost for example.

inforsgit, 2019-12-17:
For the current temperature issue, I also find it a bit annoying that in the lovelace thermostat card there's a big "0" in the middle of the thermostat :(

I don't think this is an issue with this support though - as you mention @ayufan - it currently sends the information that there's not current temperature support to ESPHome. However, I think this is an HA issue - in their climate platform (https://developers.home-assistant.io/docs/en/entity_climate.html) I don't see an option to specify that current temperature is not supported.

Therefore three questions:

1. @OttoWinter - what does ESPHome do when `supports_current_temperature` is set to `false`? Does it pass this on to HA, and if so, how?
2. If this is an HA/Lovelace issue, should we raise something with them there?
3. Should this component report current temperature as target temperature, as the HA EQ3 integration (https://www.home-assistant.io/integrations/eq3btsmart/) does?

ayufan, 2019-12-17:
> Should this component report current temperature as target temperature, as the HA EQ3 integration

Well, this seems like a bad idea and workaround. It is not real temperature.

inforsgit, 2019-12-17:
@ayufan - I agree - it's a very nasty way to make the Lovelace thermostat card look right!

inforsgit, 2019-12-26:
Hi! Is there any reason why MQTT is mandatory for this component? I'm using the Home Assistant API connection so I don't actually need the MQTT but it won't compile without it. In addition I'm having some issues with MQTT, as reported in https://github.com/esphome/issues/issues/951.

Ilpredatore, 2020-02-14:
Hello everybody! any news regarding developing eq3 integration?
thank you very much

aqualx, 2020-02-14:
> Hi! Is there any reason why MQTT is mandatory for this component? I'm using the Home Assistant API connection so I don't actually need the MQTT but it won't compile without it. In addition I'm having some issues with MQTT, as reported in [esphome/issues#951](https://github.com/esphome/issues/issues/951).

MQTT is mandatory only if you have _expire_after_ setting:
```
    valve: # optional, allows to see valve state in %
      expire_after: 61min
```

aqualx, 2020-02-14:
> Hello everybody! any news regarding developing eq3 integration?
> thank you very much

It is already working with [custom_component ](https://github.com/ayufan/esphome-components) made by @ayufan.
Works nice without any major issues at all.
In my case ESP-32 should be not farther than 3-5m from EQ-3 for stable connection.

Gravitate1, 2020-03-07:
@ayufan Your component works really well.
However, I find that it becomes unstable and constantly loses wifi connection when linked to more than a single EQ-3.

Does this happen for anyone else?

Is there any configuration that would help prevent this as it's not really viable for me to set up one esp32 board for every EQ-3 in the house?

Truck0ff, 2020-06-25:
Can we implement a custom Bluetooth sensor that can be associated with the esp32 Bluetooth module and read its values.
For me I have door switches and the climate controlled modules.
Sorry my python is very rusty and getting into esphome again after a few years.
I have 3 esp32, and 3 Bluetooth door sensors from Bunnings in the grid connect series. Along with the Bluetooth climate sensors.
I haven’t pulled them apart yet to look at the board or the architecture but don’t know yet if the integration to the esp32 through Bluetooth would be a massive gain.

Abbadon89, 2020-08-13:
> > Is there any way we can make this work with esp32_ble_tracker at the same time?
> 
> This is made on purpose. No, it just doesn't work stable with the concurrent WiFi/BT access otherwise. Maybe it would be possible to run it, but I prefer to avoid that :)

I tried to use one esp as 'bridge' for one eq3 and few miflora devices but esp32 ble tracker is required to do so. Any way to bypass that limitation?

yunnanpl, 2021-11-19:
In order to do so (multiple functionalities on esp32), and as I had some stability issues with https://github.com/softypit/esp32_mqtt_eq3, I decided to rewrite it in python (I am less bad at python :D).
At the moment it has basic eq3 functionality, and it reads Mijia thermometers. I will add ATC thermometers later.
Furthermore, it has homeassistant mqtt autodiscovery functionality, so there is no need to configure the climate entities (as it is quite burdensome). I see no issue to add miiflora to it.
In addition, it scans BLE every 1-2 minutes, as a preparation for device tracking and presence detection. You are free to check it out: https://github.com/yunnanpl/esp32_python_eq3

h0jeZvgoxFepBQ2C, 2022-09-19:
With the new bluetooth proxy support, would it be possible to just forward the eq3 bluetooth data to home assistant via proxy?

h0jeZvgoxFepBQ2C, 2022-09-20:
Sorry if I ask you @OttoWinter , but is this something which Home assistant has to implement or is there a support code required in esphome to make eq3 bt thermostat running via bluetooth proxy?

nagyrobi, 2022-09-20:
The Bluetooth proxy of ESPHome currently only provides Home Assistant with passive sensor data that is advertised by certain devices. Not all devices are supported and ESPHome does not decode or keep a list.

To find out if your device is supported, please search for it in the [Home Assistant Integrations](https://www.home-assistant.io/integrations/) list.

The Individual device integrations in Home Assistant (such as BTHome) will receive the data from the Bluetooth Integration in Home Assistant which automatically aggregates all ESPHome bluetooth proxies with any USB Bluetooth Adapters you might have.

h0jeZvgoxFepBQ2C, 2022-09-20:
Ok, so I guess to make the eq3 thermostat running Home Assistant would have to add some code for it as far as I understood? At least to get the passive sensor data (temperature)? Active control is not possible then I guess?

inforsgit, 2022-09-20:
I'm really hopeful that the bluetooth proxy function will one day allow for EQ3 control via home assitant but indeed, my understanding is that it's not currently there. In the Home Assitant release notes it mentions indeed that 

> Bluetooth proxies are currently limited to forwarding passive data. Active connections are coming in a future release.

So I assume this means that EQ3 control would need this future release to make it happen.

In terms of an EQ3 integration for Home Assitant, I believe it's already there: https://www.home-assistant.io/integrations/eq3btsmart/ - it would just be a case of this using the proxies.

h0jeZvgoxFepBQ2C, 2022-09-20:
So you think the passive temperature transmission should be working already with the bluetooth proxy?

nagyrobi, 2022-09-20:
Wow that's a really smart one! A thermostat without a built-in temperature sensor! Who would have thought...

ayufan, 2022-09-20:
> Wow that's a really smart one! A thermostat without a built-in temperature sensor! Who would have thought...

It has sensor, but it is not exposed via BT since the control of thermostat is done on separate IC while BT is after-though on separate IC.

h0jeZvgoxFepBQ2C, 2022-09-20:
> As the device doesn’t contain a temperature sensor ([read more](https://forum.fhem.de/index.php/topic,39308.15.html)), we report target temperature also as current one.

Ahhhh damn, forgot about this. Yeah, then the bluetooth proxy is no solution at all at the moment unfortunately :( Maybe then when the active bluetooth relaying comes in HA...

000al000, 2022-10-06:
In hass eq3 component is broken now

h0jeZvgoxFepBQ2C, 2022-10-06:
Is there a issue already created on github? If yes, could you link it?

000al000, 2022-10-06:
https://github.com/home-assistant/core/issues/76040

000al000, 2022-10-13:
I tried Active Bluetooth Proxy - no success :(

[6 more replies]

closed 2024-02-21 by bbbenji