esphome/feature-requests #89: Add native API Home Assistant event fire action

OttoWinter, 2019-02-28:
**Describe the problem you have/What new integration you would like**

ESPHome has triggers like on_click etc that can be used in automations - in HA terms these are events. Currently it's a bit hard to get these triggers into HA, you need to set up a `homeassistant.service` action:

```yaml
binary_sensor:
- platform: abc
  # ..
  on_press:
  - homeassistant.service:
      service: script.execute
      # ... etc
```

that is a whole bunch of yaml for something quite simple: getting events across these two systems.

So, let's add an `homeassistant.event` action like this

```yaml
binary_sensor:
- platform: abc
  # ..
  on_press:
  - homeassistant.event: esphome.my_gpio_triggered
```

So this will generate an event in the event bus in HA called `esphome.my_gpio_triggered` with no args.

```yaml
automation:
  trigger:
    platform: event
    event_type: esphome.my_gpio_triggered
```

Optionally, the user can also supply arguments:

```yaml
binary_sensor:
- platform: abc
  # ..
  on_press:
  - homeassistant.event: 
      event: esphome.my_gpio_triggered
      data:
        hello: world
        here: !lambda 'return "this";'
```

xalexslx, 2019-03-09:
I would like this feature as well. It's nice to have specially if you want to expose multiple clicks trigger to HA.

tatunts, 2019-03-11:
This is amazing! Thanks @OttoWinter for the great work.

balloob, 2019-03-25:
For security purposes we should not allow ESPHome devices be able to fire **any** event. Limit it to either events prefixed with `esphome.` or `esphome_` or just have a single `esphome_event` and make it mandatory for users to specify a `type` in the event data.

closed 2019-06-18 by OttoWinter