esphome/feature-requests #92: ESP8266 capacitive touch sensor using GPIO

redskinhu, 2019-03-03:
**Describe the problem you have/What new integration you would like**
I would like create a capacitive touch switch using ESP8266 GPIO's.

**Please describe your use case for this integration and alternatives you've tried:**
I would like to use the ESP8266 GPIO's instead of using TTP223 or ESP32. I saw many of versions but they use the Arduino Capacitive Sensing Library. See: http://playground.arduino.cc/Main/CapacitiveSensor

Is it possible to integrate this feature or someone have / can create Custom Binary Sensor code for this problem? Or any other solution?

Thanks
T

BananaPukeh, 2019-03-05:
I managed to implement this by toggling the ground of the switch when you touch is, so the button would be reset.

```
binary_sensor:
  - platform: gpio
    pin:
      number: D1
      mode: INPUT_PULLUP
    name: 'MCU 4 Button'
    id: 'button'
    on_press:
      then:
        - if: 
            condition:
              - switch.is_off: 'button_switch'
            then:
              - delay: 1s
              - if: 
                  condition:
                    - binary_sensor.is_on: 'button'
                  then:
                    - switch.turn_on: 'button_switch'
                    - delay: 200ms
                    - switch.turn_off: 'button_switch'
    
    
switch:
  - platform: gpio
    name: "button switch"
    id: button_switch
    pin: D2
    internal: true
```

redskinhu, 2019-03-05:
Hello

Thanks for the quick answer. Is the hardware like this? http://digitalmedia.risd.edu/pbadger/PC/index.php?n=Devices.CapacitiveSensing

Thx
T

redskinhu, 2019-03-05:
Hi

I've checked the code. I think you are talking about toggling of one pushbutton...? Isn't it?

I'm still looking for a very simple capacitive touch button solution.

Thx
T

BananaPukeh, 2019-03-05:
Im using this button, it has an integrated light which turns on if the button is on. Unfortunately you cannot control the state/led of this button using it pins. So I’ve hacked something together that resets the button when it is pushed by turning the ground of the button off for a moment. 

https://s.click.aliexpress.com/e/GPSG0d2

BananaPukeh, 2019-03-05:
Isnt it possible to use a captive touch button with the binary_sensor component checking if a pin is pulled high/low?

redskinhu, 2019-03-06:
Follow the link in the 1st post and read it.

Short extract: 
It needs 2 pins, one for charge one for sensing, connected by high resistance and the sensing surface is connetced to the sensing pin too. Create a loop to toggle the charge pin and measure the the time when the sense pin reaches same state like the charge pin. If the sensor is touched the charging time is different than with the untouched sensor. The capacitiveSensor Arduino library can do this.

I want to create a simple (small) light switch but I don't need the performance/pins of the ESP32 (it is supports the touch) see: https://esphome.io/components/binary_sensor/esp32_touch.html

And i don't want to use any unecessary hw like TTP223.

I need only one resistor and one cap and the sw support from the esphome side. I'm just a power user not a developer so i can't implement it. Maybe i can but it takes the half of my life... ;-)

I have resistor and cap but there is no sw support yet. I'm looking forward to.

Joshfindit, 2019-10-17:
@BananaPukeh You're using different hardware in your solution. It's possible to use just 2 pins with a resistor between them as per @redskinhu's link http://digitalmedia.risd.edu/pbadger/PC/index.php?n=Devices.CapacitiveSensing

Following as I'm also curious about capacitive touch on the ESP8266 (D1 mini)

Mossfeldt, 2020-04-10:
Would also like this to be implemented. Could be of great use for several of my DIY-projects. :)

RubenKelevra, 2021-12-13:
Well, you can use the analog input pin for this. There's the unpowered version:

https://siytek.com/touch-switch-esphome/

And a powered version:

https://www.instructables.com/DIY-Touch-SensorSwitch-ESP8266-Node-MCU/

Padm59, 2022-03-29:
Somebody made any progress here?

closed 2022-07-01 by nagyrobi