esphome/feature-requests #93: Docker ARM Images (for RPi

ichbestimmtnicht, 2019-03-03:
**Describe the problem you have/What new integration you would like**

Please add Docker ARM images. This is kind of an issue because when using the commands (on latest dietpi [with "Docker version 18.06.3-ce, build d7080c1"
 installed - i think so]) I am getting this error: standard_init_linux.go:190: exec user process caused "exec format error"

**Please describe your use case for this integration and alternatives you've tried:**

Having a simple ARM servernode from where you can update your configuration at all times (from mobile for example) is the way i wanna integrate it into the system. (+haproxy for a wakeupcall but this isn't directly related and already tested on x64.)

**Additional context**

OttoWinter, 2019-03-03:
https://hub.docker.com/u/esphome - they are already there

closed 2019-03-03 by OttoWinter

ichbestimmtnicht, 2019-03-03:
I am sorry, should I (or you) add it to the documentation to help new users out? (No need for an answer if you do it yourself or don't want it.

Great Response time!

OttoWinter, 2019-03-05:
Sure, that's possible. I think this would be the best place for it: https://esphome.io/guides/faq#docker-reference