esphome/feature-requests #94: iBeacon tracking

rdehuyss, 2019-03-03:
**Describe the problem you have/What new integration you would like**
BLE tracking - can't we use the UUID in esphome yaml files to track phones?

**Please describe your use case for this integration and alternatives you've tried:**
I have an app on my Android phone that mimics an iBeacon and emits a constant UUID. Can't this be used to do room based presence detection.

**Additional context**
In BeaconScope (an Android app) I can emit different types of beacons and configure the uuid. Can I use esphome to track this iBeacon.

fakezeta, 2019-03-23:
Hi, as asked from @OttoWinter in Discord I'm posting the logs of my ble_tracker with VERY_VERBOSE logging.

The iBeacon peripheral UUID is 4A13716B-21B1-442F-9246-76A870AB4D66 with  Service UUID: 1111

```
Showing logs:

[12:22:23][VV][esp32_ble_tracker:545]: Parse Result:
[12:22:23][VV][esp32_ble_tracker:565]:   Address: FB:61:BD:53:04:94 (RANDOM)
[12:22:23]) wifi: state: assoc -> run (10)
[12:22:23][VV][esp32_ble_tracker:568]:   N
[12:22:23][VV][esp32_ble_tracker:568]:   Name: 
[12:22:23][VV][esp32_ble_tracker:576]:   Ad Flag: 4
[12:22:23][VV][esp32_ble_tracker:581]:   M, channel 6
[12:22:23][VV][esp32_ble_tracker:582]:   Service data
[12:22:23][0;778) wifi: pm start, type: 0
[12:22:23]
[12:22:23]I 0;38m[VV][esp32_ble_tracker:597
[12:22:23][VV][esp32_ble_tracker:597]: Adv data: 02.01.04.1B.FF.4C.00.02.15.EB.EF.D0.83.70.A2.47.C8.98.37.E7.B5.63.4D.F5.24.00.01.00.01.CB.59. (31 bytes)
[12:22:23][VV][esp32_ble_tracker:313]: Xiaomi no service data
[12:22:23][D][esp32_ble_tracker:438]: Found device FB:61:BD:53:04:94 RSSI=-84
[12:22:23][D][esp32_ble_tracker:459]:   Address Type: RANDOM
[12:22:25][VV][esp32_ble_tracker:545]: Parse Result:
[12:22:25][VV][esp32_ble_tracker:565]:   Address: 4B:BA:D4:65:57:06 (RANDOM)
[12:22:25][VV][esp32_ble_tracker:567]:   RSSI: -53
[12:22:25][VV][esp32_ble_tracker:568]:   Name: 
[12:22:25][VV][esp32_ble_tracker:576]:   Ad Flag: 26
[12:22:25][VV][esp32_ble_tracker:581]:   Manufacturer data: 'L'
[12:22:25][VV][esp32_ble_tracker:582]:   Service data: ''
[12:22:25][VV][esp32_ble_tracker:597]: Adv data: 02.01.1A.0A.FF.4C.00.10.05.07.1C.99.4F.FF. (14 bytes)
[12:22:25][VV][esp32_ble_tracker:313]: Xiaomi no service data
[12:22:25][D][esp32_ble_tracker:438]: Found device 4B:BA:D4:65:57:06 RSSI=-53
[12:22:25][D][esp32_ble_tracker:459]:   Address Type: RANDOM
[12:22:27][VV][esp32_ble_tracker:545]: Parse Result:
[12:22:27][VV][esp32_ble_tracker:565]:   Address: 4B:BA:D4:65:57:06 (RANDOM)
[12:22:27][VV][esp32_ble_tracker:567]:   RSSI: -46
[12:22:27][VV][esp32_ble_tracker:568]:   Name: 
[12:22:27][VV][esp32_ble_tracker:576]:   Ad Flag: 26
[12:22:27][VV][esp32_ble_tracker:581]:   Manufacturer data: 'L'
[12:22:27][VV][esp32_ble_tracker:582]:   Service data: ''
[12:22:27][VV][esp32_ble_tracker:597]: Adv data: 02.01.1A.1A.FF.4C.00.0C.0E.00.E2.67.C0.73.5A.F7.8E.D4.6C.93.59.3F.8C.10.05.0B.1C.99.4F.FF. (30 bytes)
[12:22:27][VV][esp32_ble_tracker:313]: Xiaomi no service data
[12:22:27][V][esp32_ble_tracker:432]: Already discovered device 4B:BA:D4:65:57:06
[12:22:30][VV][esp32_ble_tracker:545]: Parse Result:
[12:22:30][VV][esp32_ble_tracker:565]:   Address: FB:61:BD:53:04:94 (RANDOM)
[12:22:30][VV][esp32_ble_tracker:567]:   RSSI: -83
[12:22:30][VV][esp32_ble_tracker:568]:   Name: 
[12:22:30][VV][esp32_ble_tracker:576]:   Ad Flag: 4
[12:22:30][VV][esp32_ble_tracker:581]:   Manufacturer data: 'L'
[12:22:30][VV][esp32_ble_tracker:582]:   Service data: ''
[12:22:30][VV][esp32_ble_tracker:597]: Adv data: 02.01.04.1B.FF.4C.00.02.15.EB.EF.D0.83.70.A2.47.C8.98.37.E7.B5.63.4D.F5.24.00.01.00.01.CB.59. (31 bytes)
[12:22:30][VV][esp32_ble_tracker:313]: Xiaomi no service data
[12:22:30][V][esp32_ble_tracker:432]: Already discovered device FB:61:BD:53:04:94
[12:22:30][D][esp32_ble_tracker:165]: Starting scan...
[12:22:30][D][WiFiGeneric.cpp:345] _eventCallback(): Event: 7 - STA_GOT_IP
[12:22:30][D][WiFiGeneric.cpp:389] _eventCallback(): STA IP: 192.168.1.72, MASK: 255.255.255.0, GW: 192.168.1.1
[12:22:30][V][wifi:1251]: Event: Got IP static_ip=192.168.1.72 gateway=192.168.1.1
[12:22:30][D][WiFiGeneric.cpp:345] _eventCallback(): Event: 7 - STA_GOT_IP
[12:22:30][D][WiFiGeneric.cpp:389] _eventCallback(): STA IP: 192.168.1.72, MASK: 255.255.255.0, GW: 192.168.1.1
[12:22:30][V][wifi:1251]: Event: Got IP static_ip=192.168.1.72 gateway=192.168.1.1
[12:22:30][I][wifi:356]: WiFi connected!
[12:22:30][C][wifi:244]:   SSID: 'SweetHome'
[12:22:30][C][wifi:245]:   IP Address: 192.168.1.72
[12:22:30][C][wifi:247]:   BSSID: 22:FA:2D:24:56:99
[12:22:30][C][wifi:248]:   Hostname: 'ingresso_tracker'
[12:22:30][C][wifi:252]:   Signal strength: -48 dB ▂▄▆█
[12:22:30][C][wifi:253]:   Channel: 6
[12:22:30][C][wifi:254]:   Subnet: 255.255.255.0
[12:22:30][C][wifi:255]:   Gateway: 192.168.1.1
[12:22:30][C][wifi:256]:   DNS1: 1.1.1.1
[... 95 lines cut ...]
[12:22:52][D][esp32_ble_tracker:459]:   Address Type: RANDOM
[12:22:52][D][esp32_ble_tracker:461]:   Name: 'Blank'
[12:22:52][VV][esp32_ble_tracker:545]: Parse Result:
[12:22:52][VV][esp32_ble_tracker:565]:   Address: 4B:BA:D4:65:57:06 (RANDOM)
[12:22:52][VV][esp32_ble_tracker:567]:   RSSI: -48
[12:22:52][VV][esp32_ble_tracker:568]:   Name: Blank
[12:22:52][VV][esp32_ble_tracker:576]:   Ad Flag: 26
[12:22:52][VV][esp32_ble_tracker:579]:   Service UUID: 11:1111
[12:22:52][VV][esp32_ble_tracker:581]:   Manufacturer data: 'L'
[12:22:52][VV][esp32_ble_tracker:582]:   Service data: ''
[12:22:52][VV][esp32_ble_tracker:597]: Adv data: 02.01.1A.03.03.11.11.06.09.42.6C.61.6E.6B.0A.FF.4C.00.10.05.0B.1C.99.4F.FF. (25 bytes)
[12:22:52][VV][esp32_ble_tracker:313]: Xiaomi no service data
[12:22:52][V][esp32_ble_tracker:432]: Already discovered device 4B:BA:D4:65:57:06
[12:22:54][VV][esp32_ble_tracker:545]: Parse Result:
[12:22:54][VV][esp32_ble_tracker:565]:   Address: 4B:BA:D4:65:57:06 (RANDOM)
[12:22:54][VV][esp32_ble_tracker:567]:   RSSI: -47
[12:22:54][VV][esp32_ble_tracker:568]:   Name: Blank
[12:22:54][VV][esp32_ble_tracker:576]:   Ad Flag: 26
[12:22:54][VV][esp32_ble_tracker:579]:   Service UUID: 11:1111
[12:22:54][VV][esp32_ble_tracker:581]:   Manufacturer data: 'L'
[12:22:54][VV][esp32_ble_tracker:582]:   Service data: ''
[12:22:54][VV][esp32_ble_tracker:597]: Adv data: 02.01.1A.03.03.11.11.06.09.42.6C.61.6E.6B.0A.FF.4C.00.10.05.0B.1C.99.4F.FF. (25 bytes)
[12:22:54][VV][esp32_ble_tracker:313]: Xiaomi no service data

```

rdehuyss, 2019-03-27:
Here is a solution on how to learn about the service uuid: https://platformio.org/lib/show/1841/ESP32%20BLE%20Arduino.

Basically, one needs to connect to the server and request the service with the configured uuid.

Or, so I think it should work :-)

tottka, 2019-07-03:
This would be a fantastic upgrade.

atomicpapa, 2019-09-04:
HUGE + 1 on this request.  Every freaking phone in my house randomizes BT Mac addresses

Lumpusz, 2019-10-16:
> Hi, as asked from @OttoWinter in Discord I'm posting the logs of my ble_tracker with VERY_VERBOSE logging.
>
> The iBeacon peripheral UUID is 4A13716B-21B1-442F-9246-76A870AB4D66 with Service UUID: [...]

This looks very promising. Can you post source of your version of the tracker? Have you found any downsides to it?

Lumpusz, 2019-10-17:
I took a closer look at this. Sercive UUIDs are already parsed, but they are not matched with expectations. I believe the sensor needs to be modified to accept not just a MAC, but a uuid as well, and the sensor would match that too. Raw uuid is not exposed though as it is protected, so api needs to change for that.

Lumpusz, 2019-10-18:
I was able to track my android phone with a custom sensor, based on svc UUID. See logs:

```[12:07:07][D][my_ble_rssi:032]: Found device with Address: 5E:DF:DE:1C:66:69 by Service UUID: 11:1111
[12:07:07][D][sensor:092]: 'Android Phone 1': Sending state -80.00000  with 0 decimals of accuracy
[12:07:09][D][my_ble_rssi:032]: Found device with Address: 5E:DF:DE:1C:66:69 by Service UUID: 11:1111
[12:07:09][D][sensor:092]: 'Android Phone 1': Sending state -80.00000  with 0 decimals of accuracy
[12:07:12][D][my_ble_rssi:032]: Found device with Address: 5E:DF:DE:1C:66:69 by Service UUID: 11:1111
[12:07:12][D][sensor:092]: 'Android Phone 1': Sending state -83.00000  with 0 decimals of accuracy
[12:07:22][D][my_ble_rssi:032]: Found device with Address: 5B:45:7F:6D:57:76 by Service UUID: 11:1111
[12:07:22][D][sensor:092]: 'Android Phone 1': Sending state -80.00000  with 0 decimals of accuracy
[12:07:27][D][my_ble_rssi:032]: Found device with Address: 5B:45:7F:6D:57:76 by Service UUID: 11:1111
[12:07:27][D][sensor:092]: 'Android Phone 1': Sending state -81.00000  with 0 decimals of accuracy```

I restarted the beacon, so the MAC address changed but the custom sensor was still able to identify it based on svc UUID.
Code is messy, I will work on a proper PR when I have more time.

Lumpusz, 2019-10-21:
@OttoWinter I was testing the fix locally, on latest dev branch, but the sw that I compile can not run on the board.

Steps to reproduce:

1. create local clone of latest dev branch
1. run pip install ./esphome
1. run python -m esphome test_sensor.yaml compile
1. serial flash with ESPHome-Flasher-1.1.0-windows-x64.exe
1. press EN on board

```
esphome:
  name: test_sensor
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "xxx"
  password: "xxx"

esp32_ble_tracker:
  scan_interval: 10s

sensor:
  - platform: ble_rssi
    mac_address: AC:37:43:77:5F:4C
    name: "Test MAC"

api:

ota:
```

```
[21:09:44]ets Jun  8 2016 00:22:57
[21:09:44]
[21:09:44]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[21:09:44]configsip: 0, SPIWP:0xee
[21:09:44]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[21:09:44]mode:DIO, clock div:2
[21:09:44]load:0x3fff0018,len:4
[21:09:44]load:0x3fff001c,len:952
[21:09:44]load:0x40078000,len:6084
[21:09:44]load:0x40080000,len:7944
[21:09:44]entry 0x40080310
```
Is this a known issue?