esphome/feature-requests #96: Include EmonLib for non-intrusive measurements on power lines

tubalainen, 2019-03-06:
**Describe the problem you have/What new integration you would like**
It would be fantastic if we could use the sample measuring library of EmonLib for ei measuring power consumption.
Reference to such a project in Arduino IDE. This is in correspondence with a "Split Core Current Transformer" connected to the A0 pin on an NodeMCU (or similar).
(https://github.com/digiblur/digiDryerMon)

**Please describe your use case for this integration and alternatives you've tried:**
Reading accurate data on the A0 pin without the use of this library would be quite a feat of "lambda hack". Not saying it cannot be done though :)...

**Additional context**

animavitis, 2019-04-08:
i have working custom component for this (works only on esp8266):

```
esphome:
  name: animapower
  platform: ESP8266
  board: nodemcuv2
  includes:
  - customEmonlib.h
  libraries:
  - "EmonLib"

wifi:
  ssid: 'Anima_24'
  password: 'kfiatuszek'

mqtt:
  broker: '192.168.1.13'
  username: ''
  password: ''

sensor:
- platform: custom
  lambda: |-
    auto emonLibSensor = new CustomEmonLib();
    App.register_component(emonLibSensor);
    return {emonLibSensor};

  sensors:
    name: "Power Meter"
    unit_of_measurement: W
    accuracy_decimals: 2

logger:
  level: debug
````

customEmonlib.h
```
#include "esphome.h"
#include "EmonLib.h"                   // Include Emon Library

using namespace esphome;

class CustomEmonLib : public PollingComponent, public sensor::Sensor {
 public:
  EnergyMonitor emon1;                   // Create an instance
  // constructor
  CustomEmonLib() : PollingComponent(15000) {}

  void setup() override {
    // This will be called by App.setup()
    emon1.current(1, 111.1);
  }
  void update() override {
    // This will be called every "update_interval" milliseconds.
    double Irms = emon1.calcIrms(1480);
    publish_state(Irms*230.0);
  }
};
```

animavitis, 2019-05-02:
I made new version of custom component, this one do not have emonlib dependency & works on both ESP8266 & ESP32.
All credits go to https://github.com/capella-ben/emon_esp32 

file PowerMon.h:
```
#include "esphome.h"

using namespace esphome;

#define ADC_BITS 10 //12 for esp32 & 10 for esp8266
#define ADC_COUNTS (1 << ADC_BITS)
const float ICAL = 111.1; // default calibration factor: 90.9 for ESP32, 111.1 fro esp8266
const int powerPin = A0;  // ADC PIN for powerMon: A0 for esp8266, 36 for esp32
double filteredI = 0;
double offsetI, sqI, sumI, Irms;
int sample = 0;

class CustomPowerMon : public PollingComponent, public sensor::Sensor
{
  public:
    CustomPowerMon() : PollingComponent(500) {}

    double calcIrms(unsigned int Number_of_Samples, int pin)
    {
        for (unsigned int n = 0; n < Number_of_Samples; n++)
        {
            sample = analogRead(pin);
            // Digital low pass filter extracts the 2.5 V or 1.65 V dc offset,
            //  then subtract this - signal is now centered on 0 counts.
            offsetI = (offsetI + (sample - offsetI) / 1024);
            filteredI = sample - offsetI;
            // Root-mean-square method current
            // 1) square current values
            sqI = filteredI * filteredI;
            // 2) sum
            sumI += sqI;
        }
        double I_RATIO = ICAL * ((3300 / 1000.0) / (ADC_COUNTS));
        Irms = I_RATIO * sqrt(sumI / Number_of_Samples);
        sumI = 0;
        return Irms;
    }

    void setup() override
    {
        delay(10);
        offsetI = ADC_COUNTS >> 1;
    }
    void update() override
    {
        double Irms = calcIrms(1480, powerPin); // Calculate current
        double dPower = Irms * 235; // Irms * volts
        publish_state(dPower);
    }
};
```

yaml configuration:
```
esphome:
  name: PowerMon
  platform: ESP8266
  board: nodemcuv2
  includes:
  - powerMon.h

sensor:
  - platform: custom
    lambda: |-
      auto powerMonSensor = new CustomPowerMon();
      App.register_component(powerMonSensor);
      return {powerMonSensor};
    sensors:
     name: "Power Meter"
     unit_of_measurement: W
     accuracy_decimals: 0
     filters:
       - sliding_window_moving_average:
           window_size: 150
           send_every: 150
           send_first_at: 150
```

It will measure the consumption every half seconds, but publish average power consumption of the last 150 measurements.

tubalainen, 2019-05-06:
> I made new version of custom component, this one do not have emonlib dependency & works on both ESP8266 & ESP32.
> All credits go to https://github.com/capella-ben/emon_esp32
>
> file PowerMon.h:
>
> ``` [...]

Thanks! <3

Im a "Lamda n00b", would it be easy to re-calculate and create another sensor that measures the current as well as Watts with a input parameter defining the voltage. :)

animavitis, 2019-05-07:
custom component can publish more than one value, so its possible to expose both power & current value at same time.... 
https://esphome.io/components/sensor/custom.html#bonus-sensors-with-multiple-output-values
this article explain it great. For sure u will be able to modify PowerMon.h by yourself :)

jesserockz, 2019-06-08:
This is implemented here: https://esphome.io/components/sensor/ct_clamp.html

closed 2019-06-08 by OttoWinter